freakent/dbus-mqtt-devices #8: How to remove a device?

stundenblume, 2022-04-01:
How can I remove a device?
Is that even necessary? 

While testing I created a lot of different devices. Now I have some unused device instances. Should they be released or doesn't that matter?

freakent, 2022-04-01:
If you go into the regular venus console, there should be an option at the bottom of the menu saying "Remove disconnected devices", that will partially clean things up. Unfortunately the persisted settings are permanent - that's just how Venus uses dbus. Only thing to do is create a clean install of Venus.
[image]

stundenblume, 2022-04-01:
OK, thank you. I don't have that entry.
But doesn't matter. I Don't plan to have that much unused sensors on the production system.

[image]

closed 2022-04-01 by stundenblume