freakent/dbus-mqtt-devices #9: Error with 0.4.x

stundenblume, 2022-04-01:
Both 0.4.x versions don't work for me.
0.3.1 with my own services.yml is working as expected.

0.3.1:
```
@4000000062476b8932614914 INFO:device_manager:[Connected] Result code 0
@4000000062476b8e32c3cfbc ERROR:mqtt_gobject_bridge:[Disconnected] Lost connection to broker
@4000000062476b8e32cfaecc INFO:mqtt_gobject_bridge:[Disconnected] Set timer
@4000000062476b9925897d4c *** CCGX booted (0) ***
@4000000062476b9f23dc80fc *** starting dbus-mqtt-devices ***
@4000000062476bb53a99a004 -------- dbus_mqtt_devices, v0.3.1 is starting up --------
@4000000062476bb53a99bf44 INFO:logger:Loglevel set to INFO
@4000000062476bb60378b1dc INFO:device_manager:Using portalId <id>
@4000000062476bb603a284e4 INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@4000000062476bb60f357c6c INFO:device_manager:[Connected] Result code 0
@4000000062476bd832c6e4cc INFO:device_manager:Received device status message {'clientId': 'a01', 'connected': 1, 'version': 'v1.0 ALPHA', 'services': {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'}}
@4000000062476bd832f38694 INFO:device:**** Registering device: a01, services: {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'} ****
@4000000062476bd83347402c INFO:device:Registering Service temperature for client a01
@4000000062476bd833475b84 INFO:device_service_config:About to open config file
@4000000062476bd91947949c INFO:device_service:Registering service temperature for client a01 at path com.victronenergy.temperature.mqtt_a01_t1
@4000000062476bd91b568edc INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_t1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000062476bd92024c85c INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.temperature.mqtt_a01_t1
@4000000062476bd9205c4f34 INFO:device_service:Registered Service com.victronenergy.temperature.mqtt_a01_t1 under DeviceInstance 5
@4000000062476bd92067094c INFO:device:Registering Service pvinverter for client a01
@4000000062476bd920702d24 INFO:device_service_config:About to open config file
@4000000062476bda05ab2a0c INFO:device_service:Registering service pvinverter for client a01 at path com.victronenergy.pvinverter.mqtt_a01_pv1
@4000000062476bda06489ea4 INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_a01_pv1/CustomName does not exist yet or must be adjusted
@4000000062476bda0bc298e4 INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_pv1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000062476bda0f5387d4 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.pvinverter.mqtt_a01_pv1
@4000000062476bda10fe4204 INFO:device_service:Registered Service com.victronenergy.pvinverter.mqtt_a01_pv1 under DeviceInstance 5
@4000000062476bda10fe652c INFO:device:Registering Service pvinverter for client a01
@4000000062476bda10fe74cc INFO:device_service_config:About to open config file
@4000000062476bda2e74cca4 INFO:device_service:Registering service pvinverter for client a01 at path com.victronenergy.pvinverter.mqtt_a01_pv2
@4000000062476bda2f4ad59c INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_a01_pv2/CustomName does not exist yet or must be adjusted
@4000000062476bda3a43fa3c INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_pv2/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000062476bdb017d1c74 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.pvinverter.mqtt_a01_pv2
@4000000062476bdb0217a30c INFO:device_service:Registered Service com.victronenergy.pvinverter.mqtt_a01_pv2 under DeviceInstance 6
@4000000062476bdb0217c634 INFO:device:Registering Service grid for client a01
@4000000062476bdb0217d5d4 INFO:device_service_config:About to open config file
@4000000062476bdb228d738c INFO:device_service:Registering service grid for client a01 at path com.victronenergy.grid.mqtt_a01_g1
@4000000062476bdb25354a9c INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_g1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000062476bdb28113d84 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.grid.mqtt_a01_g1
@4000000062476bdb28a1dcf4 INFO:device_service:Registered Service com.victronenergy.grid.mqtt_a01_g1 under DeviceInstance 1
@4000000062476bdb28c69f1c INFO:device_manager:publish {'portalId': '<id>', 'deviceInstance': {'t1': '5', 'pv1': '5', 'pv2': '6', 'g1': '1'}} to device/a01/DBus, status is 0
@4000000062476bdb29027f3c INFO:device_service:setting changed, setting: CustomName, old: PV Inverter MQTT, new: PV Inverter MQTT
@4000000062476bdb2916a764 INFO:device_service:setting changed, setting: CustomName, old: PV Inverter MQTT, new: PV Inverter MQTT
```

0.4.0:
```
@40000000624769772b8457bc INFO:device:Registering Service grid for client a01
@40000000624769772b8bbe44 INFO:device_service_config:About to open config file
@400000006247697802186a44 INFO:device_service_config:No configuration for Service grid, please update services.yml
@4000000062476978025caaec INFO:device_service:Registering service grid for client a01 at path com.victronenergy.grid.mqtt_a01_g1
@400000006247697805875a14 *** starting dbus-mqtt-devices ***
@400000006247697928efb6a4 -------- dbus_mqtt_devices, v0.4.1 is starting up --------
@400000006247697928efd1fc INFO:logger:Loglevel set to INFO
@400000006247697929ddbc3c INFO:device_manager:Using portalId <id>
@40000000624769792a01de3c INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000624769792c684364 INFO:device_manager:[Connected] Result code 0
@40000000624769b61c22146c INFO:device_manager:Received device status message {'clientId': 'a01', 'connected': 1, 'version': 'v1.0 ALPHA', 'services': {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'}}
@40000000624769b61c466d1c INFO:device:**** Registering device: a01, services: {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'} ****
@40000000624769b61c4e4104 INFO:device:Registering Service temperature for client a01
@40000000624769b61c56a95c INFO:device_service_config:About to open config file
@40000000624769b62fb347e4 INFO:device_service:Registering service temperature for client a01 at path com.victronenergy.temperature.mqtt_a01_t1
@40000000624769b63165d9e4 INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_t1/ClassAndVrmInstance does not exist yet or must be adjusted
@40000000624769b633d2421c INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.temperature.mqtt_a01_t1
@40000000624769b634269bdc INFO:device_service:Registered Service com.victronenergy.temperature.mqtt_a01_t1 under DeviceInstance 5
@40000000624769b6342fcb6c INFO:device:Registering Service pvinverter for client a01
@40000000624769b634389d3c INFO:device_service_config:About to open config file
[... 15 lines cut ...]
@40000000624769b9229f172c -------- dbus_mqtt_devices, v0.4.1 is starting up --------
@40000000624769b9229f366c INFO:logger:Loglevel set to INFO
@40000000624769b9237464a4 INFO:device_manager:Using portalId <id>
@40000000624769b92396c184 INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000624769b925e6e75c INFO:device_manager:[Connected] Result code 0
```

0.4.0.1:
```
[...]
@40000000624767c70bb62564 *** starting dbus-mqtt-devices ***
@40000000624767c82d14fe24 -------- dbus_mqtt_devices, v0.4.1 is starting up --------
@40000000624767c82d15197c INFO:logger:Loglevel set to INFO
@40000000624767c82df3adf4 INFO:device_manager:Using portalId <id>
@40000000624767c82e1612a4 INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000624767c8305924d4 INFO:device_manager:[Connected] Result code 0
@400000006247681a2621cd7c INFO:device_manager:Received device status message {'clientId': 'a01', 'connected': 1, 'version': 'v1.0 ALPHA', 'services': {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'}}
@400000006247681a2645727c INFO:device:**** Registering device: a01, services: {'t1': 'temperature', 'pv1': 'pvinverter', 'pv2': 'pvinverter', 'g1': 'grid'} ****
@400000006247681a264d2724 INFO:device:Registering Service temperature for client a01
@400000006247681a26550e94 INFO:device_service_config:About to open config file
@400000006247681b03a7dffc ERROR:device_service_config:Unexpected error: <class 'yaml.parser.ParserError'>
@400000006247681b044fa14c INFO:device_service:Registering service temperature for client a01 at path com.victronenergy.temperature.mqtt_a01_t1
@400000006247681b0605e0b4 *** starting dbus-mqtt-devices ***
[...]
```

stundenblume, 2022-04-01:
0.4.0 without "grid":
```
@4000000062476f8924c4fc9c INFO:device:Registering Service pvinverter for client a01
@4000000062476f8924cdf57c INFO:device_service_config:About to open config file
@4000000062476f89374446fc INFO:device_service:Registering service pvinverter for client a01 at path com.victronenergy.pvinverter.mqtt_a01_pv2
@4000000062476f8937f5ef74 INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_a01_pv2/CustomName does not exist yet or must be adjusted
@4000000062476f8a003f6484 INFO:settingsdevice:Setting /Settings/Devices/mqtt_a01_pv2/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000062476f8a02c023c4 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.pvinverter.mqtt_a01_pv2
@4000000062476f8a0345de24 INFO:device_service:Registered Service com.victronenergy.pvinverter.mqtt_a01_pv2 under DeviceInstance 6
@4000000062476f8a036a0024 INFO:device_manager:publish {'portalId': '<id>', 'deviceInstance': {'t1': '5', 'pv1': '5', 'pv2': '6'}} to device/a01/DBus, status is 0
@4000000062476f8a03a7247c INFO:device_service:setting changed, setting: CustomName, old: PV Inverter MQTT, new: PV Inverter MQTT
@4000000062476f8a03b99724 INFO:device_service:setting changed, setting: CustomName, old: PV Inverter MQTT, new: PV Inverter MQTT
```

freakent, 2022-04-01:
Hi @stundenblume, As always, thanks for doing this.
Couple of observations,
1) 0.4.0 didn't include grid in the services.yml file, there is a warning in the log file but then it looks like it crashed which is not good. I'll raise a separate issue for that.
2) The services.yml file in 0.4.0.1 has a small formatting error so the parser errored. I just fixed the formatting error and pushed the change up to github. You can replace your version with this one.
https://raw.githubusercontent.com/freakent/dbus-mqtt-devices/main/services.yml

Out of interest, are you updating the logs to manually remove your portal id or is that not working on your system? :-)

stundenblume, 2022-04-01:
with the updated 0.4.0.1 it works!

I replaced the portalID.

closed 2022-04-01 by stundenblume

freakent, 2022-04-02:
Top tip, if there appears to be something not right with the services.yml go to http://www.yamllint.com/ and copy paste your entire services.yml file in the site and press Go. That's how I found the formatting error.