freakent/dbus-mqtt-devices #11: Dokumentation request

Hofyyy, 2022-04-30:
Hello Martin,

first of all. Thanks for the great project.
I try to connect a "Powerfox" Grid meter to my Victron management system.
Until now I have a MQTT Topic which updates every 3 sec. the WATT number from all phases.

I am not sure if this questions are right in the Issue section, but i did not find a discussion page or something like that.

Now I have to ask stupid questions. But maybe this helps to enhance the documentation.

1. clientId
- On which MQTT server or client is this generated?
- How can I find it?
At the moment I use my own mosquito, Redflow and MQTT Explorer to try to generate exact the tuple your driver would need.
So I play locally before pushing wrong data to your driver.

2.  And maybe you could also add a "grid" example?

freakent, 2022-04-30:
I have no problem with people using Raising Issues to ask for help. Better here than on Victron forum as I get automatic alerts from GitHub.

freakent, 2022-04-30:
The ClientId is actually up to you. Just pick a meaningful name, but don't use any special characters like :-/ etc as DBUS will complain. 

In theory it would be good to use the same Client Id in your MQTT initialisation but because of the DBUS special character limitations I didn't enforce that.

freakent, 2022-04-30:
An example for Grid is a good idea. The Grid support was contributed by @stundenblume, perhaps he can provide an example too.

freakent, 2022-04-30:
The correct MQTT topic is “device” singular.  Thanks for finding the typo!

Hofyyy, 2022-04-30:
One more typo.
In the "Install and Setup" section you use the "Install Version" that makes trouble after each release.

Current examples:

- wget -O dbus-mqtt-devices.zip https://github.com/freakent/dbus-mqtt-devices/archive/refs/tags/**v0.4.1.zip**
- ./dbus-mqtt-devices-**0.3.1**/bin/setup.sh
- ln -s /data/drivers/dbus-mqtt-devices-**0.2.1/**bin/service /service/dbus-mqtt-devices

freakent, 2022-04-30:
Thanks for finding these. I clearly need to find a way to update all of these when I update the version number. I'll put some thought to it.

stundenblume, 2022-04-30:
> An example for Grid is a good idea. The Grid support was contributed by @stundenblume, perhaps he can provide an example too.

Hi @Hofyyy,
you should connect to the victronOS via MQTT-Explorer, so you can observe the devices created with this driver. After that, you can write to the corresponding topics with the software of your choice. A grid device looks like this:
[image]
Some notes:
1. the victronOS sends for 60sec MQTT data after writing something to `R/<portalID>/system/0/Serial`
2. writing the L1 power to a grid meter is done with topic `W/<portalID>/grid/<assignedInstance>/Ac/L1/Power` and `W/<portalID>/grid/<assignedInstance>/Ac/Power` Payload: `{"value": 100}` Notice the leading **W**!

Is that clear?
Are you the one who would like to use the Tasmota device? Than I can provide you my rules.

Hofyyy, 2022-05-01:
Thank you for the Detailed Information. 
I use Powerfox for Grid
and two tasmota behind the pv inverter. 

Bit I can not test AT the Moment because I got install errors. See second github issue

stundenblume, 2022-05-01:
OK, a friend of mine will read his smartmeter with Tasmota and will use this driver for venusOS.
I use this driver for my inverters.
For me the install for 4.0.0.1 on venusOS works without issues.

Tasmota rules:
1. get your instance number `ON System#Boot DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON`
2. this is the example for my inverter, on grid you will have all three phases `ON System#Boot DO Backlog mem1 <VRM portalID>; mem2 <assignedInstance>; mem3 <phase[1,2,3]>; mem4 <maxKwp>;PowerDelta 101; TelePeriod 0 ENDON`

minimal rule set:
```
Rule1 
   ON System#Boot DO Backlog mem1 <VRM portalID>; mem2 <assignedInstance>; mem3 <phase[1,2,3]> ENDON
   
   ON System#Boot DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
   
   ON Energy#Power[2] DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
```

full set an a Sonoff DUAL R3:
```
Rule1 
 ON System#Boot DO Backlog mem1 <VRM portalID>; mem2 <assignedInstance>; mem3 <phase[1,2,3]>; mem4 <maxKwp>;PowerDelta 101; TelePeriod 0 ENDON
 
 ON System#Boot DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 
 ON Energy#Power[2] DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
  
 ON Energy#Current[2] DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/Current {"value": %value%} ENDON
 ON Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON
 ON Energy#Total DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Energy/Forward {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Energy/Forward {"value": %value%} ENDON
```

Hofyyy, 2022-05-01:
Maybe total stupid question. Which Software handels the rulessets you described?

stundenblume, 2022-05-01:
[Tasmota](https://tasmota.github.io/docs/Rules/) itself

Hofyyy, 2022-05-01:
Ahh thats great. Then I could have a cool Setup. For Grid I will use an EM24 original Meter. but for the two inverter I have tasmota. So I should be able to get that working

stundenblume, 2022-05-01:
That is exactly my setup. EM24 for grid and three Tasmota devices as pvinverters.
The friend refused to by the expensive EM24. There we will use a Tasmota device witch [reads](https://tasmota.github.io/docs/Smart-Meter-Interface/) his SML Smartmeter and writes via the rules into the venusOS.
By the way, we ( 6 persons with growing PV installations) are located near Oldenburg.

Hofyyy, 2022-05-01:
Cool. #Hannover. Thats Sounds great. I hope I will be able to install it Direct on my gxcc. because I will add one 800W multiplus 24V With 2x60kg 210ah AGM Truck Batterie. Thanks for the Support.

stundenblume, 2022-05-01:
Some of  us are using Pylontech Batterys and others pure Lifepo Cells.
Are you using new AGM Batteries or used ones?

Here is a ruleset for a pvinverter where no further action **should** be required.
Right now it doesn't work because Tasmota reads the "VRM portalID" in upper case and you have to **input it yourself**.
```
Rule1 
 ON System#Init DO Backlog mem3 1; mem4 0.25; PowerDelta 101; TelePeriod 0 ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 ON Event#DBusInfo#portalId DO mem1 <VRM portalID> ENDON
 ON Event#DBusInfo#deviceInstance#wr DO mem2 %value% ENDON

 ON System#Boot DO Publish W/%mem1%/pvinverter/%mem2%/Ac/MaxPower {"value": %mem4%} ENDON
 
 ON Energy#Power[2] DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
  
 ON Energy#Current[2] DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON
 ON Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON
 
 ON Energy#Total DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Energy/Forward {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Energy/Forward {"value": %value%} ENDON
```

Hofyyy, 2022-05-01:
THX. New Batteries Direct from the Research and Produktion line. My Familie works in a big Company In Hannover. So I get some test Batteries.

Hofyyy, 2022-05-01:
> PowerDelta

Hello @stundenblume 
I try to understand every rule now.
Could you explain me what the PoweDelta does and why it is 101?

If I understand this correct, it defines a threshold until when further things are reportet.
Because my panels are 0W to 600W on one tasmota I would set the threshhold to 5W!?!

Best regards

stundenblume, 2022-05-01:
[PowerDelta](https://tasmota.github.io/docs/Commands/#power-monitoring): 1 - 100 means 1 to 100% change of the power. From 101 it means power change in W with a offset of 100W.
1 = 1%
10 = 10%
100 = 100%
101 = 1W
110 = 10W
2100 = 2000W

I don't know if it's really mandatory to set that parameter.

`TelePeriod 0` is to not overload the Victron MQTT Broker.

After [talking](https://discord.com/channels/479389167382691863/790182324939128832/970296838848663584) to some people who know Tasmota, it's unlikely that it will be possible to read the VRM portalID with rules. Rules will always parse and change to upper case.

Here the latest version. Just fill mem1, mem3 and mem4:
```
Rule1 
 ON System#Init DO Backlog mem1 <VRM deviceID>; mem3 <phase[1,2,3]>; mem4 <kWpMax>; PowerDelta 101; TelePeriod 0 ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 ON Event#DBusInfo#deviceInstance#wr DO mem2 %value% ENDON
 
 ON System#Boot DO Publish W/%mem1%/pvinverter/%mem2%/Ac/MaxPower {"value": %mem4%} ENDON
 
 ON Energy#Power[2] DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
 
 ON Energy#Current[2] DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON
 ON Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON
 
 ON Energy#Total DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Energy/Forward {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Energy/Forward {"value": %value%} ENDON
```

Hofyyy, 2022-05-01:
Thanks!

Id as String? "ABC" or ABC
Phases=1
KWP=0.6 or 600W?

stundenblume, 2022-05-01:
`ON System#Init DO Backlog mem1 b7eb8b3b; mem3 1; mem4 0.6; PowerDelta 101; TelePeriod 0 ENDON`

stundenblume, 2022-05-01:
On which device did you flash Tasmota? If you just have one channel for power your rule looks a little different.
This would be a minimal version:
```
Rule1 
 ON System#Init DO Backlog mem1 <VRM deviceID>; mem3 <phase[1,2,3]> ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 ON Event#DBusInfo#deviceInstance#wr DO mem2 %value% ENDON
 
 ON Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
```
current, voltage, energy counter, maxPower are all just optinal

Hofyyy, 2022-05-02:
No. All is fine

Gosund SP1 v23

Hofyyy, 2022-05-02:
Bähm! Its working perfekt on the test mosquito server now. I can test it finally, wen the driver does work on my real CCXG. Then I will switch the MQTT Server

stundenblume, 2022-05-02:
Perfect!

freakent, 2022-05-02:
Great collaboration here, guys 👏👏. I wonder if we should have a wiki page on this project where you can share all this knowledge? @stundenblume, If I set it up would you add your Grid tips?

Hofyyy, 2022-05-03:
Just for Info: This tasmota helper tool is amasing: https://github.com/jziolkowski/tdm/

stundenblume, 2022-05-03:
> Great collaboration here, guys clapclap. I wonder if we should have a wiki page on this project where you can share all this knowledge? @stundenblume, If I set it up would you add your Grid tips?

Yes, I would fill it with information for "pvinverter" and later "grid" when I had a change to gain some experience. I already have a small documentation for myself.
I also would like to write something on the victron forum to gain some attention to your extremely useful project.

freakent, 2022-05-04:
@stundenblume, do you see an edit button here?
https://github.com/freakent/dbus-mqtt-devices/wiki/PV-Inverter

stundenblume, 2022-05-04:
> @stundenblume, do you see an edit button here? https://github.com/freakent/dbus-mqtt-devices/wiki/PV-Inverter

Yes, thanks!

closed 2022-05-04 by stundenblume

Hofyyy, 2022-05-04:
@stundenblume 
Now I was able to do a real test.
Registration is running.

But the Values are not on the right topic.
Is used the ClientID instead of the deviceinstance in the MQTT Path .. any idea?
So MEM2 seems to be wrong ...

Rule1 
 ON System#Init DO Backlog mem1 f833311fbeda; mem3 1 ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 ON Event#DBusInfo#deviceInstance#wr DO mem2 %value% ENDON
 
 ON Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON

Hofyyy, 2022-05-05:
@stundenblume 
Hello,
short summary. All is working very well now.

Two small open points:
1.
I need to set "mem2 manually" -> mem2 1 and mem2 2 then all is working fine.
If I understand it correct: "DO mem2 %value% ENDON" is not working.

2.
I get mqtt updates in my Tasmota every 200ms.
Can I change the update intervall of the Rule?

But it works already amazing!
[image]

stundenblume, 2022-05-05:
> @stundenblume Hello, short summary. All is working very well now.
> 
> Two small open points: 1. I need to set "mem2 manually" -> mem2 1 and mem2 2 then all is working fine. If I understand it correct: "DO mem2 %value% ENDON" is not working.
> 
> 
> I get mqtt updates in my Tasmota every 200ms. Can I change the update intervall of the Rule?
> 
> But it works already amazing! [image]

1. In Tasmota on ESP8266 devices "subscribe" [is not included in the precompiled binaries](https://tasmota.github.io/docs/Subscribe-%26-Unsubscribe/). So ` Event#DBusInfo#deviceInstance#wr` will never be triggered an mem2 won't be filled with the "instanceID".
Solutions:
   1. Set mem2 manually on the first line in Rule1 after you know it. `ON System#Init DO Backlog mem1 <VRMdeviceID>; mem2 <deviceInstance>; mem3 <phase[1,2,3]>; mem4 <kWpMax>; PowerDelta 101; TelePeriod 0; VoltRes 1; SetOption129 1 ENDON`
   2. compile a binary with
    ```
    #ifndef SUPPORT_MQTT_EVENT
    #define SUPPORT_MQTT_EVENT
    #endif
    ```
   3. use [ESP32 hardware](https://templates.blakadder.com/esp32.html)

2. 1. You can use Time#Minute as [trigger](https://tasmota.github.io/docs/Rules/#rule-trigger)
    2. You can prefix the sensor witch `Tele-` then it will be send with every TelePeriod

This example should send every 11 seconds:
```
Rule1 
 ON System#Init DO Backlog mem1 <VRMdeviceID>; mem2 <deviceInstance>; mem3 <phase[1,2,3]>; mem4 <kWpMax>; TelePeriod 11 ENDON
 
 ON mqtt#connected DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 
 ON Tele-Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
 ```
 You can change TelePeriod by typing `teleperiod 11` into the web console.

 I ordered your hardware, than I can test it myself.

Hofyyy, 2022-05-05:
Thank you for the perfect answer. I will change mem2 to hardcoded value. Thats totaly fine for me.

And I will try to change the intervall to 2 seconds. I think that should be more then enough.

stundenblume, 2022-05-05:
[TelePeriod](https://tasmota.github.io/docs/Commands/#mqtt):
```See current value and force publish STATE and SENSOR message0 = disable telemetry messages1 = reset telemetry period to firmware default (TELE_PERIOD)10..3600 = set telemetry period in seconds (default = 300)```

stundenblume, 2022-05-05:
With RuleTimers it would be possible to get intervals < 10 sec.
Will send later an example.

stundenblume, 2022-05-05:
This should work (untested). Just set `mem5 2`:
```
Rule1 
 ON System#Init DO Backlog mem1 <VRMdeviceID>; mem2 <deviceInstance>; mem3 <phase[1,2,3]>; mem5 <intervall> TelePeriod 0NDON
 
 ON mqtt#connected DO Backlog Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}}; RuleTimer1 %mem5% ENDON

 ON Rules#Timer=1 DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %var1%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %var1%}; RuleTimer1 %mem5% ENDON
 
 ON Energy#Power DO var1 %value% ENDON
```

Hofyyy, 2022-05-05:
Darf ich fragen was die letzte Zeile tut?

stundenblume, 2022-05-05:
The last line updates `var1` on every new "Power" reading. As far as I know, there is no shorter way.

Hofyyy, 2022-05-05:
If I see it correct, I only have mem1, mem2 and mem3. Further more 10 sec. are fine.

I will test the following rules tomorrow

PV1

Rule1 
 ON System#Init DO Backlog mem1 <ID>; mem2 1; mem3 1; PowerDelta 101; TelePeriod 10 ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 
 ON Tele-Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
ON Tele-Energy#Current DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON
 ON Tele-Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON

PV2

Rule1 
 ON System#Init DO Backlog mem1 <ID>; mem2 2; mem3 1; PowerDelta 101; TelePeriod 10 ENDON
 
 ON mqtt#connected DO Backlog Subscribe DBusInfo, device/%deviceid%/DBus; Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 
 ON Tele-Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
ON Tele-Energy#Current DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON
 ON Tele-Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON

stundenblume, 2022-05-06:
> Gosund SP1 v23

@Hofyyy
My Gosund devices SP1-C_V2.4 arrived today. Was it possible to use [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert) or did you flash it via serial adapter?

Hofyyy, 2022-05-06:
I di it With some wlan hack without open the Device.

stundenblume, 2022-05-07:
OK. The first one I flashed via serial adapter. The next one I will try Tuya-Convert. Did you use that software or something else?

The SP1 is a very nice and well designed device. It could be a bit smaller but the price is also very good and suited for 16A is always on the safe site.

Hofyyy, 2022-05-07:
Hey, I flashed the software some years ago. so i do not know what i used. sorry. But Tuya convert sounds good. yea 16A was also a reason from my side.

Hofyyy, 2022-05-07:
@stundenblume
I had the problem that after a "power shutdown", that the devices were not registeres anymore in the Victron.
A second manual reboot of the SP1s solved the issue.

It seems to be that the SP1s booted first, after power was on again (ok they shoult boot fast).
Could you test to power down the Victron Main System or reboot it?

Maybe its also a bug in the driver, because the driver may start after the MQTT broker on the system.
But the SPs1 do register right after MQTT broker is online?!? So we might need a registration delay of some seconds in the rule ...

I would like to get it 100% robust. And I do not want to have a fixed boot order.

I will do some futher testing.

stundenblume, 2022-05-07:
See my answer [here](https://github.com/freakent/dbus-mqtt-devices/issues/17#issuecomment-1120306646).

stundenblume, 2022-05-15:
@Hofyyy
today I noticed that "powerDelta" overrides "telePeriod".
That means if you set `PowerDelta 110` you will get updates of the power reading if it changed by 10W, even faster than 10sec.

stundenblume, 2022-05-16:
> Rule1 ON System#Init DO Backlog mem1 "dsffddffs"; mem2 1; mem3 1; PowerDelta 101; TelePeriod 10 ENDON
> 
> ON mqtt#connected DO RuleTimer1 10 ENDON ON Rules#Timer=1 DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
> 
> ON Tele-Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON ON Tele-Energy#Current DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON ON Tele-Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON

@Hofyyy 
You also should publish your "Total Energy", I think otherwise VRM won't be able the show/graph you pv data.
[image]

Hofyyy, 2022-05-17:
@stundenblume 
Thanks for the info. I am not sure if that was necessary. 

Thats at the moment my final rule.

Rule1 
 ON System#Init DO Backlog mem1 "VICTRONID"; mem2 1; mem3 1; PowerDelta 0; TelePeriod 10 ENDON

 ON mqtt#connected DO RuleTimer1 30 ENDON
 ON Rules#Timer=1 DO Publish device/%deviceid%/Status {"clientId":"%deviceid%","connected":1,"version":"v1.0 ALPHA","services":{"wr":"pvinverter"}} ENDON
 
 ON Tele-Energy#Power DO Backlog Publish W/%mem1%/pvinverter/%mem2%/Ac/Power {"value": %value%}; Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Power {"value": %value%} ENDON
 ON Tele-Energy#Current DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Current {"value": %value%} ENDON
 ON Tele-Energy#Voltage DO Publish W/%mem1%/pvinverter/%mem2%/Ac/L%mem3%/Voltage {"value": %value%} ENDON
 ON Tele-Energy#Total DO Publish W/%mem1%/pvinverter/%mem2%/Ac/Energy/Forward {"value": %value%} ENDON

Hofyyy, 2022-05-17:
Step by step its running :-)

[image]

stundenblume, 2022-05-17:
Looks good! :)

`Tele-Energy#Total` and `Tele-Energy#Current` can be written to two topics `.../Ac/xxx` and `.../Ac/L1/xxx` I don't know if it's necessary but just in case it doesn't work as expected.  
 
 ```
   ON Tele-Energy#Total DO Backlog Publish W/%mem1%/%mem5%/%mem2%/Ac/Energy/Forward {"value": %value%}; Publish W/%mem1%/%mem5%/%mem2%/Ac/L%mem3%/Energy/Forward {"value": %value%} ENDON
   ON Tele-Energy#Current DO Backlog Publish W/%mem1%/%mem5%/%mem2%/Ac/L%mem3%/Current {"value": %value%};  Publish W/%mem1%/%mem5%/%mem2%/Ac/Current {"value": %value%} ENDON
```

Hofyyy, 2022-05-17:
I checked it. And I found that it should not be needed on L1. Also in the GGCX its looks good this way.

stundenblume, 2022-05-17:
OK, nice to know. `Power` has to be written to both topics otherwise it hadn't worked for me.