freakent/dbus-mqtt-devices #12: Error during install script

Hofyyy, 2022-04-30:
Hello,

I got the following error:

Collecting PyYAML==6.0
  Downloading PyYAML-6.0.tar.gz (124 kB)
     |████████████████████████████████| 124 kB 905 kB/s 
  Installing build dependencies ... error
  ERROR: Command errored out with exit status -9:
   command: /usr/bin/python /usr/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /var/volatile/tmp/pip-build-env-6l75klcn/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel Cython
       cwd: None
  Complete output (7 lines):
  Collecting setuptools
    Downloading setuptools-62.1.0-py3-none-any.whl (1.1 MB)
  Collecting wheel
    Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting Cython
    Downloading Cython-0.29.28-py2.py3-none-any.whl (983 kB)
  Installing collected packages: setuptools, wheel, Cython
  ----------------------------------------
ERROR: Command errored out with exit status -9: /usr/bin/python /usr/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /var/volatile/tmp/pip-build-env-6l75klcn/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel Cython Check the logs for full command output.
Set up Victron module libraries
Set up device service to autorun on restart
grep: /data/rc.local: No such file or directory
Setup dbus-mqtt-devices complete
root@ccgx:/data/drivers# 
Session was closed

Hofyyy, 2022-04-30:
Device: Color Control GX
Firmware: v.2.85
Clean Image

freakent, 2022-05-01:
Both @stundenblume and myself are running this on Raspberry Pi's. Last night I tried to install on a CCGX device. The installation caused the device to reset and will no longer boot. I recommend you do not try to install this on CCGX until I find out why.

Hofyyy, 2022-05-01:
Thank you for the information!

freakent, 2022-05-01:
I managed to get the driver to install on CCGX but the MQTT broker is incredibly unstable. The connection from the driver to MQTT is continuously disconnecting and reconnecting. It maybe that CCGX just isn't up to the job. Which is a shame because my whole reason for writing this was to deploy on CCGX. I'll keep investigating.

Hofyyy, 2022-05-01:
Thats sounds great and bad :-)

gharshaw, 2022-05-01:
Bummer.  I was just about to try this on my CCGX v.2.85. I'm reading bluetooth temp sensors and bluetooth SeeLevel Tank sensors via Node-Red PI into the CCGX with MQTT. I've been using multiple "dummy d-bus services" to add these devices. Hoping you are able to find a solution as I like the idea of  the self registration.
Thanks for your work on this.

Hofyyy, 2022-05-02:
My tasmota Device is Running perfectly. So wie need a solution here 😜

freakent, 2022-05-02:
Hey, I've been experimenting! 😀 If anyone else has any ideas I'm listening.

freakent, 2022-05-02:
In release 0.41 I am using pip to install PYyaml. On my CCGX as soon as I tried to install PYyaml using pip, the device reset and would no longer boot. I guess it was something to do with compiling PYyaml to native code. I looked around for a pure python alternative only to end up back at PYyaml. It turns out you can install a pure python version of PYyaml. I did this on my CCGX and it installed fine. I then ran dbus-Mqtt-devices directly from the command line and found it was continuously connecting and disconnecting with a protocol error (rc = 1). I have been comparing how dbus-mqtt (from Victron) and dbus-mqtt-devices are run. So far I haven't found the magic switch that stabilises things.

gharshaw, 2022-05-02:
I found this comment in an MQTT troubleshooting blog about rc=1 -- Maybe it will shed some light :-) 
"this sometimes happens, when one program is running twice an connects from the same IP with the same client ID. The broker than closes the "old connection" which is still held by the process running in the background."

gharshaw, 2022-05-02:
I see that dbus-mqtt (from Victron) has a MAX_TOPIC_AGE = 60s keep-alive arg in  dbus-mqtt. I don't see the arg in your script. 

parser.add_argument('-k', '--keep-alive', default=MAX_TOPIC_AGE, help='keep alive interval in seconds', type=int)

I don't know if this is the issue - just throwing out thoughts that might shed more light.

freakent, 2022-05-03:
Thanks, very useful. Two things I can investigate. If you find any more let me know.

stundenblume, 2022-05-03:
> Hey, I've been experimenting! grinning If anyone else has any ideas I'm listening.

Hi Martin,
sorry I don't own a CCGX nor I'm a experienced programmer. Sorry!
Sebastian

gharshaw, 2022-05-04:
Martin, I'd be curious to know if Markus is having the same mqtt stability problem on the Cerbo GX

gharshaw, 2022-05-04:
Martin, 
BTW, I'm willing to help you with any testing on my CCGX

Gene

Hofyyy, 2022-05-04:
@freakent 
I can do testing too. Because my "Color Control GX" is not used until now and I have no problem with flashing stock if needed. My real installation will start next weekend.

I did now:
- Add all folders described
- Copied script
- Checked right (they were fine) I did not need chmod
- Run script without problems
- did "reboot"

System is online again with remote console and SSH

What can I test?
Or do you need VPN access?

I will now change my tasmota to the real MQTT Broker :-) and play a little bit.

Hofyyy, 2022-05-04:
First results:
- I seems to be that the service is not running or does not exist
Do I need to unpack a different packet first?

See:
root@ccgx:/data/drivers/dbus-mqtt-device-0.4.0.3/bin# ls
setup-ccgx.sh
root@ccgx:/data/drivers/dbus-mqtt-device-0.4.0.3/ext# ls
__pycache__             dbus_mqtt.py            ext                     mqtt_gobject_bridge.py  velib_python

gharshaw, 2022-05-04:
> Hi Gene,
> 
> I think the thing to do here is to get the installation working then we can see if the stability issue is just on my device. I do you want to remind you that my CCGX completely crashed and I had to re-flash a standard firmware by hand to get it working again. I think we can avoid that happening again but I strongly suggest that you have a recovery SD card standing by and that you make a note of any settings you don't want to loose.
> 
> Here are Victron's instructions for manual upgrade, just in case. https://www.victronenergy.com/live/ccgx:firmware_upgrade_to_v2 These instructions will revert your ccgx to an older version, but you then perform an online upgrade to bring it back up to the latest version.
> 
> Are you happy with this?

Yes - I recently had to restore back to 2.18 with Sdcard due to ethernet interface remaining "unplugged"

freakent, 2022-05-04:
Take a look at the Troubleshooting instructions. 

https://github.com/freakent/dbus-mqtt-devices#troubleshooting

I would like to see the contents of your log file /var/log/dbus-mqtt-devices/current

If it is constantly disconnecting and re connecting it should be very evident from the log.

freakent, 2022-05-04:
@Hofyyy I think I see an extra ext directory inside your main ext directory. That would suggest you ran the set up script from the wrong folder. Try following the steps in the install instructions in the readme, except using the setup-ccgx.sh.

Hofyyy, 2022-05-04:
@freakent 
All is running fine now. As I can see.
Also my inverters are registered. I troubleshooting now, why no WATT are displayed. Seems to be that the device instance is not saved correctly in my rule.

gharshaw, 2022-05-04:
@freakent 
Marty,  The driver is so far running  very stable. I have not seen it restart at all in the logs.  I have 5 temp sensors and 4 tank sensors being registered and updated. 
I do see one issue after I reboot the CCGX.  I need to re-register my <device/<client id>/Status> payload and when I do, the local settings like "custom name" get overwritten  to the default of "My Tank Sensor" or "My Temperature Sensor. 
Is this expected? 
Do you perhaps update these custom names dynamically by publishing a mqtt message to the custom name value?

Awesome work!!

Gene

freakent, 2022-05-04:
Well it's good to know it's running stable. 

You need to register with the status message every time you connect to Mqtt. Especially if you reboot the GX device. That is a requirement of Victron's DBUS implementation

gharshaw, 2022-05-04:
Here is log file after rebooting and running the re-register. I am getting the same device ID's unfortunately local settings are getting updated to what appears to be the values in the  services.yml file. 
[log.txt](https://github.com/freakent/dbus-mqtt-devices/files/8627263/log.txt)

freakent, 2022-05-04:
@gharshaw Gene, I just noticed you are running 0.4.0.1, can you update to the latest 0.4.0.3 ?

gharshaw, 2022-05-04:
@freakent 

> @gharshaw Gene, I just noticed you are running 0.4.0.1, can you update to the latest 0.4.0.3 ?

I think I'm running dbus_mqtt_devices, v0.4.1 -  the latest release. 
starting dbus-mqtt-devices ***
@400000006272e42b0fe3d154 -------- dbus_mqtt_devices, **v0.4.1** is starting up **--------******

freakent, 2022-05-05:
Gene, Sorry, please ignore that suggestion.

freakent, 2022-05-05:
It sounds like the ccgx installation solution is working. So I will close this issue for now. I am preparing a new 0.5 release with the setup-ccgx.sh script.

closed 2022-05-05 by freakent