freakent/dbus-mqtt-devices #13: Feature Request: List, Add, Remove devices

Hofyyy, 2022-05-01:
Hello,

to spread this tool as robust as possible into the world I would like to use it the following way.
I would like to:
- Install the tool via your installer (ok)
- I would like to add 1 grid and 2 pv inverter, without knowledge (feature request)
--> go via SSH on the system
--> call: ./dbus-mqtt-devices add grid NAME ATTR1 ATTR2 ...
--> call: ./dbus-mqtt-devices add pvinverter NAME_1 ATTR1 ATTR2 ...
--> call: ./dbus-mqtt-devices add pvinverter NAME_2 ATTR1 ATTR2 ...

In this case the mqtt magic behind add and remove would be hardcoded per device via python.
In my ideal world no "normal" user would need to understand this.

After the devices are created they should be visible in victron (first magic moment).
--> Then i can read some docu per device type and browse with MQTT Explorer on real existing topics per device.

Further more, for cleanup and testing a "list" and "remove" would be great
--> call: ./dbus-mqtt-devices list all
---> Would output a list of all added devices with the important information

--> call: ./dbus-mqtt-devices remove all
--> call: ./dbus-mqtt-devices remove MAGIC_ID //would remove (connect=0) this specific device.

From my knowlegde as programmer I would hope, that this is not a big task to code, but would massive push all users and reduce documentation needs for so much detailes, because the complex stuff would be capsuled.

Best regards
Sebastian

freakent, 2022-05-01:
> I would like to add 1 grid and 2 pv inverter, without knowledge (feature request)

You are missing a vital component in your solution.  You need some device to actually send the readings to the GX device via MQTT. The registering protocol I have designed MUST be followed every time that device connects to the Victron MQTT. Victron's DBUS handles the saving of any attributes that need to be preserved. The registration protocol is not something you do one time only. Creating a command line tool would serve no benefit.

freakent, 2022-05-01:
> In my ideal world no "normal" user would need to understand this.

They don't need to, if you follow the registration protocol on the device sending the data.

freakent, 2022-05-01:
> After the devices are created they should be visible in victron (first magic moment).
Victron's design calls for devices to only be visible when they are connected to the dbus.

Hofyyy, 2022-05-01:
Ok thank you for the Information. Then I Do not understand connect=0 but.  Then I will play some weeks With your solutiob.

closed 2022-05-01 by Hofyyy

freakent, 2022-05-01:
Sorry for breaking up my reply in multiple posts, I am on my iPhone and it's easier to respond to specific points that way.