freakent/dbus-mqtt-devices #14: GPS Driver

trinler007, 2022-05-04:
Hi,

on my camper I've a router with an external GPS antenna. This device could easily send the position data to my Cerbo GX via MQTT.

I’m already using dbus-mqtt-devices for tank and temperature devices. Would it be possible to add a GPS driver?

Thanks!

Markus

freakent, 2022-05-04:
Hi Markus,

Great to hear you have Temperature and Tank devices working? What device do you use as a client?

I also have a GPS enabled router, a Teltonika RUTX-11. I have looked into this but so far it seems Victron do not expose GPS interface through DBUS-MQTT. It seems the simplest solution for GPS is actually to use SOCAT. 
https://community.victronenergy.com/questions/17470/venus-gx-to-pull-gps-coordinates-from-onboard-rout.html

trinler007, 2022-05-04:
Hi Martin,

I use various Homematic and Zigbee sensors for the temperature, and get the filling level of the tank via ModbusTCP from a barometric sensor. I process this information with node-red and send it to dbus-mqtt-devices.

For GPS (and LTE) I'm using a Mikrotik LtAP (https://mikrotik.com/product/ltap_lte6_kit). These devices are able to use MQTT in scripts. 

Since I'm already processing the position whit a script and sending it to a tracking server, it would have been very simple to send the position to the Victron device with adding just a few lines of additional code.

If I connect a USB GPS device, I see the determined data in the MQTT. Does this help in any way?

freakent, 2022-05-04:
VRM just updated, and guess what, it works!
[image]

Even I'm surprised!

trinler007, 2022-05-04:
Awesome! That was extremely fast! Thanks!

freakent, 2022-05-04:
If I share my updates services.yml file with you, would you be able to replace your existing services.yml with this patched version?

gharshaw, 2022-05-04:
Martin, Not to hijack this GPS thread but I'd be curious to know if Markus is having the same mqtt stability problem on the Cerbo GX

Gene

trinler007, 2022-05-04:
> If I share my updates services.yml file with you, would you be able to replace your existing services.yml with this patched version?

Yes, of course. I'm very happy to test that.

trinler007, 2022-05-04:
Hi Gene, so far I haven't found any issues with MQTT on my Cerbo.

freakent, 2022-05-04:
Find a zip file attached. I also included my simulator script so you can see which mqtt topics to use.
[gps-dbus-mqtt-devices.zip](https://github.com/freakent/dbus-mqtt-devices/files/8621495/gps-dbus-mqtt-devices.zip)

freakent, 2022-05-04:
To see the GPS device in the Venus GX Console, from the main menu choose settings > GPS. You have to scroll down to find both options.

freakent, 2022-05-04:
OK, I'll plan to make another release with the GPS support added. In the mean time if you encounter any problems with it please raise another issue so I can decide whether it is safe for making generally available.