freakent/dbus-mqtt-devices #16: Does set up need to be re-run after a CCGX firmware update

gharshaw, 2022-05-06:
@freakent 
Hi Martin,
Looks like Victron has released 2.90~3 as a release candidate. Will I need to rerun setup-ccgx.sh if I do a firmware update?

Regards,
Gene

freakent, 2022-05-06:
No.

freakent, 2022-05-06:
At least, I don't think so, until I test it 😀

freakent, 2022-05-06:
The real concern is whether anything has changed in 2.9 to affect compatibility with this driver. We should leave this issue open to track experience with upgrades to 2.9.

gharshaw, 2022-05-10:
@freakent 
Martin,
After updating CCGX to new Firmware 2.90 ~6, and  as I kinda anticipated, PIP and yaml are not included in the new firmware images.
@40000000627a715f16674484   File "/data/drivers/dbus-mqtt-devices-0.4.1/device_manager.py", line 17, in <module>
@40000000627a715f16674484     import yaml
@40000000627a715f16674484 ModuleNotFoundError: No module named 'yaml'

I updated to dbus-mqtt-devices-0.5.1 and re-ran setup-ccgx.sh to reinstall pip and yaml successfully. 

I then did  a bit of cleanup :
 1. deleted "ln -s /data/drivers/dbus-mqtt-devices-0.4.1/bin/service /service/dbus-mqtt-devices" from the rc.local file 
 2. deleted the  /data/drivers/dbus-mqtt-devices-0.4.1 folder 

Successful install on 2.90 ~6

Regards,
Gene

closed 2022-12-14 by freakent