freakent/dbus-mqtt-devices #20: ERROR:mqtt_gobject_bridge:[Disconnected] Lost connection to broker

BoudewijnWeenink, 2022-09-08:
When running : more /var/log/dbus-mqtt-devices/current
I get this error message: ERROR:mqtt_gobject_bridge:[Disconnected] Lost connection to broker
I think this prevents me from adding sensors and values
Any suggestions on how to solve?

I''m running a R.Pi 4 (1.15) so can't run an official version download from Victron...

Thanks for clear description :)

Thanks,

Boudewijn

freakent, 2022-09-08:
Can you upload a copy of your log file ?

Regards,
Martin

+44 7776490601

freakent, 2022-09-08:
> I''m running a R.Pi 4 (1.15) so can't run an official version download from Victron...

What exactly do you mean by this? How did you install VenusOS and what version are you running?

BoudewijnWeenink, 2022-09-09:
Thanks Freakent for your prompt reply.

> Can you upload a copy of your log file ?

Sure :)
@400000006319d39f133d50dc *** starting dbus-mqtt-devices ***
@400000006319d3a010835544 -------- dbus_mqtt_devices, v0.5.1 is starting up --------
@400000006319d3a010836cb4 INFO:logger:Loglevel set to INFO
@400000006319d3a0112d8f64 INFO:device_manager:Using portalId e45f01bce286
@400000006319d3a02240ca8c INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@400000006319d3a0236009b4 INFO:device_manager:[Connected] Result code 0
@400000006319d4a71b6f663c *** CCGX booted (0) ***
@400000006319d4ae08b2016c *** starting dbus-mqtt-devices ***
@400000006319d4b5057b965c -------- dbus_mqtt_devices, v0.5.1 is starting up --------
@400000006319d4b5057ba9e4 INFO:logger:Loglevel set to INFO
@400000006319d4b50c0f9bbc INFO:device_manager:Using portalId e45f01bce286
@400000006319d4b52fd29b6c INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@400000006319d4b532cd841c INFO:device_manager:[Connected] Result code 0
**@400000006319e7882e527794 ERROR:mqtt_gobject_bridge:[Disconnected] Lost connection to broker**
@400000006319e7882e5bed74 INFO:mqtt_gobject_bridge:[Disconnected] Set timer
@400000006319e78d2e76a164 INFO:mqtt_gobject_bridge:[Reconnect] start
@400000006319e78d2eadf18c INFO:mqtt_gobject_bridge:[Reconnect] success
@400000006319e78d2ec02d84 INFO:device_manager:[Connected] Result code 0
@40000000631a21fb18c174ac *** CCGX booted (0) ***
@40000000631a221031b6fb6c *** starting dbus-mqtt-devices ***
@40000000631a221736139314 -------- dbus_mqtt_devices, v0.5.1 is starting up --------
@40000000631a22173613a69c INFO:logger:Loglevel set to INFO
@40000000631a22173783b92c INFO:device_manager:Using portalId e45f01bce286
@40000000631a221932929dfc INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000631a22193526fcac INFO:device_manager:[Connected] Result code 0
@40000000631a28511ad25734 *** CCGX booted (0) ***
@40000000631a2857298904bc *** starting dbus-mqtt-devices ***
@40000000631a285f24af7cb4 -------- dbus_mqtt_devices, v0.5.1 is starting up --------
@40000000631a285f24af903c INFO:logger:Loglevel set to INFO
@40000000631a285f2991f1e4 INFO:device_manager:Using portalId e45f01bce286
@40000000631a28610662dd64 INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000631a28610c27c354 INFO:device_manager:[Connected] Result code 0
@40000000631a2d44237118e4 *** CCGX booted (0) ***
@40000000631a2d4a32da93dc *** starting dbus-mqtt-devices ***
@40000000631a2d5329ca631c -------- dbus_mqtt_devices, v0.5.1 is starting up --------
@40000000631a2d5329ca76a4 INFO:logger:Loglevel set to INFO
@40000000631a2d533009fb34 INFO:device_manager:Using portalId e45f01bce286
@40000000631a2d542533569c INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@40000000631a2d5427c4909c INFO:device_manager:[Connected] Result code 0

> > I''m running a R.Pi 4 (1.15) so can't run an official version download from Victron...
> 
> What exactly do you mean by this? How did you install VenusOS and what version are you running?

I've tried a lot of the official Victron images, but none of them work (common for newest Pi 4s unfortunat).
The only version I could get to work, I found at : 
https://atbbox.com/venus-image-raspberrypi4-20220221234425-v2.85_1.zip
(I tried to find the site where I found the download link, but couldn't find it)

Venus also shows v.2.8.5 ~1 in the firmware screen..
[image]

I've got an older Pi4 (v. 1.12) that should be compatible but so far I couldn't get to working either with an official download.. I can free it up, but then up I need to migrate a lot of home automation stuff that is running on it.... the new Pi doesn't accept the SD from the old Pi... to bad Pi3's aren't available at the moment, that may have made my life a lot easier ;)

If you expect it is the Pi 4 (v 1.15) or the image that I've used, then I'll free up my old Pi 4 (v 1.12) then I can test that....

Thanks,

Boudewijn

freakent, 2022-09-09:
How much RAM does your pi have? I had a similar problem to this running this driver on my CCGX, although it works fine in my Pi.

BoudewijnWeenink, 2022-09-09:
Thanks again :) for your quick reply.

If I check https://www.victronenergy.com/live/venus-os:start the CCGX had 256MB
My new Pi 4 has 4GB (only one available for purchase), so it should be enough..
but it might be that in the background also this version of Venus OS doesn't like this board/revision...
FYI my old Pi 4 has 2GB

BoudewijnWeenink, 2022-09-12:
I've gave my old Pi4 a fresh SD image of Venus OS (2.90 large).
After installation per your instruction, I have no more errors in the log :)

My conclusion is that the version I used on my new Pi4 that appeared to run, didn't run 100% in the background. as a consequence your bridge will also not work, but that has nothing to do with your bridge :)

Again thanks for the bridge and support.

Boudewijn

closed 2022-09-12 by BoudewijnWeenink