freakent/dbus-mqtt-devices #22: pvinverter not in dashboard

stromibaer, 2022-10-17:
Hi,
have installed dbus-mqtt-devices latest version on Cerbo v2.91. 
Pvinverters will be registered and are listed as devices in VRM with deviceInstance number.
I can also publish values to them and values are showed in Cerbo remote control.

[image]

But the pvinverters are not showed on the VRM Portal Dashbord neither on Cerbo display.
In the configuration of the pvinverters i have no "Setup" entry:

[image]

[image]

PV Inverter 31 is a ET112 which is metering whole energy of the inverters, but i want to display each pvinverter standalone in the VRM Portal and Cerbo.

All pvinverters are on AC-OUT, but how can i change this? Because Setup entrypoint is not available.

Thanks,
Peter

freakent, 2022-10-17:
Can you share your code that you used to register your device on MQTT to dbus?

stromibaer, 2022-10-17:
on startup: 
   // Loop all inverters and register @Victron Venus OS
    for (uint8_t i = 0; i < Hoymiles.getNumInverters(); i++) {
        auto inv = Hoymiles.getInverterByPos(i);

        char buffer[sizeof(uint64_t) * 8 + 1];
        snprintf(buffer, sizeof(buffer), "%0x%08x",
            ((uint32_t)((inv->serial() >> 32) & 0xFFFFFFFF)),
            ((uint32_t)(inv->serial() & 0xFFFFFFFF)));
        String subtopic = String(buffer);

        // Register @Victon Venus
        String hoyserial = String(subtopic);
        mqttClient->subscribe(String("device/" + hoyserial + "/DBus").c_str(),0);

        DynamicJsonDocument serviceDoc(256);
        serviceDoc[hoyserial] = F("pvinverter");
        JsonObject serviceObj = serviceDoc.as<JsonObject>();

        DynamicJsonDocument rootDoc(1024);
        rootDoc[F("clientId")] = hoyserial;
        rootDoc[F("connected")] = 1;
        rootDoc[F("version")] = inv->DevInfo()->getHwVersion();
        rootDoc[F("services")] = serviceObj;
        JsonObject rootObj = rootDoc.as<JsonObject>();

        char data[1024];
        serializeJson(rootObj, data);

        String topic = ("device/" + hoyserial + "/Status");
        mqttClient->publish(topic.c_str(), 0, 1, data);
    }

On MqTT Message receive:

        // Victron message only on startup after subscribe
        // device/116181045449/DBus = JSON 
        char* rest = &token_topic[strlen("device")];
        serial_str = strtok_r(rest, "/", &rest);

        if (serial_str == NULL) {
            return;
        }

        serial = strtoull(serial_str, 0, 16);

        auto inv = Hoymiles.getInverterBySerial(serial);

        if (inv == nullptr) {
            Serial.print(F("Can not register inverter: "));
            Serial.println(serial);
            return;
        }

        char* strlimit = new char[len + 1];
        memcpy(strlimit, payload, len);
        strlimit[len] = '\0';
        
        DynamicJsonDocument docDbus(512);
        deserializeJson(docDbus, strlimit);
        VictronPortalId = docDbus["portalId"];
                    
        DynamicJsonDocument docInstance(256);
        docInstance = docDbus["deviceInstance"];
        String deviceInstance = docInstance[serial_str];
    
        String inverter = serial_str;
        VictronDeviceInstance.insert({inverter, deviceInstance});

        if (VictronDeviceInstance.find(inverter)!=VictronDeviceInstance.end()) {
            String valfound = VictronDeviceInstance[inverter];
            Serial.print(F(" to Venus OS with portalId: "));
            Serial.print(VictronPortalId);
            Serial.print(F(" and deviceInstance: "));
            Serial.println(valfound);
        }
        return;

And publishing values with:

        // Victron message only on startup after subscribe
        // device/116181045449/DBus = JSON 
        char* rest = &token_topic[strlen("device")];
        serial_str = strtok_r(rest, "/", &rest);

        if (serial_str == NULL) {
            return;
        }

        serial = strtoull(serial_str, 0, 16);

        auto inv = Hoymiles.getInverterBySerial(serial);

        if (inv == nullptr) {
            Serial.print(F("Can not register inverter: "));
            Serial.println(serial);
            return;
        }

        char* strlimit = new char[len + 1];
        memcpy(strlimit, payload, len);
        strlimit[len] = '\0';
        
        DynamicJsonDocument docDbus(512);
        deserializeJson(docDbus, strlimit);
        VictronPortalId = docDbus["portalId"];
                    
        DynamicJsonDocument docInstance(256);
        docInstance = docDbus["deviceInstance"];
        String deviceInstance = docInstance[serial_str];
    
        String inverter = serial_str;
        VictronDeviceInstance.insert({inverter, deviceInstance});

        if (VictronDeviceInstance.find(inverter)!=VictronDeviceInstance.end()) {
            String valfound = VictronDeviceInstance[inverter];
            Serial.print(F(" to Venus OS with portalId: "));
            Serial.print(VictronPortalId);
            Serial.print(F(" and deviceInstance: "));
            Serial.println(valfound);
        }
        return;

--> I have extended the code from OpenDTU to send the special messages for Victron part
With MqTT Explorer everything looks fine:

[image]

stromibaer, 2022-10-17:
I think registration is just be done fine:

@40000000634d804e3a5cdd54 INFO:device:**** Registering device: 116181045449, services: {'116181045449': 'pvinverter'} ****
@40000000634d804e3a72268c INFO:device:Registering Service pvinverter for client 116181045449
@40000000634d804e3a880c04 INFO:device_service_config:About to open config file
@40000000634d804f268f3934 INFO:device_service:Registering service pvinverter for client 116181045449 at path com.victronenergy.pvinverter.mqtt_116181045449_116181045449
@40000000634d804f2a3997dc INFO:settingsdevice:Setting /Settings/Devices/mqtt_116181045449_116181045449/ClassAndVrmInstance does not exist yet or must be adjusted
@40000000634d804f2e937834 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.pvinverter.mqtt_116181045449_116181045449
@40000000634d804f2f91259c INFO:device_service:Registered Service com.victronenergy.pvinverter.mqtt_116181045449_116181045449 under DeviceInstance 12
@40000000634d804f2fd268a4 INFO:device_manager:publish {'portalId': '48e7da889711', 'deviceInstance': {'116181045449': '12'}} to device/116181045449/DBus, status is 0

freakent, 2022-10-17:
It's hard to say for sure, but it looks like you are using the driver correctly. The driver doesn't have any special handling for pvinverters so maybe the problem is outside the driver. I know there are others using the driver for pvinverters (I'm not) maybe they can offer more advice.

freakent, 2022-10-17:
@hoffy Any ideas?

Hofyyy, 2022-10-18:
Hi, yes it is running on my side without any problems.
- I think you need to activate inverters so that it is shown like the critical math output. but i am not sure. if it is listed in the hw it should be ok. do you have correct values in the extended web view?
- I think it is not possible to show them stand alone. i had two inverters on die 230V side and they were grouped in the overview. otherwise the picture does get much to big. Ok they were standalone in the hw list. so i had two devices in the list. bit only one icon in the overwiew

Hofyyy, 2022-10-18:
[image]
[image]

stromibaer, 2022-10-18:
Which release do you have on Cerbo?
And can you post me your services.yml file?

stromibaer, 2022-10-18:
After doing some testing i following out
 - Switch off MqTT sender
 - Delete all pvinverter instances in portal
 - Reboot Cerbo
 - I have a working VRM Portal with realtime values
 - I have values in Cerbo

After switch on MqTT sender and registering again the pvinverters
 - Cerbo do not have values in the overview
 - I have no ESS assistent anymore
 - pvinvertes are in as device and sending values
 - VRM Portal is showing last values, but no realtime, because Cerbo has troubles

In which logfiles can find some more infos about the Cerbo system and what is going wrong?

stromibaer, 2022-10-18:
Einen Unterschied hab ich mittlerweile entdeckt --> die Formatierung der "Allowed Roles" ist bei allen MqTT PV Invertern anders als bei einem Gridzähler oder einem Zähler der die Rolle pvinverter inne hat:

[image]

[image]

[image]

[image]

Während am MqTT Explorer alle Werte zyklisch verschwinden und wieder erscheinen, bleiben diese hier permanent stehen:

[image]

Und lassen sich mit:

[image]

[image]

auch nicht überschreiben. 
Die Position lässt sich überschreiben.

Was steht da bei euch drin an "Allowed Role"?

stromibaer, 2022-10-18:
@stundenblume i have seen from the victron community, that you has worked on same topic and had also some issues. Do you have now a working environment and can share it?

Hofyyy, 2022-10-18:
Sieht bei mir komplett anders aus. Ich gehe mit dem MQTT Browser direkt auf meinen Victron Device V2.90.
Ich hab deine Allowed Werte garnicht.
[image]

stromibaer, 2022-10-20:
Lange Rede nun gefunden: `String fieldvalue = String(inv->Statistics()->getChannelFieldValue(channel, fieldId));`
Die Wertausgabe muss ein Float sein, dann läuft alles.

closed 2022-10-20 by stromibaer

freakent, 2022-10-20:
> Closed #22 as completed.
>
> [
> {
> "target": "https://github.com/freakent/dbus-mqtt-devices/issues/22#event-7634411807",
> "url": "https://github.com/freakent/dbus-mqtt-devices/issues/22#event-7634411807",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Thanks for closing, but could you please post a short summary so that any else who comes here with the same problem can benefit?Regards,Martin+44 7776490601

freakent, 2022-10-20:
So just to be clear, is this a change you made to your client side code? Is this the data you are sending to the Victron MQTT topics for pvinverter?

stromibaer, 2022-10-21:
Yes, client side. I have adapted den Open-DTU with some additions to publish the values from Hoymiles inverters directly to Venus OS.

stundenblume, 2022-11-01:
> @stundenblume i have seen from the victron community, that you has worked on same topic and had also some issues. Do you have now a working environment and can share it?

sorry, was on vacation.

[Open-DTU](https://github.com/tbnobody/OpenDTU) seems to be a very nice project!
That would be a reason to by Hoymiles inverters instead of the APSystems.