freakent/dbus-mqtt-devices #23: Adopt services.yml for pvinverters

stromibaer, 2022-10-20:
Pls add following lines and/or adopt some lines in the pvinverter section:

[image]

to:

 ```
 AllowedRoles:
    default: ["grid", "pvinverter", "genset", "acload"]
  Role:
    description: "grid, pvinverter, genset, acload"
    persist: true
    default: "pvinverter"
```

The default line in AllowedRoles has some format issues with ''.

And adopt the comment field:

[image]

to:

 ```
 ProductId:
    default: 45069 # 45069 ET340, 45068 ET112 - 0xB00D, ESS Demo mode uses 45058
```

ET340 is for 3 phases and ET112 is for one phase metering.

With this changes it is possible to change default location of the inverter from AC-OUT to AC-IN or vice versa in the Cerbo settings menue.

Thanks,
Peter

pos-ei-don, 2023-01-19:
Hi!
Why to you use a grid meter (ET***) as PvInverter? They cannot produce power?!?
Shouldn't a PV-Inverter use a ProductID like 45069 (=Fronius)?
So you get the power shown in VRM and other Dashboards?
In my tests, this looks much more promising.

freakent, 2023-01-19:
I really don't think the ProductId has any effect on how the system actually works.

pos-ei-don, 2023-01-19:
But i tested it and saw the difference.

stundenblume, 2023-01-20:
You can use energy meters to measure PV energy.

[Victron:](https://www.victronenergy.com/accessories/energy-meter)
_[...] Also they can be used to measure the output of a PV Inverter, to display the data on GX device [...]_

In my experience (Software versions april 2022), only the definition of the type has an impact on the display in the VRM portal.

freakent, 2025-04-20:
Dear @stromibaer, @pos-ei-don, @stundenblume, @patatman,

This issue has been outstanding for a long long time. I have looked at the config and can not see anything that needs to be changed. Please confirm if this can be finally closed.

Martin

freakent, 2025-04-20:
@pos-ei-don, could I ask what you are using instead?

patatman, 2025-04-20:
I've also moved away from this, since I couldn't get it working stable. 
I'm now using 2 dedicated power meters to measure the power generated from my PV inverters. 

There was nog other way, since the Dynamic ESS needs a high refresh rate.

closed 2025-04-27 by freakent