freakent/dbus-mqtt-devices #25: Help needed with OpenHab sending JSON

Joeknx, 2022-11-15:
Thank you for the great tool. Exactly what I need.

I have a working keep alive rule (DSL rule)

`val actions = getActions("mqtt", "mqtt:broker:VenusOS")
        actions.publishMQTT("R/xxxxxxxx/keepalive", '["grid/40/Ac/Frequency", "grid/40/Ac/Energy/Forward"]')
        logInfo("Solar", "CCGX MQTT Keep Alive Timer fired!!")`

I can't manage to send the JSON 

Example: 
`actions.publishMQTT(„device/fe001/Status“, '[{ "clientId": "fe001", "connected": 1, "version": "v1.0 ALPHA", "services": {"t1": "temperature"} }]')`

...is not working.

`actions.publishMQTT(„device/fe001/Status“, '{ "clientId": "fe001", "connected": 1, "version": "v1.0 ALPHA", "services": {"t1": "temperature"} }')`

also not.

closed 2022-11-15 by Joeknx

freakent, 2022-11-15:
Can I just confirm status following your last message? You said you can not send JSON? So is this now working or not?

Joeknx, 2022-11-15:
It is working now. Thank you! 
`val actions = getActions("mqtt", "mqtt:broker:VenusOS")
        actions.publishMQTT("device/fe001/Status", '{ "clientId": "fe001", "connected": 1, "version": "v1.0 ALPHA", "services": {"t1": "temperature"} }')
  `

But i am still struggling with the value. 
`actions.publishMQTT("W/XXXXXX/temperature/<device instance>/Temperature", '{"value": 24.91}')`

What is the device instance? t1?

Thank you!

freakent, 2022-11-15:
No, the device instance is the number you get back from the dbus by sending the registration message. You need to subscribe to "device/<client id>/DBus"before you publish the connect message. Please read the protocol section within the docs.

Joeknx, 2022-11-16:
Thank you. It is working now.
I tried to understand the readme, but did not quit get it.

Sorry for the beginner questions!

freakent, 2022-11-16:
No problem. I'm glad you got it working. I'm always happy to receive suggestions that improve the documentation.