freakent/dbus-mqtt-devices #26: PV Inverter not showing up

Joeknx, 2022-11-17:
Hi, 

i try to push "AC/Power" to the venus (with OpenHab rule Javascript).

`var energie = items['Active_power'];
actions.get('mqtt', 'mqtt:broker:VenusOS').publishMQTT('W/xxxxxxx/pvinverter/1/Ac/Power', '{"value": ' + energie + ' }');`

Init:
`actions.get('mqtt', 'mqtt:broker:VenusOS').publishMQTT('device/pv001/Status', '{ "clientId": "pv001", "connected": 1, "version": "v1.0 ALPHA", "services": {"pv1": "pvinverter"} }');`

PV-Inverter shoes up in Settings (Remote console und VRM Portal) but not in the dashboard (Remote console und VRM Portal)
I changed the "services.yml" already to get the role thing, but it is still not showing up.

What am I missing or doing wrong?

freakent, 2022-11-17:
When you register your device in the, what deviceId do you get back in the device/pv001/DBus topic?
You have hardcoded 1 for the deviceId in the topic W/xxxxxxx/pvinverter/1/Ac/Power

`var energie = items['Active_power']; actions.get('mqtt', 'mqtt:broker:VenusOS').publishMQTT('W/xxxxxxx/pvinverter/1/Ac/Power', '{"value": ' + energie + ' }'); 
`

freakent, 2022-11-17:
I highly recommend you test your messages in MQTT-Explorer before implementing them in code.

Joeknx, 2022-11-17:
I get 1 in device/pv001/DBus topic.

`publishMQTT('W/xxxxxxx/pvinverter/1/Ac/Power',` '{"value":  100}');`

for example shows 100 W in the device list from Venus os

freakent, 2022-11-17:
The driver has no control on what gets sent by the GX device to VRM. When you are sending values to your device you are not using the driver at all. 

Perhaps it is just a time lag? The GX sends data to VRM in batches, usually every 15 mins. 

If in doubt, test the data you are sending by using MQTT-Explorer.

Joeknx, 2022-11-20:
The PV Inverter needs all values to be displayed. 
I tried only to forward "total power" of the Inverter.

With all values (L1, L2, L3 etc.) it showed up instantly in the Remote console und VRM Portal.

freakent, 2022-11-20:
Awesome. Thanks for sharing the solution.

Joeknx, 2022-11-21:
I also noticed, that sending the "total power" of the inverter to `W/xxxxxxx/pvinverter/1/Ac/Power` is still showing 0W in the Remote console und VRM Portal.
Sending the "total power" to `W/xxxxxxxx/pvinverter/1/Ac/L1/Power` works!

closed 2022-12-14 by freakent