freakent/dbus-mqtt-devices #31: Request for support com.victronenergy.solarcharger

pos-ei-don, 2023-01-11:
I am currently trying to add a solarcharger,
to feed in data from an additional power-source (a balkony system
where i measure the power with tasmota in mqtt).
My goal is to change the mqtt-Data from Tasmota via Nodered to be compatible to this
driver so i can see the data in the vrm plots and graphs.

My current work in Progress is below.
I see the device popping up and showing some numbers, but it stays in not activ status.
Can someone help me where  i might have an error?
Has someone a working example? I tried to stay basic, then i added addidional values.
If I add the commented out  # ErrorCode, it does not show up at all. Why?

best regards, Joe

solarcharger:
  ProductId:
    default: 41074 # 
  CustomName:
    default: "Solarcharger"
    persist: true
  NrOfTrackers:
    description: "Trackers"
    default: 1
  Pv/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
  Pv/0/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
  Pv/0/V: #
    description: "V"
    default: 0
    format: "{} V"
  Pv/V: # DD
    description: "V"
    default: 0
    format: "{} V"
  Dc/0/Current: # DD
    description: "A"
    default: 0
    format: "{} A"
  Dc/0/Voltage: # DD
    description: "V"
    default: 52
    format: "{} V"
  Yield/Power: #
    description: "W"
    default: 0
    format: "{} W"
  Yield/SYSTEM: #
    description: "W"
    default: 0
    format: "{} W"
  Yield/User: #
  Link/NetworkStatus:
    description: ""
    default: 4 # 0x8
  Link/NetworkMode:
    description: ""
    default: 13 # 0x8
  Link/ChargeCurrent:
    description: ""
    default: 20 # 0x8
#  ErrorCode
#    description: "0=No Error"
#    default: 0 # not showing in in console???
#  Connected:
#    description: ""
#    default: 1 #
  Mode:
    description: ""
    default: 1 #
  MppOperationMode:
    description: ""
    default: 2 #

freakent, 2023-01-11:
Hi, sounds like an interesting project.

>I see the device popping up and showing some numbers, but it stays in not activ status.
Can you explain more? If you stop the driver or disconnect from MQTT with a Last Will message (as recommended) the then dbus will show the device as disconnected. 

Have you check the troubleshooting tips in the Readme?

freakent, 2023-01-11:
For examples, take a look in the test-data folder and the various simulator scripts provided.

pos-ei-don, 2023-01-11:
Yes i did, i even wrote a test-data script for this charger-tests.
My current guess is that it does not work because of the missing "ErrorCode"-Part.
But i cannot configure this because then the Device dissappears totally. I hoped that someone finds the error or maybe there is a change needed somewhere in the code?

```
import paho.mqtt.client as mqtt
import json
import copy

clientid = "ch001"

registration = {
  "clientId": clientid,
  "connected": 1,
  "version": "v0.2",
  "services": {
    "ch1": "solarcharger"
  }
}

unregister = copy.deepcopy(registration)
unregister["connected"] = 0

data = {
    "ProductId": 41074,
    "NrOfTrackers": 1,
#    "Connected": 1,
    "Pv/P": 22,
    "Pv/0/P": 23,
    "Pv/V": 81,
    "Pv/0/V": 81,
    "Dc/0/Current": 3,
    "Dc/0/Voltage": 53,
#    "Link/NetworkMode": "13",
#    "MppOperationMode": "2",
#    "Yield/Power": 300,
#    "ErrorCode": 0,
    "State": 1,
#    "Mode": 1,
}
```

freakent, 2023-01-11:
Can you upload the whole script? How are you using that data variable?

pos-ei-don, 2023-01-11:
Here the 2 Files. 

BTW. Thank you A LOT for this project!!!!

As soon as i uncomment the ErrorCode - Part in the service.yml, it doesnt work.

[charger-simulator.zip](https://github.com/freakent/dbus-mqtt-devices/files/10391815/charger-simulator.zip)

pos-ei-don, 2023-01-11:
BTW , maybe we can post the nodered-flow to fill the data with values in the installation instructions?
Maybe this helps others aswell, and as victron includes nodered in venusos directly, it is a very easy way to use such virtual devices!
I can provide my currend flow if needed!

freakent, 2023-01-11:
> As soon as i uncomment the ErrorCode - Part in the service.yml, it doesnt work.

Is there an error in the log file?
`more /var/log/dbus-mqtt-devices/current`

Does the error happen during device registration or when you publish data.  Do you know which part of your code triggers the error?

pos-ei-don, 2023-01-11:
The Error seems to be 

> ERROR:device_manager:Unexpected error: <class 'yaml.scanner.ScannerError'

freakent, 2023-01-11:
That sounds like a YAML parsing error. Looking at your settings files, I can see you have a missing colon ":" after ErrorCode

ErrorCode: # colon required here. 
#    description: "0=No Error"
#    default: 0 # not showing in in console???
#  Connected:
#    description: ""
#    default: 1 #

If in doubt, copy paste your yaml file into any of the online yaml verification tools you can find online.

freakent, 2023-01-11:
Haha, I didn't see your edit. Glad you spotted the error too.

pos-ei-don, 2023-01-11:
Sorry, didn't want you to invest time in an already found error....

Unfortunately, I cannot get it to work.
Do you know anything about this errormessage?
How can I debug this? The errormessage of your driver does not give any hint what is going on.....

[image]

freakent, 2023-01-11:
Again, let's split the problem in half. Is that after registration or registration + sending data?

pos-ei-don, 2023-01-12:
OK, i checked that. This Message appears right after the registration.
So i have to change the service.yml to find the problem, right?

pos-ei-don, 2023-01-12:
Ok, things change without any change.
After a reboot, i can register the device and it pops up. THe message about beeing incompatible
now just appears when sending data.
I changed the data  a lot, but the message keeps appearing. I think I am at a dead-end here.... :(

freakent, 2023-01-12:
I can see some incorrect paths in your services.yml 
See https://github.com/victronenergy/venus/wiki/dbus#solar-chargers

/SYSTEM should be /System for example.

pos-ei-don, 2023-01-12:
Okay, i see, but this is outcommented long ago since i don't need is currently.
I am just trying to find a minimal configuration which works, but i am unable.
When i compare the topics to my hardware-device, there are some differences. maybe the dokumentation is
little outdated. I'm lost here, so unfortunately I am giving up...

stundenblume, 2023-01-13:
God morning.
Without reading the whole issue, this is my [Tasmota approach](https://github.com/freakent/dbus-mqtt-devices/wiki#get-tasmota-to-act-like-an-victron-recognized-pv-inverter).
Sebastian

pos-ei-don, 2023-01-13:
@stundenblume, thank you. But you register it as pvinverter,
I need to register it as solarcharger...

stundenblume, 2023-01-13:
OK.
As I understand, you have already a real solarcharger.
My approach for a up to than unsupported device for this driver (pv-inverter) was:
0. install MQTT Explorer
1. get all topics via MQTT Explorer of the real, working device
2. add solarcharger to service.yml
3. let the driver register a solarcharger
4. fill some topics of the the new charger with data via MQTT Explorer

Maybe you already did all that.
Right now I don't have a test system running for active help.

pos-ei-don, 2023-01-13:
Yes I did that. Thank you anyway for trying to help me!!
There are differences between my real systems and the Dokumentation. For example /Pv/V and /Pv/0/V

I didn't find out which one works, as in this case no solution worked.
But I do use the virtual battery and other devices and this works very well!!

pos-ei-don, 2023-01-16:
I have a Question for understanding: 
What is the better way to try: Better fill services.yml with the values and then try to fill them with mqtt, or
keep services.yml small and try different values with mqtt (are values not configured in services.yml ignored?), or do they have to be in sync at every time?
Do you have a restart-script for the Service? Currently i reboot at every change.

pos-ei-don, 2023-01-16:
Thank you, I will test it.
Unfortunately, I cannot have "long" downtimes of your great service, because i use the virtual battery in production
and it is very helpful/stable! A Really great service you developed here!!

Yes, I still get the same message, regardless what values i try to send. It worked for a very short time at the beginning, but i am not able to reproduce this, although the config which worked is available.
I thought that maybe i have some spare-values in my victron-config as vrm shows me a vebus-device in the gui which does not exist and which does not show up in the device-list.  So maybe in need a factory-reset?!

stundenblume, 2023-01-16:
Can you set up a second RasPi for testing? There should no other hardware be necessary.
I did that when I was playing with the services.yml and pv-inverter.

pos-ei-don, 2023-01-17:
I did. Same error message. I think I just need the first correct values to Start... 😏

pos-ei-don, 2023-01-17:
Maybe i missunderstood ProductId:.
When to use a real ProductId, when to use 65535# 
 I saw noth, when I choose 65535, it is displayed.

freakent, 2023-01-18:
I am just reading the Victron docs for Solar Charger devices. I don't think you should be mixing numbered and unnumbered attributes, for example:

```
  Pv/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
  Pv/0/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
```

The docs say:

``` 
PV Array voltages and currents:
/Pv/V             <- PV array voltage, path exists only for single tracker product (all common MPPTs)
/Pv/0/V           <- PV array voltage from 1st tracker
/Pv/x/V           <- PV array voltage from tracker x+1; today's max number of trackers in a single Victron product is 4.
/Pv/0/P           <- PV array power (Watts) from 1st tracker
/Pv/x/P           <- PV array power from tracker no. x+1. 
/Yield/Power      <- Total PV power (Watts).

  Two examples:
  1) a solar charger with one tracker has 2 paths:
  /Pv/V
  /Yield/Power

  2) a solar charger with two trackers has 5 paths:
  /Pv/0/V
  /Pv/0/P
  /Pv/1/V
  /Pv/1/P
  /Yield Power

```
This suggests to me you should be using one or the other not both.

freakent, 2023-01-18:
How did you determine 13 for the /Link/NetworkMode path?

```

/Link/NetworkMode     <- Bitmask
                      0x1 = External control
                      0x4 = External voltage/current control
                      0x8 = Controled by BMS (causes Error #67, BMS lost, if external control is interrupted).

```

The docs suggest to me that it should be 1, 4 or 8 (in decimal). I would think 1 would be a good place to start.

freakent, 2023-01-18:
The Victron docs list two settings, so I think you should have these defined with "persist: true".

```
/Settings/BmsPresent         <- BMS in the system. External control is expected.
                                This happens automatically if NetworkMode is set to expect a BMS.
/Settings/ChargeCurrentLimit <- The maximum configured (non-volatile) charge current. This is the same as set by VictronConnect.

```

freakent, 2023-01-18:
I just noticed you have persist: true on a couple of parameters that are not settings, I would remove the persist where it is not needed. 

```
Pv/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
  Pv/0/P: # 
    description: "W"
    persist: true
    default: 0
    format: "{} W"
```

pos-ei-don, 2023-01-18:
This is all testdata.
As the ProductID i used seems to block every setting for the device, i had to many different
datas for tests.
Now with a non defined ProductID, it seems to work and i can try to find a complete config to share.
Thank you a lot!!

freakent, 2023-01-18:
> Do you have a restart-script for the Service? Currently i reboot at every change.

All services on Venus OS run under daemintools (https://cr.yp.to/daemontools.html)

From memory, you should be able to use the following command (from ssh command line) to top the service

```
svc -d dbus-mqtt-devices 
```

and the following command to restart it:

```
svc -u dbus-mqtt-devices
```

closed 2023-03-03 by freakent