freakent/dbus-mqtt-devices #32: Request for support com.victronenergy.evcharger

seliSoft, 2023-01-17:
https://github.com/victronenergy/venus/wiki/dbus#grid-and-genset-meter

Would it be possible to add evcharger devices, so i can include my wallboxes?
Thank you very much!

stundenblume, 2023-01-17:
Hi,
you can edit [services.yml](https://github.com/freakent/dbus-mqtt-devices/blob/main/services.yml) according to the [evcharger](https://github.com/victronenergy/venus/wiki/dbus#evcharger) and it should work.
Sebastian

pos-ei-don, 2023-01-18:
Had the same need:

here is my currend work in progress:
you can see it in vrm and monitor power with this

Unfortunately, i think we need help fom @freakent 
* How can i set it connected/disconnected? it seems that it does nor work changing this value
*  MaxCurrent does not work persisdent.
* StartStopp thows an error

```
evcharger:
  ProductId:
    default: 65535 # 0xFFFF
  CustomName:
    default: "MyCharger"
    persist: true
  Ac/Energy/Forward:
    description: "wh"
    format: "{:.2f}W"
  Ac/L1/Power:
    description: "W"
    format: "{} W"
  Ac/L2/Power:
    description: "w"
    format: "{} W"
  Ac/L3/Power:
    description: "w"
    format: "{} W"
  Ac/Power:
    description: "w"
    format: "{} W"
  AutoStart:
#    description: "w"
    default: "0"
    persist: true
#    format: "{}"
  ChargingTime:
    description: "in seconds"
    format: "{} s"
  Current:
    description: "A"
    default: 0
    format: "{}"
#    persist: true
  EnableDisplay:
    description: ""
    format: "{}"
  MaxCurrent:
    description: "A"
    default: 0
#    format: "{}"
    #persist: true   #wird nicht gespeichert reset nach disconnect/reconnect
  Mode:
    description: ""
    default: 0
    format: "{}"
  Model:
    description: "w"
    default: "AC22NS"
    format: "{}"
  Position:
    description: ""
    format: "{}"
    #persist: true   # geht nicht
  Role:
    description: "w"
    format: "{}"
  SetCurrent:
    description: "w"
    format: "{}"
#  StartStop:           #mqtt deamon sbeendet sich fatal wenn gesetzt
#    description: "w"
#    format: "{}"
#    persist: true
  Status:
    description: "w"
    default: 1
```

pos-ei-don, 2023-01-18:
[image]
[image]
[image]

pos-ei-don, 2023-01-18:
This is the StartStop-Errormessage
```
INFO:device_manager:Received device status message {'clientId': 'ev002', 'connected': 1, 'version': 'v0.9', 'services': {'ev1': 'evcharger'}}
exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
Traceback (most recent call last):
  File "/data/drivers/dbus-mqtt-devices-0.5.1/ext/velib_python/ve_utils.py", line 24, in exit_on_error
    return func(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 60, in _on_message
    if self._status_is_valid(status):
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 109, in _status_is_valid
    if services.get(service_id) not in self.service_types: # as defined in services.yml
TypeError: argument of type 'NoneType' is not iterable
```

pos-ei-don, 2023-01-18:
> Is the services.yml extract you have provided the exact version giving this error?

yes!!

> 
> The StartStop section looks OK, apart from having persist: true. 
> Only use persist: true for static device settings. The problem is now that you have declared this a persisted value I think dbus will always retain the value you last set.

so i have to save the state in nodered and send it after restarts?
Is there any chance to delete this? 
i already had to reinstall my production-venusOS because of issues regarding this... ;-)

> 
> The AutoStart property does have a problem, the description is commented out. Every section MUST have a description attribute. Also you have comments around the default value, but I think that should be a numeric value not a string.
> 
> ```
>   AutoStart:
> #    description: "w"
>     default: "0"
>     persist: true
> #    format: "{}"
> ```

ok, changed this. thank you. 
Do you know why Charging Current is greyed out?

freakent, 2023-01-18:
> Is there any chance to delete this?
Have a look at a victron tool called dbus-spy, it will help you see what is actually help in the dbus.

> Do you know why Charging Current is greyed out?
Sorry, I don't

Here are a few resources for you to cross reference:

1) The Victron modbus-TCP register list. You can request a download of the spreadsheet from Victron, which includes a section for evcharger: https://www.victronenergy.com/panel-systems-remote-monitoring/color-control#technical-information

2) Victron's dbus-modbus implemtation for evcharger: https://github.com/victronenergy/dbus-modbus-client/blob/master/ev_charger.py

3) Open_EVSE charger Victron implementation : https://github.com/JuWorkshop/dbus-evsecharger/blob/main/dbus-evsecharger.py

pos-ei-don, 2023-01-18:
Here you can see /Connected as parameter
https://github.com/victronenergy/venus/wiki/dbus#evcharger

This thets the vehicle to be connected.
But when I try to use it, I get:
> #"Can't register the object-path handler for '/Connected': there is already a handler"

Is this not possible to use because of internals of this driver?

freakent, 2023-01-18:
Where do you see this message? It's not one I recognise.

pos-ei-don, 2023-01-18:
When i add " Connected:" to the services.yml for the evcharger profile
and then try to start the driver in the console, i see this error.

I thought that this is because your internal varuable uses "conncted" for the virtual device? but this is just a guess...

freakent, 2023-01-18:
> I thought that this is because your internal varuable uses "conncted" for the virtual device? but this is just a guess...

I doubt it, very different.

Please show the relevant section from your services.yml, /Connected is not in the extract you provided earlier.

freakent, 2023-01-18:
>When i add " Connected:" to the services.yml for the evcharger profile
and then try to start the driver in the console, i see this error.

Please also provide full text from the logging so I can see the context.

freakent, 2023-01-18:
If you get weird messages like this, change the client id you send in the registration message (currently ev002) so that you create a new, clean set of dbus entries.

freakent, 2023-01-18:
I noticed you have lots of 

```
  format: "{}"
```. 

Remove these, they add no value. Only use a format if the default format is not what you want. My worry is that you may be trying to apply formats incorrectly.

pos-ei-don, 2023-01-18:
ok, i did.

here is the full errormessage

```
INFO:logger:Loglevel set to INFO
INFO:device_manager:Using portalId b827eb20de35
INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
INFO:device_manager:[Connected] Result code 0
INFO:device_manager:Received device status message {'clientId': 'ev003', 'connected': 1, 'version': 'v0.9', 'services': {'ev3': 'evcharger'}}
INFO:device:**** Registering device: ev003, services: {'ev3': 'evcharger'} ****
INFO:device:Registering Service evcharger for client ev003
INFO:device_service_config:About to open config file
INFO:device_service:Registering service evcharger for client ev003 at path com.victronenergy.evcharger.mqtt_ev003_ev3
INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_ev003_ev3/CustomName does not exist yet or must be adjusted
INFO:settingsdevice:Setting /Settings/Devices/mqtt_ev003_ev3/ClassAndVrmInstance does not exist yet or must be adjusted
INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.evcharger.mqtt_ev003_ev3
exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
Traceback (most recent call last):
  File "/data/drivers/dbus-mqtt-devices-0.5.1/ext/velib_python/ve_utils.py", line 24, in exit_on_error
    return func(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 62, in _on_message
    self._process_device(status)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 143, in _process_device
    self._devices[clientId] = device = MQTTDevice(device_mgr=self, device_status=status)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device.py", line 30, in __init__
    device_service = MQTTDeviceService(self, id, service)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_service.py", line 45, in __init__
    self._set_up_dbus_paths()
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_service.py", line 103, in _set_up_dbus_paths
    dbus_service.add_path("/"+k, value=value, description=v.get('description'), writeable=True, gettextcallback=textformatcallback, onchangecallback=changecallback)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/ext/velib_python/vedbus.py", line 107, in add_path
    item = VeDbusItemExport(
  File "/data/drivers/dbus-mqtt-devices-0.5.1/ext/velib_python/vedbus.py", line 468, in __init__
    dbus.service.Object.__init__(self, bus, objectPath)
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 489, in __init__
    self.add_to_connection(conn, object_path)
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 578, in add_to_connection
    connection._register_object_path(path, self._message_cb,
KeyError: "Can't register the object-path handler for '/Connected': there is already a handler"
```

pos-ei-don, 2023-01-18:
But i think in this description and context, this value is used to tell the device that the vehicle is connected to the powerwall.
Is there a workaround possible?

freakent, 2023-01-18:
No, it's fundamental. I haven't seen any Victron documentation that suggests that it indicates whether it is connected to the car. It would also mean that the definition of Connected is different for ev chargers than any other service.

pos-ei-don, 2023-01-18:
ok. But whats about StartStop?
Here is the log

```
*** starting dbus-mqtt-devices ***
-------- dbus_mqtt_devices, v0.5.1 is starting up --------
INFO:logger:Loglevel set to INFO
INFO:device_manager:Using portalId b827eb20de35
ERROR:device_manager:Unexpected error: <class 'yaml.scanner.ScannerError'>
INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
INFO:device_manager:[Connected] Result code 0
INFO:device_manager:Received device status message {'clientId': 'ev003', 'connected': 1, 'version': 'v0.9', 'services': {'ev3': 'evcharger'}}
exit_on_error: there was an exception. Printing stacktrace will be tried and then exit
Traceback (most recent call last):
  File "/data/drivers/dbus-mqtt-devices-0.5.1/ext/velib_python/ve_utils.py", line 24, in exit_on_error
    return func(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 60, in _on_message
    if self._status_is_valid(status):
  File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 109, in _status_is_valid
    if services.get(service_id) not in self.service_types: # as defined in services.yml
TypeError: argument of type 'NoneType' is not iterable

```

freakent, 2023-01-18:
It's definitely a YAML parse error:
```
<class 'yaml.scanner.ScannerError'>
```

pos-ei-don, 2023-01-18:
Interessant, das habe ich sicher drei oder vier mal gecheckt, es dürfte aber mit dem Kommentar dahinter zu tun haben. Als ich das gelöscht habe, gings.
Vielen Dank, und sorry für die häufigen Rückfragen...

seliSoft, 2023-01-18:
Thank you!
With this config, "position" becomes editable:
---------------
  Position:
    persist: true
    default: 0
    min: 0
    max: 1
    description: "0=OUTPUT; 1=INPUT"
-------------
[image]
-------------
Now, my chargers are visible in VRM.
(How) could this implementation be used to do "PV-Überschussladen" without other tools?

pos-ei-don, 2023-01-19:
Here is a current Screenshot: Works great.
Sadly, Victron seems to have no "soc" field for the car.
Would be a nice info aswell.

[image]

freakent, 2023-01-19:
Why don't you post a suggestion on the Victron community forum to add an SOC field to EVCharger on the dbus. No need to mention this driver, it's the dbus api that needs to be extended.

pos-ei-don, 2023-01-19:
Thank you! I'm not "into" it , I am just new to victron so i am learning a lot currently.
I need to get a useraccount there, so yes, i will try to do so the next days.

freakent, 2023-01-19:
You could also register the car's battery on the system and push SOC to the battery.

stundenblume, 2023-02-15:
@pos-ei-don
I tried your expanded services.yml from your PR.
In venusOS 2.82 this driver did not register a new device. I will try it on a newer version of venusOS.

pos-ei-don, 2023-02-15:
I have multiple instances running, di I think it should work.
I am so happy with it the at I even migtated another Inverter from the shelly plugin to this one. I live it to be able to manipilulate some values while the GX is up and running

stundenblume, 2023-02-15:
Maybe I did something wrong.
This driver runs smooth since one year, so I'm not in the flow right now... :-D
I would like to monitor my dumpload as evcharger.

I love It too. It's so flexible and easy to use! I don't know why there is not more attention. Or maybe it is, but just a view people have problems. ;-)

pos-ei-don, 2023-02-15:
I think we should Post easy to use node Red flows to create Devices and fill them with values. This should help others to get startet much faster. 
I did some programming in NodeRed, but currently it does not run without manual setting a value, so it needs some more attention 😉

stundenblume, 2023-02-15:
Off topic: Also reading a existing Smartmeter with Tasmota and send it directly to venusOS is on the list. That would eliminate the need of a (relative expensive) EM24.

pos-ei-don, 2023-02-15:
I use this, and it works well! Also hoymiles can be integrated directly.  (and the EV charger for the car, but dlthis is a different Topic)

freakent, 2023-02-15:
> I think we should Post easy to use node Red flows to create Devices and fill them with values. This should help others to get startet much faster. 
> 
> I did some programming in NodeRed, but currently it does not run without manual setting a value, so it needs some more attention 😉

The best place to post sample flows for NodeRed is https://flows.nodered.org. I don't mind adding links in the documentation to to sample flows on https://flows.nodered.org.

stundenblume, 2023-02-17:
@pos-ei-don
what are the minimal information the victron system needs, to show up the charger in the overview?
With just power and energy it didn't.

pos-ei-don, 2023-02-17:
I've never testes "the minimal" settings.
Here are the values I set from the nodered to this device.
Some other settings i set in the console (charge-current,...) and send it the car. 

[image]

Are you shure that the registration succeded? You can monitor the calues in MQTT, did the device in the /N/ path (N/<id>/evcharger/... ) take the values? be shure to send numbers, not strings!

stundenblume, 2023-02-17:
[image]
Here everything looks fine.
But not in the overview. But that's maybe because it's just a RasPi with nothing else connected. I Will try it on my main system.
[image]

pos-ei-don, 2023-02-17:
I dont use the overview.  But i dont See it there either. I use the online vrm.
Do you See it there?

stundenblume, 2023-02-17:
You are right. It shows up in VRM.
Thanks :)
[image]

pos-ei-don, 2023-02-17:
You are charging with 51w? 😉.
Do you plan to connect a solar charger with this?

stundenblume, 2023-02-17:
This is just a test environment.
I Use this driver to send temperatures and power of 6 APSystems inverters to the victron system.
I would like to use the "evcharger" for monitoring a dumpload.
A friend will use it to monitor charging of his motorcycle.
I hope you services.yml will be included in the next version of the driver.

50W is just a example Power. :-D

stundenblume, 2023-02-21:
@pos-ei-don
I have it running now for a few days without problems.
It's nice to see a ev charging but are there other benefits?
1. I was expecting a calculation of how much energy did go to the ev like the battery (direct use, to battery, to grid, to ev).
2. Can the Victron system control the charger? Start/stop, power control, ...

pos-ei-don, 2023-02-21:
1. You can add an additional battery device, but i am unshure if its worth it.
2. You can Start and stop, but its on you to Code the backen connection to your charger. My EV-charger works with 0-10V input, so i can change the charge-current within Victron/NodeRed as needed.

stundenblume, 2023-02-21:
Is start/stop and charge current native with VenusOS possible or only with separate logic like NodeRed on VenusOS lage?

I can't start/stop charging using the GUI, the slider is grayed out.

stephwe, 2023-02-26:
The wallbox can be controlled with VRM or the Cerbo GX via openWB.
However, it doesn't make any sense because openWB controls the wallbox very well and you have a lot more setting options.

Otherwise everything works great. Unfortunately, the SoC from the car is not transmitted, but it is not that important either.

The flow for Node Red openWB - Cerbo GX is available here:
https://openwb.de/forum/viewtopic.php?p=79599#p79599

stundenblume, 2023-02-27:
@stephwe 
Thank You Stephan.
But I don't have a real Smart EV charger. I have two use cases.
1. Electric motorcycle with a Tasmota flashed power plug.
2. A dump load controller (Tasmota).
Right now I control them from ioBroker. But I think it would be nice if the Victron system could control them. Just to have a "cleaner" system.

pos-ei-don, 2023-02-27:
The on/off switch does not work for me either.
But Automatic/Manual charging works. So i used this one for turning something on/off.
Its only a workaround, i hope that victron will add several needed features to its EV module.

closed 2023-03-03 by freakent

disaster123, 2023-11-19:
Is therre any reason why this is closed? The service file still does not contain evcharger or pvcharger.

freakent, 2023-11-19:
Looks like the thread went off topic and a working config was not verified for evcharger. Going forward, I am replacing the single services yaml file with separate config files for each service. It's in beta at the moment. I'm not going to try to merge in any more community changes to the services file only to have to migrate them later. I also need contributors to provide tests so that I can verify any new services work as I have no way of testing them.

disaster123, 2023-11-19:
I already use the 0.7.5-beta which works great for me. I'm happy to test new service files.

pos-ei-don, 2023-11-20:
I stopped using vrm at the Moment so i am not able to change this to the new Format. I cannot test it. 
I hope someone will take over 😉.

dersch81, 2024-03-05:
> Looks like the thread went off topic and a working config was not verified for evcharger. Going forward, I am replacing the single services yaml file with separate config files for each service. [...]

I can confirm that this config is working (since a couple of month now)

It would be great to have it included, then the mqtt EV Charger user don't need to inlcude this manually after every upgrade of dbus_mqtt_devices.

```
# https://github.com/victronenergy/venus/wiki/dbus#evcharger
evcharger:
  ProductId:
    default: 65535 # 0xFFFF
  CustomName:
    default: "MyCharger"
    persist: true
  Ac/Energy/Forward:
    description: "wh"
    format: "{:.2f}W"
  Ac/L1/Power:
    description: "W"
    format: "{} W"
  Ac/L2/Power:
    description: "w"
    format: "{} W"
  Ac/L3/Power:
    description: "w"
    format: "{} W"
  Ac/Power:
    description: "w"
    format: "{} W"
  AutoStart:
    description: "w"
    default: 0
#    persist: true
  ChargingTime:
    description: "in seconds"
    format: "{} s"
  Current:
    description: "A"
    default: 0
#    persist: true
  EnableDisplay:
    description: ""
  MaxCurrent:
    description: "A"
    default: 0
#    persist: true   #wird nicht gespeichert reset nach disconnect/reconnect
  Mode:
    description: ""
    default: 0
  Model:
    description: "w"
    default: "AC22NS"
  Position:
    persist: true
    default: 0
    min: 0
    max: 1
    description: "0=OUTPUT; 1=INPUT"
#  Connected:
#    description: "w"
  Role:
    description: "w"
  SetCurrent:
    description: "w"
#  StartStop:
#    description: "w"
#    persist: true
  Status:
    description: "w"
    default: 1
```

freakent, 2024-03-06:
@dersch81 Thanks for raising this again. This thread went so off topic it got lost. I think this service description still needs refinement. I will open a new issue and will propose some improvements. I would welcome your collaboration (and anyone else interested) on this.