freakent/dbus-mqtt-devices #38: Cerbo GX crashes when writing to grid L1/L2 topics

mtucker502, 2023-02-02:
Note: My current setup is off-grid. I want to validate that I can get the sensor values into ESS before I make the grid connection.

I am able to write to Ac/Energy and Ac/Power without issue. I can also write to L1/Voltage and L2/Voltage. But if I write to L*/Power or L*/Current the screen crashes. If I leave my client connected long enough entire Cerbo will reboot.

### dbus-mqtt-device logs:

```
@4000000063dc161125893314 INFO:device_manager:Received device status message {'clientId': 'ctsensor', 'connected': 1, 'version': '1', 'services': {'L1': 'grid'}}
@4000000063dc1611259a0f7c INFO:device:**** Registering device: ctsensor, services: {'L1': 'grid'} ****
@4000000063dc161125a5db04 INFO:device:Registering Service grid for client ctsensor
@4000000063dc161125b19ad4 INFO:device_service_config:About to open config file
@4000000063dc16112e44eebc INFO:device_service:Unregistered mqtt_ctsensor_L1 from dbus
@4000000063dc16120f1c474c INFO:device_service:Registering service grid for client ctsensor at path com.victronenergy.grid.mqtt_ctsensor_L1
@4000000063dc16121233c07c INFO:settingsdevice:Setting /Settings/Devices/mqtt_ctsensor_L1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000063dc161217c1a464 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.grid.mqtt_ctsensor_L1
@4000000063dc161218a6b11c INFO:device_service:Registered Service com.victronenergy.grid.mqtt_ctsensor_L1 under DeviceInstance 1
@4000000063dc161218d81d74 INFO:device_manager:publish {'portalId': '102c6b64af99', 'deviceInstance': {'L1': '1'}} to device/ctsensor/DBus, status is 0
@4000000063dc163d19a44ee4 INFO:device_manager:Received device status message {'clientId': 'ctsensor', 'connected': 0, 'version': '1'}
@4000000063dc163d1d1f5c44 INFO:device_service:Unregistered mqtt_ctsensor_L1 from dbus
@4000000063dc163d1d1f7f6c INFO:device:Removed Service L1 from client ctsensor
@4000000063dc163d1d1f8b24 INFO:device_manager:**** Unregistering device ctsensor ****
^C
```

### MQTT Explorer:
[image]

### Mosquitto_pub:

```
❯ mosquitto_sub -h venus.local -t "device/#" -t "W/#" -v
device/ctsensor/Status {"clientId": "ctsensor", "connected": 1, "version": "1", "services": {"L1": "grid"}}
device/ctsensor/DeviceInstance {"L1": "1"}
device/ctsensor/DBus {"portalId": "102c6b64af99", "deviceInstance": {"L1": "1"}}
W/102c6b64af99/grid/1/Ac/L1/Voltage {"value": "120.2"}
W/102c6b64af99/grid/1/Ac/L2/Voltage {"value": "120.2"}
W/102c6b64af99/grid/1/Ac/L1/Current {"value": "11.98"}
W/102c6b64af99/grid/1/Ac/L2/Current {"value": "18.51"}
W/102c6b64af99/grid/1/Ac/L1/Power {"value": "1411.1"}
W/102c6b64af99/grid/1/Ac/L2/Power {"value": "2154.1"}
W/102c6b64af99/grid/1/Ac/Energy/Forward {"value": "30.12"}
W/102c6b64af99/grid/1/Ac/Power {"value": "3565"}
device/ctsensor/Status {"clientId": "ctsensor", "connected": 0, "version": "1"}
```

### Before mqtt client connects:
[image]

### After mqtt client writes to L1/L2 current/power:
[image]

freakent, 2023-02-02:
Hi @mtucker502, I've not used grid myself. If this driver is registering your device correctly then it is doing its job, the interface you are using to publish data values is Victron's own dbus-mqtt interface. 

Having said that, could the cause of your issue be the same as the one reported earlier this, posting strings instead on numeric values? You have quotes around numeric values. 

W/102c6b64af99/grid/1/Ac/L1/Voltage {"value": "120.2"}
W/102c6b64af99/grid/1/Ac/L2/Voltage {"value": "120.2"}
W/102c6b64af99/grid/1/Ac/L1/Current {"value": "11.98"}
W/102c6b64af99/grid/1/Ac/L2/Current {"value": "18.51"}

freakent, 2023-02-02:
One more thing that looks a bit odd in your log file:

INFO:device_manager:**** Unregistering device ctsensor ****

Are you publishing data after the driver has stopped or been killed? I'm not sure how the dbus will behave if your device is not connected.

mtucker502, 2023-02-03:
I had the pip install error earlier :)

This one though you were right on the money. I'm sending floats as strings. Is there a preferred method for decoding bytes from paho-mqtt into the proper type?

Currently I'm decoding as "utf-8" and for these sensor values then converting to floats. Example: `sensor_value = float(message.payload.decode("UTF-8"))`

freakent, 2023-02-03:
I couldn't tell you. As far as I know you just need to ensure your numeric values are floats or integers.

freakent, 2023-02-03:
As you found the problem was, can this issue be closed?

closed 2023-02-03 by mtucker502