freakent/dbus-mqtt-devices #39: sending data from ESP32 to cerbo gx

sheldonadjah, 2023-02-03:
Hi there, I'm working on a solar project which its battery voltage is 384VDC. i will design a esp32 board to monitor the battery voltage, power from inverter, and also start and stop generator when needed. but i will like to remotely monitor the esp32 through a cerbo gx. can you assist me ?

stundenblume, 2023-02-03:
Hi,
you can use this driver to create a new battery and write data via MQTT to that "battery device". The battery will be shown as if you use a Victron supported battery.
Sebastian

mtucker502, 2023-02-07:
> Hi there, I'm working on a solar project which its battery voltage is 384VDC. i will design a esp32 board to monitor the battery voltage, power from inverter, and also start and stop generator when needed. but i will like to remotely monitor the esp32 through a cerbo gx. can you assist me ?

Are you using anything such as esphome on the esp32?

stundenblume, 2023-02-08:
Yes, I use Tasmota. There I can provide some help.

mtucker502, 2023-02-08:
I use esphome and hopefully this config will help you get pointed in the right direction.

Special thanks to @ssieb who helped me get this working.

```yaml
web_server:
  port: 80
  include_internal: true
  local: true
  version: 2
  id: webserver

substitutions:
# Change the disp_name to something you want  
  disp_name: Cottage
# Interval of how often the power is updated
  update_time: 10s
# Current transformer calibrations:
  current_cal: '24900'   # self calibrated
# AC Transformer voltage calibration 9VAC Jameco 157041: 38302
# For meter <= v1.3:
#         42080 - 9v AC Transformer - Jameco 112336
#         32428 - 12v AC Transformer - Jameco 167151
# For meter > v1.4:
#         37106 - 9v AC Transformer - Jameco 157041
#         38302 - 9v AC Transformer - Jameco 112336
#         29462 - 12v AC Transformer - Jameco 167151
# For Meters >= v1.4 rev.3
#         3920 - 9v AC Transformer - Jameco 157041

# For Meters >= v1.4 rev3: 3920
  # voltage_cal: '38302' # reports 655.3V
  # voltage_cal: '7014'  # reports 211.1V -- should change to 3987.115111321648508
  voltage_cal: '3953'
  client_id: 'ctsensor'
  sensor_name: 'L1'

globals:
  - id: vrm_portal_id
    type: std::string
    restore_value: no
    initial_value: ''
  - id: sensor_id
    type: int
    restore_value: no
    initial_value: '0'
    
esphome:
  name: cottage-ct-main

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret api_key
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 10.52.84.100
[... 186 lines cut ...]
mqtt:
  topic_prefix: grid
  broker: 10.52.84.212
  discovery: false  # we do not need homeassistant here
  birth_message:
    topic: device/${client_id}/Status
    # topic: grid/status
    # payload: online
    payload: '{"clientId": "${client_id}", "connected": 1, "version": "1", "services": {"L1": "grid"}}'
    # retain: False
  will_message:
    topic: device/${client_id}/Status
    payload: '{"clientId": "${client_id}", "connected": 0, "version": "1"}'
  on_json_message:
    topic: device/${client_id}/DBus
    then:
      - globals.set:
          id: vrm_portal_id
          value: !lambda |-
            return x["portalId"];
      - globals.set:
          id: sensor_id
          value: !lambda |-
            return x["deviceInstance"]["${sensor_name}"];
```

mtucker502, 2023-02-16:
@stundenblume Did the esphome configuration help you?

stundenblume, 2023-02-16:
@mtucker502
I'm a happy Tasmota user. @sheldonadjah is the one who needed help.

freakent, 2023-03-03:
Can this be closed now?

closed 2023-04-14 by stundenblume

sheldonadjah, 2023-05-08:
yes this was very helpful. sorry for the late respond.  i went out of town.