freakent/dbus-mqtt-devices #41: Energy Meter missing under settings

mtucker502, 2023-02-16:
My energy meter is showing correctly under the normal device list:
[image]

I can see all of the values as expected:
[image]

But if I navigate to Settings > Energy Meters it's blank:
[image]
[image]

Is this an issue with my config or an expected behavior as the GX can't control or change any settings on my MQTT meter?

freakent, 2023-02-16:
I assume that since your energy meter is custom that it's settings are under "My Grid Meter" not on Victron's Settings screen.

stundenblume, 2023-04-14:
Even a "Victron" EM24 won't be listed there.

closed 2023-04-14 by stundenblume