freakent/dbus-mqtt-devices #42: ModuleNotFoundError: No module named 'dataclasses'

Aikawa24, 2023-02-21:
`root@einstein:/data/drivers# ./dbus-mqtt-devices-0.5.1/bin/setup.sh
Setup dbus-mqtt-devices in /data/drivers/dbus-mqtt-devices-0.5.1 started
Ensure Python's Pip is installed
pip 20.0.2 from /usr/lib/python3.8/site-packages/pip (python 3.8)
Pip install module dependencies
Collecting PyYAML==6.0
  Using cached PyYAML-6.0.tar.gz (124 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 18, in <module>
    from setuptools.dist import Distribution
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 42, in <module>
    from setuptools.config import setupcfg, pyprojecttoml
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/config/__init__.py", line 10, in <module>
    from . import setupcfg
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/config/setupcfg.py", line 18, in <module>
    from setuptools.extern.packaging.requirements import Requirement, InvalidRequirement
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/requirements.py", line 8, in <module>
    from ._parser import parse_requirement
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_parser.py", line 10, in <module>
    from ._tokenizer import DEFAULT_RULES, Tokenizer
  File "/var/volatile/tmp/pip-build-env-lhpek7qm/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_tokenizer.py", line 3, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'

Set up Victron module libraries
Set up device service to autorun on restart
Setup dbus-mqtt-devices complete`

Aikawa24, 2023-02-21:
@4000000063f445eb2f18a5f4 *** starting dbus-mqtt-devices ***
@4000000063f445ee1463b03c Traceback (most recent call last):
@4000000063f445ee1463d74c   File "/data/drivers/dbus-mqtt-devices-0.5.1/dbus_mqtt_devices.py", line 34, in <module>
@4000000063f445ee1463e6ec     from device_manager import MQTTDeviceManager
@4000000063f445ee1463eebc   File "/data/drivers/dbus-mqtt-devices-0.5.1/device_manager.py", line 17, in <module>
@4000000063f445ee1463fe5c     import yaml
@4000000063f445ee14640244 ModuleNotFoundError: No module named 'yaml'

stundenblume, 2023-02-21:
For reference: For an RasPi see [this](https://github.com/freakent/dbus-mqtt-devices/issues/40#issuecomment-1434786528) post.

Aikawa24, 2023-02-21:
OK,
after run `opkg update && opkg install python3-modules python3-pip` as suggested

`root@einstein:~# opkg update && opkg install python3-modules python3-pip
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/einstein/Packages.gz.
Updated source 'einstein'.
Installing python3-debugger (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-debugger_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-tkinter (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-tkinter_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-pydoc (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-pydoc_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-mailbox (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-mailbox_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-idle (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-idle_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-smtpd (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-smtpd_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-syslog (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-syslog_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-xmlrpc (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-xmlrpc_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-terminal (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-terminal_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-venv (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-venv_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-image (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-image_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-audio (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-audio_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-mmap (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-mmap_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-profile (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-profile_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-resource (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-resource_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libjitterentropy2 (2.2.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libjitterentropy2_2.2.0-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libsysfs2 (2.1.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libsysfs2_2.1.0-r5_cortexa7hf-neon-vfpv4.ipk.
Installing libpanelw5 (6.2) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libpanelw5_6.2-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libgdbm6 (1.18.1) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libgdbm6_1.18.1-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-doctest (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-doctest_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-pip (20.0.2) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-pip_20.0.2-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-misc (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-misc_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing rng-tools (6.9) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/rng-tools_6.9-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-curses (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-curses_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libgdbm-compat4 (1.18.1) on root
[... cut ...]
  Building wheel for PyYAML (PEP 517) ... done
  Created wheel for PyYAML: filename=PyYAML-6.0-cp38-cp38-linux_armv7l.whl size=45330 sha256=0cb040097f0bbf283b0d149b259e0d89f273cb8e8a52b44816c6574ecdb710b1
  Stored in directory: /data/home/root/.cache/pip/wheels/52/84/66/50912fd7bf1639a31758e40bd4312602e104a8eca1e0da9645
Successfully built PyYAML
Installing collected packages: PyYAML
Successfully installed PyYAML-6.0
Set up Victron module libraries
Set up device service to autorun on restart
Setup dbus-mqtt-devices complete`

closed 2023-02-21 by Aikawa24

freakent, 2023-02-21:
Thanks for confirming I will merge this change into the main branch now.