freakent/dbus-mqtt-devices #43: Setting up a "Multi" inverter

thucar, 2023-03-02:
I have successfully imported the basics of my Hybrid inverter functionality to Venus OS. But there is one thing that I have not managed to figure out.

AC In and AC Out only show up as single phase. But when I check VRM, then I can see and display all three phases separately
[image]
vs
[image]

This is what I have in my services.yml for my inverter setup:
[image]

stundenblume, 2023-03-02:
It seems you commented out L2 and L3.
You should remove that and it should be fine.

thucar, 2023-03-02:
Hi @stundenblume 
thank you for taking the time to look at this.

Those are additional AC inputs I have commented out. The inverter has more than one input for AC. All are three phase inputs and I have defined  different data types for each phase of each input like this:
[image]

stundenblume, 2023-03-02:
in service.yml you only define a "template". You will generate your inverters (instances) with the status topic.
I should not be necessary for you to modify service.yml for normal use.
Expect: you would like to add a until now unsupported device.

freakent, 2023-03-02:
As @stundenblume said, the services.yml only defines the parameters. It's the data you send to dbus-Mqtt topics that is important. You'd need to share your code for publishing messages to those topics.

thucar, 2023-03-02:
Thank you for the help with this. 
I'm sending live data to all the topics I showed above. The data is arriving correctly because I can choose to display it in a graph on VRM. All three AC phases have Power, Current, Voltage and Frequency data coming in correctly. But when I look at the Multi inverter in the Remote Console, I only see L1 values for both, AC in and AC out.

This is the registration message I use. After that it's just posting live data to all the endpoints seen in the picture above.
`{"clientId":"deye","connected":1,"version":"v1.0 ALPHA","services":{"wr":"multi"}}
`

stundenblume, 2023-03-02:
Can you please post your complete service.yml.
In which topics do you write your data?

freakent, 2023-03-02:
Also be aware that if you register a device and then subsequently comment out parameters those parameters will still exist on the dbus. So that might give strange results. Regards,Martin+44 7776490601

"target": "https://github.com/freakent/dbus-mqtt-devices/issues/43#issuecomment-1451694099",
"url": "https://github.com/freakent/dbus-mqtt-devices/issues/43#issuecomment-1451694099",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

thucar, 2023-03-02:
[services.yml.txt](https://github.com/freakent/dbus-mqtt-devices/files/10870743/services.yml.txt)

Since my services file is quite a long one, I uploaded it instead of just pasting the entire thing.

The topics I'm publishing to are:
```
W/c0619ab2fb5f/multi/2/Ac/In/1/L1/P
W/c0619ab2fb5f/multi/2/Ac/In/1/L1/I
W/c0619ab2fb5f/multi/2/Ac/In/1/L1/F
W/c0619ab2fb5f/multi/2/Ac/In/1/L1/V
W/c0619ab2fb5f/multi/2/Ac/In/1/L2/P
W/c0619ab2fb5f/multi/2/Ac/In/1/L2/I
W/c0619ab2fb5f/multi/2/Ac/In/1/L2/F
W/c0619ab2fb5f/multi/2/Ac/In/1/L2/V
W/c0619ab2fb5f/multi/2/Ac/In/1/L3/P
W/c0619ab2fb5f/multi/2/Ac/In/1/L3/I
W/c0619ab2fb5f/multi/2/Ac/In/1/L3/F
W/c0619ab2fb5f/multi/2/Ac/In/1/L3/V
```

and for AC Out:
```
W/c0619ab2fb5f/multi/2/Ac/Out/L1/I
W/c0619ab2fb5f/multi/2/Ac/Out/L1/P
W/c0619ab2fb5f/multi/2/Ac/Out/L1/V
W/c0619ab2fb5f/multi/2/Ac/Out/L1/F
W/c0619ab2fb5f/multi/2/Ac/Out/L2/I
W/c0619ab2fb5f/multi/2/Ac/Out/L2/P
W/c0619ab2fb5f/multi/2/Ac/Out/L2/V
W/c0619ab2fb5f/multi/2/Ac/Out/L2/F
W/c0619ab2fb5f/multi/2/Ac/Out/L3/I
W/c0619ab2fb5f/multi/2/Ac/Out/L3/P
W/c0619ab2fb5f/multi/2/Ac/Out/L3/V
W/c0619ab2fb5f/multi/2/Ac/Out/L3/F

```

thucar, 2023-03-02:
@freakent I noticed that. So I've resorted to rebooting the Cerbo device every time I make any changes to the services file

freakent, 2023-03-02:
Rebooting won't make any difference (I think). You should check what is in the dbus using dbus-spy to be sure.
https://github.com/victronenergy/dbus-spy

thucar, 2023-03-02:
@freakent thanks for that tip. dbus-spy lets me understand better what's happening. For some reason the AC/NumberOfPhases is showing up as 0 and it is not possible to edit that value. No matter what value I enter, it stays 0. What could be the cause of such behavior? 

The data for all three phases is coming in nicely and shows up on dbus.
[image]

freakent, 2023-03-02:
The usual problem we see is sending numeric values as json strings instead of numbers.

freakent, 2023-03-02:
Have you checked the register list? https://www.victronenergy.com/upload/documents/CCGX-Modbus-TCP-register-list-2.90.xlsx

AC/NumberOfPhases is read-only by default. Have you made it writable?

thucar, 2023-03-03:
@freakent I do not need it to be writable. I thought that just defining 3 phases in the services.yml for my template was enough?

Exploring with dbus-spy I see that under com.victronenergy.multi.mqtt_deye_wr the number of phases is correctly displayed as 3. Also I see the measurement data for all three phases in and out.

But when I look under com.victronenergy.system then the number of phases for all AC inputs and consumptions is 1 and I only see L1 measurement values.

What controls this?

freakent, 2023-03-03:
That will be something in the Venus OS. This driver does not perform any additional processing after registering the device. Once you are sending data values you are talking directly to Victron's dbus-mqtt interface. Support for that is provided through the modifications space in the Victron Community forum.

thucar, 2023-03-05:
Turns out "multi" service is not (apparently?) supposed to get picked up as having three phase support. I was recommended to use vebus instead. 
With vebus I have all my AC and DC functionality covered and working correctly. I absolutely love this driver.

Now I'm in a bit of a pickle with PV stuff. Since vebus can not handle PV side, I tried adding a template for "solarcharger" but as soon as I add it, even if it does not contain anything other than ProductId, the system service starts crashing if I register a device using this service.

Is there a log file or live output I can tap into to monitor the com.victronenergy.system service?

thucar, 2023-03-06:
This looks odd to me. Two of the services I register, show up with the same VRM instance. Is that supposed to happen?

`{"wr":"2","nonessential":"3","essential":"4","solar":"4"}`

freakent, 2023-03-06:
It depends if they are different service types. DeviceIds are unique within each service type. Please show the registration payload.

freakent, 2023-05-03:
No activity for almost 2 months so closing this issue

closed 2023-05-03 by freakent

thucar, 2023-05-04:
Sorry, forgot to update this ticket. It was indeed a question of VenusOS design.