freakent/dbus-mqtt-devices #44: gi.repository missing

williamluke4, 2023-03-03:
Ran into an issue when installing this when my Cerbo was on v2.61 I believe this is due to the transition to Python 3. 

So If you encounter issues when installing this maybe try updating your Cerbo to the latest version then try again

freakent, 2023-03-03:
Can you provide more information?  I don't recognise gi.repository. Are you saying you had this driver running on Venus 2.61 but then upgraded Venus and then it stopped working? 
What version did you upgrade to?
What was the error? Was it at run time or installation?

williamluke4, 2023-03-03:
Sorry I was meant to close the issue after sign posting

closed 2023-03-03 by williamluke4

freakent, 2023-03-03:
There is insufficient information to describe this as signposting, I am going to have to delete this.