freakent/dbus-mqtt-devices #47: Temperature sensors not appearing in my VRM Device list

gharshaw, 2023-03-07:
Is it just my account or should the temperature sensors be appearing in the VRM device list as they do show on the VRM dashboard?

freakent, 2023-03-07:
Please case you provide more information and some screenshots?

gharshaw, 2023-03-07:
My temp and tank sensors that are published to the dbus appear correctly on the VRM dashboard however only the tanks appear on the VRM device list.
[image]

[image]

I am unable to remove the temp sensors from VRM since they are not on the device list. 
I'm curious if this is the case for all using this device driver

Regards
Gene

freakent, 2023-03-07:
You are correct. I just checked my test set up where I have ruuvi tags and a the temperature simulator running. The device list only shows the Ruuvi tags. 

[image]

[image]

I'm not sure exactly why that is. There are some differences in the dbus values but I can't see anything obvious. I could be that VRM is explicityly looking for the name Ruuvi somewhere.

[image]

[image]

At the end of the day, I can't see that this is really that useful. You **should** be sending a disconnect (connect=0) with a Last Will message when the device disconnects from MQTT.  The GX device will then spot disconnected devices and give you the option of removing them.

freakent, 2023-03-07:
dbus whilst devices are connected over MQTT

[image]

dbus after devices have disconnected and Last Will message sent

[image]

gharshaw, 2023-03-07:
As you see in my screenshot my IN temp sensor states it has not updated in 5 days although it is updating on my CCGX. I think it might be corrupted in VRM. Do you think sending the last will message will remove it from VRM and on a new connection will recreate it on VRM?

freakent, 2023-03-07:
I would use dbus-spy to see what is in the dbus. You might find it easier to clean up the dbus by hand.

But you should ensure you have a Last Will message set in your code.

gharshaw, 2023-03-07:
I am using node-red. I just shut it down cleanly and it did not send a last will or disconnect. According to node-red documents it should have. Do you have any experience with the node-red mqtt nodes?

Regards

freakent, 2023-03-07:
I do know NodeRed well (see my other popular project) although it's been a while so I have to rely on my memory. 

The last will message is sent by the client (in this case NodeRed) to the MQTT broker when the client **connects**. If the client disconnects unexpectedly then the **broker** sends the last will message to the topic it was intended for. 

There is a good article here:

http://stevesnoderedguide.com/configuring-the-mqtt-publish-node

Also take a look at the samples in the test-data folder. They are written in python but you should be able to understand the Mqtt last will set up.

gharshaw, 2023-03-07:
Thanks for the additional info. I have the lastwill and disconnect working now by  updating  the close and will message configuration in  the NR MQTT broker config. After making this change and forcing a close it seems to have corrected the VRM update issue

[image]

closed 2023-04-14 by stundenblume