freakent/dbus-mqtt-devices #48: Why are some lines in services.yml for "grid" commented?

stundenblume, 2023-03-22:
Hi Martin,
do you know why the [lines](https://github.com/freakent/dbus-mqtt-devices/blob/main/services.yml#L156) in services.yml for grid are commented?
I have a very small installation (HM-600 / AhoyDTU, venusOS, Smartmeter / Tasmota) for a friend and would like to emulate a gridmeter. It seems to work, but the visualisation in VRM isn't nice.

thx
Sebastian

[image]

stundenblume, 2023-03-22:
... for example setting the "role" to "grid" wouldn't be possible with the current version of the service.yml.

freakent, 2023-03-22:
Hi Sebastian,
I'm sorry I don't know whey those lines are commented. I was rely heavily on community with these device settings. 

I know I keep saying, but once other projects have quietened I am planning a update to the way servies.yml works, that will allow multiple configuration for the same service type. Just need to find the time!

stundenblume, 2023-03-22:
I had to set `N/.../system/0/Ac/ActiveIn/Source` to `1` it was `null`

derFrickler, 2023-04-20:
> I had to set `N/.../system/0/Ac/ActiveIn/Source` to `1` it was `null`

Can you tell what exactly you did?

stundenblume, 2023-04-20:
Write via MQTT `{"value": 1}` to the topic `N/.../system/0/Ac/ActiveIn/Source`
But it will be set back to `null` after some time.

klein0r, 2023-12-19:
> But it will be set back to `null` after some time.

This just happens on a Raspberry Pi when the USB-Adapter (MK-3 USB) is not plugged in (in my case). But I still have the problem that I just have "AC Input" and no "Grid".

[image]

I thought I've had to set the role to `grid` (like @stundenblume already mentioned).

https://github.com/freakent/dbus-mqtt-devices/blob/c46ee10e5169994c407fb4574981f96c33c03b8b/services.yml#L156-L161

But nothing changes when setting `Role` to `"grid"`. Any idea?

Venus OS 3.13