freakent/dbus-mqtt-devices #50: possible to add services like ac-charger and dc-system?

kr0815, 2023-04-16:
I tuned my Victron ESS with
- an additional charger to load the batteries faster when there is much sun
- an additional small 300W-Inverter with better efficiency

No i want to have them in the Venus-GUI To get correct calculations of the house consumption

I guess i have to adapt services.yml?

Anybody did this before?

What is the right service to show up in the main GUI site - only very few are possible?

I think the best would be for my purpose a DC-System?

stundenblume, 2023-04-16:
> an additional small 300W-Inverter with better efficiency

pvinverter is the right service

> an additional charger to load the batteries faster when there is much sun

https://github.com/victronenergy/venus/wiki/dbus#solar-chargers
or
https://github.com/victronenergy/venus/wiki/dbus#ac-chargers

Both are not defined in[ services.yml](https://github.com/freakent/dbus-mqtt-devices/blob/main/services.yml) right now but can be added easily .

kr0815, 2023-04-16:
PVinverter would be AC, i think i need solarcharger?

But i don´t understand the whole system
Example- i try to fake a genset
it´s on DBus, i see it in Venus as a sensor, but i don´t see a Generator in the GUI?
Is there a special product ID needed?

[image]
[image]
[image]

stundenblume, 2023-04-16:
With this driver I just have experience with pvinverter, gridmeter and temperature.

I think you already got it working and push data into the right MQTT topics, right?

The only thing missing is the display in VRM Portal?

kr0815, 2023-04-16:
Yes, the Data is pushed in the right topic
I tried vith various programs, but never got it showed up in the GUI

My problem: if the additional Battery-connected inverter pushes 300W to the house, Venus doesn´t calculate with that for house consumption

I need to "fake" a PV-Charger like in the attached Picture

[image]

stundenblume, 2023-04-17:
Than you have to define a [solar-charger](https://github.com/victronenergy/venus/wiki/dbus#solar-chargers) in [services.yml](https://github.com/freakent/dbus-mqtt-devices/blob/main/services.yml).

closed 2023-07-25 by freakent