freakent/dbus-mqtt-devices #51: Display total solar is missing - Adding SMA Inverter

phoenix-83, 2023-04-27:
Hello
I have a question about the integration of a SMA inverter.
It is a Tripower X 20-50, which is unfortunately not automatically detected by Venus OS. That is why I have inserted the values via mqtt.
This also works so far only the solar yield is not displayed in the VRM. The following (red marked) parts are missing:

[image]

First of all: I did not make any adjustments in the services.yml. 

Does anyone have an idea how I can include solar generation in the VRM portal?
Without the solar generation, unfortunately, the consumption values are not displayed correctly:
[image]

stundenblume, 2023-04-27:
Is this the topic you are looking for?: `W/.../pvinverter/6/Ac/Energy/Forward`

phoenix-83, 2023-04-28:
i hope this topic is correct. 
The current problem is that the firmware of the Tripower X does not yet provide this value via Modbus. Here a firmware update must take place first. let's see when this happens...

stundenblume, 2023-04-28:
Maybe you can calculate it by yourself or use a different source?
Everything would be better than nothing.

freakent, 2023-05-03:
As this isn't directly related to the driver I am going to close this issue now.

closed 2023-05-03 by freakent