freakent/dbus-mqtt-devices #52: How to remove an old sensor from VRM?

Ennar1991, 2023-05-17:
Hi, thank you for providing this code, enabling us to easily integrate MQTT sensors into VRM installations. It is currently running very smoothly on a Raspberry Pi in a camper installation.

I have abused the implementation somewhat, by skipping the MQTT registration entirely on my sensor (to enable deep sleep and maximize battery life). The registration and formatting to the portal ID is now done in Node Red; my sensor only publishes its own data via ESP Now.

Having worked recently on the installation with a colleague (it's not mine) we have swapped out the sensor in favor of another one. During this, i have changed the sensor IDs/names as well, requiring new topics. However, even after posting a "connected":0 key/value to the old topic, it didn't go away from the VRM portal. It is still visible and cannot be removed by the device list, since the devices through this driver do not appear there:

[image]

The VRM instance, "1" is not handed out to newer devices so some part of the system must still know that the device still exists.
Is there a possibility to remove the device remotely without SSH? It will be a few days before I can get back to the unit physically and enter it through SSH.

Best regards

freakent, 2023-05-17:
You need to see what is on the GX device not VRM. VRM is delayed by time and only displays what the GX device has collected. You should be able to access the GX console remotely from within VRM. When a device is disconnected the GX device should show an option to remove disconnected devices.

freakent, 2023-05-17:
By the way, this is how Venus OS works and not a defect in the driver. For example, I have a set of 4 Ruuvi tags that were connected to my CCGX via Bluetooth until this week. They still show up in VRM despite not being connected.

Ennar1991, 2023-05-17:
Thank you for your response.Despite sending a disconnect tag and removing the device from its internal list after disconnecting it virtually, it no longer exists in the internal list. It only shows up in the online VRM portal. It seems to be a bug there - I will watch it for a few days and then reach out to Victron.