freakent/dbus-mqtt-devices #54: Detect device failure

andre-meyer82, 2023-06-16:
Hi,
I simulate a Grid meter with an Shelly 3EM and everything is working fine.
The Shelly is polled using HTPP request.
But when stop polling the Shelly, the last valid values are still used by the Venus OS and operating continues.
Is there a possibilty to detect such an issue?
I would expect that the ESS stops working like with an original EM24 grid meter,

freakent, 2023-06-16:
Hi, please supply code and log files. Check the troubleshooting section of the README.

andre-meyer82, 2023-06-21:
Hi, am not sure what exactly do you need.
My question was more a general behaviour of the driver.
What I have figured out that if I recognize that the source of data is not responding (Shelly 3EM) I set the "Connected" state tp "0" of this client. Then this is recognized by the Venus OS and the system stops working.

freakent, 2023-06-21:
There isn't enough information here to understand what you are trying to do or the issue you are encountering. Often sharing your code is easier that trying to describe it. Please read the troubleshooting section of the readme. There is also a sample Arduino project that may help you understand how to use the driver. 

Please be aware, this driver only handles the registering of a custom or 3rd party device to the Venus OS DBUS, any data values you read or write is handled by Victron's dbus-Mqtt service.

closed 2023-07-25 by freakent