freakent/dbus-mqtt-devices #55: Q: will it future work with dbus-flashmq?

mdkeil, 2023-06-26:
[#1098](https://github.com/victronenergy/venus/issues/1098)

freakent, 2023-06-26:
Thanks for sharing this. This will have implications for the driver as it is uses the Victron MQTT libraries. I will monitor Victron's project and plan accordingly. Watch this space.

freakent, 2023-07-06:
Please provide a log file showing any errors.

mpvader, 2023-07-07:
Hi guys, the change from Mosquitto to FlashMQ includes deprecating dbus-mqtt. See our issue linked above 1098, for details.

freakent, 2023-07-07:
@mpvader, Thanks for confirmation but are you Deprecating or removing? Deprecating would mean it should continue to work for a while longer.

mpvader, 2023-07-07:
ah thanks. Removing.

There only one service can listen to a port. So what we did per v3.10~12, which is change who listens to the local websockets port from mosquitto to flashmq, is what we'll be doing more coming time.

mpvader, 2023-07-07:
ps. note that our MQTT API doesn't change much. There are some very nice improvements, especially related to the keepalive and supressing a resend of all topics; and otherwise all stays the same. Its just faster, and causes less CPU load on the GX device.

And is done via FlashMQ, which is developed by one of our own core team members in his spare time - which is maybe business wise not that important but personally I'm quite proud of that/him.!

mdkeil, 2023-07-16:
Any new progess-info for an separate flashMQ-version?

freakent, 2023-07-16:
It's unlikely I'll be able to make any significant updates for a. Few months. If you have already experienced an error please share log file, as requested previously.

mdkeil, 2023-08-16:
> If you have already experienced an error please share log file, as requested previously.

```
@4000000064dd15ca36b691a4 *** starting dbus-mqtt-devices ***
@4000000064dd15cd065e0eec Traceback (most recent call last):
@4000000064dd15cd065e35fc   File "/data/drivers/dbus-mqtt-devices-0.6.2/dbus_mqtt_devices.py", line 34, in <module>
@4000000064dd15cd065e5154     from device_manager import MQTTDeviceManager
@4000000064dd15cd065e60f4   File "/data/drivers/dbus-mqtt-devices-0.6.2/device_manager.py", line 17, in <module>
@4000000064dd15cd065e747c     import yaml
```

freakent, 2023-08-16:
That error appears to be the problem related to the missing YAML library and not specifically the new version of Venus OS.

mdkeil, 2023-08-16:
with VenusOS v3.0 it works without any error.

freakent, 2023-08-17:
That's as maybe but others are also reporting this issue with the missing PYYaml library on Venus 3.0. PYYaml is not included by Victron in any version.

mdkeil, 2023-08-28:
> We're preparing v3.10 for release within a few weeks. In that release, we'll be shipping Mosquitto, not FlashMQ. In other words, all changes so far will be taken out for that release.
> 
> FlashMQ will be included again in v3.20.

mpvader, 2023-09-21:
Hi all, there are now people thinking that there are issues with running this cope (dbus-mqtt-devices) in combination with FlashMQ.

See above link by disaster123 for an example.

Having read the thread above, this is all just a suspicion and vague.

I don't see any problem related to FlashMQ. And, note that I also don't expect any problem.

mpvader, 2023-09-21:
@freakent , am I right in concluding that this repo is expected to keep working fine as long as our dbus-mqtt project is installed?

What/where is the depedency of dbus-mqtt-devices on dbus-mqtt?

freakent, 2023-09-21:
Hi @mpvader , thanks for keeping an eye on this project :-)

I am not aware of any issue using this driver with FlashMQ, although I intend to do more testing. 

This driver has had an issue recently with a PyYaml dependency that started roughly the same time as the Venus 3.10 Beta. Some users wrongly assumed the issue was caused by Venus 3.10, when in fact it was an outdated dependency on PyYaml. This issue has now been addressed in the latest relesse. In the longer term I am working on a new version that removes the PyYaml dependency completely.

The driver does today have a minor dependency on dbus-mqtt. To simplify coding my MQTTDeviceManager class extends the MqttGObjectBridge class in the dbus-mqtt module. During installation, the setup script creates a soft link from /opt/victronenergy/dbus-mqtt to include this module in the python module search path for the driver. In the future I could either just include the source for this class directly in my own source tree or write my own version. 

Would you consider including the functionality of this driver in VenusOS directly? It would be more efficient to have a single driver for any external MQTT devices instead of one for each, when each driver more or less does the same thing? 

Thanks again.

gharshaw, 2023-11-03:
@freakent , Curious if you will be testing against 3.20~17 now that it is released with the new html5 app and flashmq? I have it installed and tanks do not show up correctly in flashmq on port 1884  but they appear correctly in mosquito mqtt on port 1883. The tanks do not show up in the new html5 app.  The new html5
 app is using local port 9001 which I believe is for flashmq.  The tanks appear correct in dbus when viewing with dbus-spy.
I'm using your 0.6.2 driver on a CCGX.
Best Regards,
Gene

freakent, 2023-11-03:
Thanks for the heads up. I'm juggling lots of projects at the moment, but I'll try to take a look next week. 

It's strange that you don't see it in the new HTML app. Does it appear in VRM?

gharshaw, 2023-11-03:
Yes. Tanks and Temps show correctly in VRM

Thanks!!

gharshaw, 2023-11-08:
@freakent  I believe I discovered the issue. In short, the device_service.py was setting the dbus path "/DeviceInstance  as a string value. I edited your code to cast it as an integer --  dbus_service.add_path('/DeviceInstance', int(self.device_instance))
 and now all my devices appear correctly in flashmq  and subsequently appeared in the html5 app and the new gui-beta app.
The following screenshots show how I discovered it. 
The 1st screenshot is when I was running my CCGX in demo mode. Tanks where populating correctly. The 2nd screenshot is with my tanks via the dbus-mqtt-devices driver. I noticed that the DeviceInstance was encapsulated in quotes when using the dbus-mqtt-devices driver

[image]

[image]

And while browsing flashmq on port 1884. dbus-flashmq was  indexing the devices incorrectly as it would only publish one of the devices with an mqqt path of 0 when the DeviceInstance was 4 as seen in this screenshot.

[image]

Gui-Beta
[image]

HTML5
[image]

Hope this helps

All the best
Gene

freakent, 2023-11-08:
@gharshaw Gene, thanks for the info but what version of the code is this in?  Please can you post the block of code you changed?

gharshaw, 2023-11-08:
@freakent I'm using your 0.6.2 driver on  CCGX v3.20~17.  Edited file device_service.py line 82
[image]