freakent/dbus-mqtt-devices #57: Installation error on RPi

Becker884, 2023-07-29:
since now I can´t install 0.6.2 on RPi with Venus 3.00 or 3.01:

```
root@raspberrypi4:/data/drivers# ./dbus-mqtt-devices-0.6.2/bin/setup.sh
dbus-mqtt-devices: Setup in /data/drivers/dbus-mqtt-devices-0.6.2 started
dbus-mqtt-devices: Checking to see if Python's Pip is installed
/usr/bin/python: No module named pip
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/raspberrypi4/Packages.gz.
Updated source 'raspberrypi4'.
Installing python3-debugger (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-debugger_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-tkinter (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-tkinter_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-mailbox (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-mailbox_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-idle (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-idle_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-db (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-db_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-smtpd (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-smtpd_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-syslog (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-syslog_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-curses (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-curses_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-xmlrpc (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-xmlrpc_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-terminal (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-terminal_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-venv (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-venv_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-image (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-image_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-mmap (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-mmap_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-profile (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-profile_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-resource (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-resource_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libjitterentropy2 (2.2.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libjitterentropy2_2.2.0-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libsysfs2 (2.1.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libsysfs2_2.1.0-r5_cortexa7hf-neon-vfpv4.ipk.
Installing python3-doctest (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-doctest_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-pip (20.0.2) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-pip_20.0.2-r0_cortexa7hf-neon-vfpv4.ipk.
Installing rng-tools (6.9) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/rng-tools_6.9-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-modules (3.8.13) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-modules_3.8.13-r0_cortexa7hf-neon-vfpv4.ipk.
Configuring python3-debugger.
Configuring python3-tkinter.
Configuring libjitterentropy2.
Configuring libsysfs2.
Configuring rng-tools.
 Adding system startup for /etc/init.d/rng-tools.
Starting random number generator daemon.
Configuring python3-mailbox.
Configuring python3-idle.
Configuring python3-db.
Configuring python3-smtpd.
Configuring python3-syslog.
Configuring python3-curses.
Configuring python3-xmlrpc.
Configuring python3-terminal.
[... 52 lines cut ...]
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/command/egg_info.py", line 551, in run
      self.add_defaults()
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/command/egg_info.py", line 589, in add_defaults
      sdist.add_defaults(self)
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/command/sdist.py", line 104, in add_defaults
      super().add_defaults()
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/sdist.py", line 251, in add_defaults
      self._add_defaults_ext()
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/sdist.py", line 336, in _add_defaults_ext
      self.filelist.extend(build_ext.get_source_files())
    File "<string>", line 204, in get_source_files
    File "/var/volatile/tmp/pip-build-env-97kyqpxl/overlay/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
      raise AttributeError(attr)
  AttributeError: cython_sources
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpep9wokuu Check the logs for full command output.
dbus-mqtt-devices: Set up Victron module libraries
dbus-mqtt-devices: Set up device service to autorun on restart
dbus-mqtt-devices: Adding device service to /data/rc.local
dbus-mqtt-devices: Setup complete
root@raspberrypi4:/data/drivers#

```

Please try yourself and confirm

freakent, 2023-07-30:
Have you tried registering a service with the driver? Is the driver running despite this error?

Becker884, 2023-07-30:
of course I tried and it is not running!
only with the workout I described https://github.com/freakent/dbus-mqtt-devices/issues/56#issuecomment-1656708500

freakent, 2023-09-17:
See release v0.6.3

closed 2023-09-17 by freakent