freakent/dbus-mqtt-devices #58: Rpi Install Error

gjonesy84, 2023-08-14:
Hi,

Having issues. Can get the pv inverter working. 

Followed the debug section at bottom of page.

root@raspberrypi4:~# svstat /service/dbus-mqtt-devices
/service/dbus-mqtt-devices: up (pid 10729) 0 seconds

I take this means its up and running?

something doesnt look quite right. Any ideas what it could be?

root@raspberrypi4:~# more /var/log/dbus-mqtt-devices/current
@4000000064da8f1d21748594 *** starting dbus-mqtt-devices ***
@4000000064da8f1e1137a184 Traceback (most recent call last):
@4000000064da8f1e1137b50c   File "/data/drivers/dbus-mqtt-devices-0.6.2/dbus_mqt                                                                                        t_devices.py", line 34, in <module>
@4000000064da8f1e1137c4ac     from device_manager import MQTTDeviceManager
@4000000064da8f1e1137d064   File "/data/drivers/dbus-mqtt-devices-0.6.2/device_m                                                                                        anager.py", line 17, in <module>
@4000000064da8f1e1137e004     import yaml
@4000000064da8f1e1137e7d4 ModuleNotFoundError: No module named 'yaml'
@4000000064da8f1e22efac14 *** starting dbus-mqtt-devices ***
@4000000064da8f1f12a91b74 Traceback (most recent call last):
@4000000064da8f1f12a92b14   File "/data/drivers/dbus-mqtt-devices-0.6.2/dbus_mqt                                                                                        t_devices.py", line 34, in <module>
@4000000064da8f1f12a93ab4     from device_manager import MQTTDeviceManager
@4000000064da8f1f12a9466c   File "/data/drivers/dbus-mqtt-devices-0.6.2/device_m                                                                                        anager.py", line 17, in <module>
@4000000064da8f1f12a9560c     import yaml
@4000000064da8f1f12a95ddc ModuleNotFoundError: No module named 'yaml'
@4000000064da8f1f24524184 *** starting dbus-mqtt-devices ***
@4000000064da8f20182a2e1c Traceback (most recent call last):
@4000000064da8f20182a3dbc   File "/data/drivers/dbus-mqtt-devices-0.6.2/dbus_mqt                                                                                        t_devices.py", line 34, in <module>
@4000000064da8f20182a4d5c     from device_manager import MQTTDeviceManager
--More-- (25% of 5441 bytes)

Any help would be much aprreciated.

:)

freakent, 2023-08-14:
Thanks for including the log, that is very helpful.  It looks like you are having same problem as others. For some reason I have not been able to work out, the PyYaml library is not being installed during the setup. The service is continuously restarting automatically. Did you see any errors when you run the setup.sh?

You could try running this from the command line:
```opkg update && opkg install python3-modules```

gjonesy84, 2023-08-14:
Hi Thanks for your fast response.

Did you see any errors when you run the setup.sh? Im not sure. I didnt ready through it all. I can delete the driver and run again and see what errors(if any) comes back with?

just ran command and came up with this.

root@raspberrypi4:~# opkg update && opkg install python3-modules
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/raspberrypi4/Packages.gz.
Updated source 'raspberrypi4'.
No packages installed or removed.
root@raspberrypi4:~#

then tried more /var/log/dbus-mqtt-devices/current

root@raspberrypi4:~# more /var/log/dbus-mqtt-devices/current
@4000000064da95210e573de4 *** starting dbus-mqtt-devices ***
@4000000064da95213a869cd4 Traceback (most recent call last):
@4000000064da95213a86b05c   File "/data/drivers/dbus-mqtt-devices-0.6.2/dbus_mqtt_devices.py", line 34, in <module>
@4000000064da95213a86bffc     from device_manager import MQTTDeviceManager
@4000000064da95213a86cbb4   File "/data/drivers/dbus-mqtt-devices-0.6.2/device_manager.py", line 17, in <module>
@4000000064da95213a86d76c     import yaml

came back with the same :(

freakent, 2023-09-17:
See release v0.6.3

closed 2023-09-17 by freakent