freakent/dbus-mqtt-devices #61: I did the installation work around but still doesn't work

Zahnpasta83, 2023-08-25:
Im trying to get it running together with the ev charing integration but nothing shows up in Venus OS. Debug shows only

[image]

freakent, 2023-08-25:
There are no errors in the log but it does look like the driver is repeatedly starting and stopping. What device and version of Venus is is this on?

Zahnpasta83, 2023-08-25:
Raspi Version 3.0.0

freakent, 2023-08-25:
> @4000000064843f82683757c INFO: device_manager: Received device status message 0.0

Should be a 1

Zahnpasta83, 2023-08-25:
Now something changed but still don't know why it says 0.0 i don't send any 0.0 message
[image]

freakent, 2023-08-25:
Please share some client code.

Zahnpasta83, 2023-08-25:
something like this? Im just sending a 1 for test purpose, as soon as it works i would like to implement a KEBA Charger

`[
    {
        "id": "8fdecdeb81b71d1a",
        "type": "tab",
        "label": "OpenWB->EVCharger",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "28908042d1d38327",
        "type": "group",
        "z": "8fdecdeb81b71d1a",
        "style": {
            "stroke": "#999999",
            "stroke-opacity": "1",
            "fill": "none",
            "fill-opacity": "1",
            "label": true,
            "label-position": "nw",
            "color": "#a4a4a4"
        },
        "nodes": [
            "e420e324306f563b",
            "f8bf19d81175b304",
            "6a01d27bbccd3cf3",
            "ccf8ff8d6f685b7e",
            "be17803b661cc51a",
            "b2ead156ac397da8",
            "79e9327f1a38d7e6",
            "7657252e806c9e76",
            "8979393b44c7c2f2",
            "21fc6dbf3302bdb6"
        ],
        "x": 74,
        "y": 1499,
        "w": 832,
        "h": 302
    },
    {
        "id": "eb0984928a8fc539",
        "type": "mqtt out",
        "z": "8fdecdeb81b71d1a",
        "name": "",
        "topic": "W/d83add0e7060/evcharger/1/Current",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "47e105b96fc4cc38",
        "x": 1270,
        "y": 180,
        "wires": []
    },
    {
        "id": "e420e324306f563b",
        "type": "inject",
        "z": "8fdecdeb81b71d1a",
        "g": "28908042d1d38327",
        "name": "Disconnect EVCharger 1 from DBUS",
        "props": [
            {
                "p": "payload"
            }
        ],
[... 1215 lines cut ...]
        "broker": "<replacewithyouropenWBBroker>",
        "port": "1883",
        "tls": "",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]`

freakent, 2023-08-25:
Are there any retained messages hanging around in MQTT? Have you checked everything with MQTT Explorer?

Zahnpasta83, 2023-08-25:
After 20 minutes it shows up in Venus OS but says not connected and except the device status i can see no data

[image]
[image]
[image]

Zahnpasta83, 2023-08-25:
[image]

Zahnpasta83, 2023-08-25:
And im still getting this dbus mqtt start up loop

freakent, 2023-08-25:
We need to solve one problem at a time. Are you still getting those messages that device manager received a status of 0.0?

freakent, 2023-08-25:
> After 20 minutes it shows up in Venus OS but says not connected and except the device status i cant see any data

Venus will say the device is Not Connected if the driver or the client is not connected to MQTT or not running.

freakent, 2023-08-25:
> something like this? Im just sending a 1 for test purpose, as soon as it works i would like to implement a KEBA Charger

As it's NodeRed I'd appreciate a screen shot of the flow. It's hard to visualise a few hundred lines of Json in my head.

Zahnpasta83, 2023-08-25:
> We need to solve one problem at a time. Are you still getting those messages that device manager received a status of 0.0?

[image]

Zahnpasta83, 2023-08-25:
[image]

Zahnpasta83, 2023-08-25:
Oops now it works, i accidentally sent the disconnection value which i removed right before i sent this picture

freakent, 2023-08-25:
So is everything working now? Can we close this issue?

Zahnpasta83, 2023-08-26:
Yes thank you very much

closed 2023-08-26 by freakent