freakent/dbus-mqtt-devices #62: New Release: 0.7.4-beta

freakent, 2023-08-26:
Dear Al, I have made a new release of the dbus-mqtt-driver that completely removes the YAML dependency from this project. I have switched to using ini files which are natively supported by Python. This feels like quite a substantial change to the internals of the driver, so I am releasing this first version without YAML as a BETA. I have performed a lot of testing myself but I would like to collect feedback from users before making a final release. 

To upgrade just follow the install instructions in the readme as before. 

If you try this release, please comment below whether it works successfully for you or not.

freakent, 2023-08-26:
This release addresses issues #55, #56 and #57.

soulraven, 2023-08-27:
something is wrong i guess, i have this payload on "device/serverRoom/Status ":

{
    "clientId": "serverRoom",
    "connected": 1,
    "version": "v0.7.4-Beta",
    "services": {
        "t1": "temperature"
    }
}

but i get:
``` @4000000064eb42c8267b91f4 INFO:device_manager:[Connected] Result code 0
@4000000064eb42d006c17054 INFO:device_manager:Received device status message {'clientId': 'serverRoom', 'connected': 1, 'version': 'v0.7.4-Beta', 'services': {'t1': 'temperature'}}
@4000000064eb42d006f8ef5c WARNING:device_manager:status.service type temperature is not supported, please check services.yml

```

freakent, 2023-08-27:
@soulraven Thanks for reporting that. New beta version 0.7.5-beta available.

soulraven, 2023-08-27:
thx, now is working, but some things are not setup,
version, and device name using instance ID.

also, is it possible to setup the "Name" directly from device topic?

[image]

freakent, 2023-08-27:
Sorry, can you clarify
>thx, now is working, but some things are not setup,version, and device name using instance ID.

soulraven, 2023-08-27:
> Sorry, can you clarify
> 
> > thx, now is working, but some things are not setup, version, and device name using instance ID.

the Device name is not set also the Firmware version. What i have see in code Firmware Versions contains the value from payload "version" and the device name is the "client id"
In log seams that

```INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_ha_sonoff_th_server_room/CustomName does not exist yet or must be adjusted```
somehow is not setup the costumName

Also will be great to have the possibility to setup directly from topic the "Name" of the sensor

freakent, 2023-08-27:
This has me puzzled, it's not an area in the code that was changed. I checked in MQTT explorer what is being returned by Venus and it does include these values:

[image]
although they do not show in the UI
[image]

I am testing on pre-release Venus 3.1-32, so could be a venus bug.  Which version of Venus OS are you testing on?

freakent, 2023-08-27:
> `INFO:settingsdevice:Setting /Settings/MqttDevices/mqtt_ha_sonoff_th_server_room/CustomName does not exist yet or must be adjusted` somehow is not setup the costumName
This is just a debug message output by the Venus dbus library, it is always output when initialised.
 
> Also will be great to have the possibility to setup directly from topic the "Name" of the sensor
I don't want to complicate the registration process. Once regiatered and you have a deviceid you can send a message to W/<portal id>/temperatire/<deviceid>/CustomName to change the name programmatically.  You can do this for any of the parameters you wish to set not just CustomName.

soulraven, 2023-08-27:
i have attached the printscreen with VenusOS version

[image]

Becker884, 2023-08-28:
Hi, i tried today with 3.01 on RPi2, setup takes only few seconds, my Gridmeter & PV runs:
[image]
only the " " weren't there before 
[image]

do you have that too?

freakent, 2023-08-28:
@Becker884 Thanks for trying it out. I'll update the grid.ini config file to remove the quotes in the next release.

[Ac/Energy/Forward]
# Total bought energy over all phases
description: "kWh"
format: "{} kWh"

[Ac/Energy/Reverse]
# Total sold energy over all phases
description: "kWh"
format: "{} kWh"
  
[Ac/Power]
# Total power of all phases, preferably real power
description: "W"
format: "{} W"

[Ac/Current]
description: "A AC"
format: "{} A"

Becker884, 2023-08-28:
Can I correct this on my own?
if yes -> where to find the grid.ini?

freakent, 2023-08-28:
Yes, you can. It's services/grid.ini

See https://github.com/freakent/dbus-mqtt-devices/blob/Config-Files/services/grid.ini

Becker884, 2023-08-28:
ok, but where can I find this file on my RPi2 ?

freakent, 2023-08-28:
/data/drivers/dbus-mqtt-devices-0.7.5-beta/services/grid.ini

mdkeil, 2023-08-28:
I used the beta with corrected grid.ini @venusos 3.10 ~32 without any problems for the moment.

mdkeil, 2023-08-28:
> although they do not show in the UI

same to me with an grid sensor. no firmware, Connection.. doesnt matter for me.

freakent, 2023-08-28:
> 

My concern is that it might affect how these devices are shown in VRM. I need to investigate this issue further.

gharshaw, 2023-08-28:
@freakent 
I installed on 3.10~32 CCGX today. I'm seeing some anomalies between  the old  MQTT and the new Flash-MQ. Victron has left old MQTT on port 1883  and the new FlashMQ on port 1884.  
Tanks and Temp sensors show up on old MQTT  correctly with all the pertinent  topics/ values using MQTT Explorer
[image]

  but do not show up on FlashMQ port 1884 using MQTT Explorer. 
[image]

Also looking at dbus-spy the topic/values are missing in all temp and tank sensors
[image]

Tanks and sensors are missing from the HTML app and new gui-v2 app as well. Most likely due to the missing data in Flash-MQ

Tanks and Sensors are appearing on the console
[image]
however device data is missing
[image]

Are you seeing the same issues?
I can help with logs and any quick code changes you might want to test

Best,
Gene

freakent, 2023-08-28:
@gharshaw, thanks for your investigations, that's all very interesting. I think I need to figure out if this issue is introduced by my code changes or the new FlashMQ  in Venus OS. I think I might have to downgrade my test Raspberry PI and run tests there to be sure.

gharshaw, 2023-08-28:
@freakent 
Further investigations. I was monitoring the Flash-MQ log on device registration and see that the devices appeared and then  immediately disappeared in Flash-MQ.  

@4000000064ed181437b9fbcc [2023-08-28 21:56:26] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.temperature.mqtt_nr001_t1' with owner ':1.605'
@4000000064ed18152bdb8bf4 [2023-08-28 21:56:27] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.temperature.mqtt_nr001_t2' with owner ':1.606'
@4000000064ed1816129d8e6c [2023-08-28 21:56:28] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.temperature.mqtt_nr001_t3' with owner ':1.607'
@4000000064ed18163470fed4 [2023-08-28 21:56:28] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.temperature.mqtt_nr001_t4' with owner ':1.608'
@4000000064ed18171a49b6a4 [2023-08-28 21:56:29] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.temperature.mqtt_nr001_t5' with owner ':1.609'
@4000000064ed18182328bd74 [2023-08-28 21:56:30] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.tank.mqtt_nr001_tk1' with owner ':1.610'
@4000000064ed181913e22044 [2023-08-28 21:56:31] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.tank.mqtt_nr001_tk2' with owner ':1.611'
@4000000064ed181a0ae4d374 [2023-08-28 21:56:32] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.tank.mqtt_nr001_tk3' with owner ':1.612'
@4000000064ed181a35af0cb4 [2023-08-28 21:56:32] [INFO] [OwnerChange] Service appeared: 'com.victronenergy.tank.mqtt_nr001_tk4' with owner ':1.613'
@4000000064ed181b008527e4 [2023-08-28 21:56:32] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.temperature.mqtt_nr001_t1' with owner ':1.605'
@4000000064ed181b04ca110c [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.temperature.mqtt_nr001_t2' with owner ':1.606'
@4000000064ed181b07eeb93c [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.temperature.mqtt_nr001_t3' with owner ':1.607'
@4000000064ed181b0b717374 [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.temperature.mqtt_nr001_t4' with owner ':1.608'
@4000000064ed181b0f2678ac [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.temperature.mqtt_nr001_t5' with owner ':1.609'
@4000000064ed181b1512a934 [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.tank.mqtt_nr001_tk1' with owner ':1.610'
@4000000064ed181b189f2b54 [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.tank.mqtt_nr001_tk2' with owner ':1.611'
@4000000064ed181b1c3e4efc [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.tank.mqtt_nr001_tk3' with owner ':1.612'
@4000000064ed181b201032d4 [2023-08-28 21:56:33] [INFO] [OwnerChange] Service disappeared: 'com.victronenergy.tank.mqtt_nr001_tk4' with owner ':1.613'
@4000000064ed181b2a3ef6dc [2023-08-28 21:56:33] [ERROR] Error on 'GetValue' from com.victronenergy.temperature.mqtt_nr001_t5: org.freedesktop.DBus.Error.Se

mdkeil, 2023-08-29:
Yes, but in future (VenusOS 3.20+) flashmq will completely replace mosquitto.

Thxeverybody, 2023-09-12:
Great driver, helps me a lot. Upgraded from 0.6.2 to 0.7.5 beta without any problems. Use it to emulate a pv inverter behind a shelly and a grid meter with Hichi/Tasmota.

freakent, 2023-09-15:
@Thxeverybody Thanks for the feedback, I'm glad to hear it is working for you. I don't hunk there will be another Beta release after I have done more testing.

gharshaw, 2023-10-13:
@freakent I just updated my CCGX to Venus OS v3.11 and dbus-mqtt-driver 0.7.5-beta without any problems. The setup on the CCGX went smoothly and everything appears to be working correctly.  Next hurdle, I imagine, will be with dbus-flashmq and the removal of  dbus_mqtt. 
Best Regards

freakent, 2023-10-13:
If you are using the driver in a production system I highly recommend you use the latest release not the Beta. 

I don't anticipate the move to FlashMQ is not such an issue. It will require a slight change to the installer that's all. This driver interacts with the dbus directly, not via dbus-Mqtt.

gharshaw, 2023-10-13:
@freakent Does the latest release remove the  YAML dependency? I thought the 0.7.x versions is where you are removing that dependency

freakent, 2023-10-13:
That's correct, 0.6.x uses pyyaml.

DJZorro, 2023-10-30:
Thank you for your driver,
With this I was able write to the dbus.
I picked the weatherinfo from Openweather.API.
Imported it in OpenRed.
Pulled out the information I wanted to display
Modified some date, Wind direction, wind speed, Sunrise and Sunset.
Pubished those data with MQTT and your driver to dbus.
Picked them up from dbus by with modify gps.ini
Pubished them on the screen with modifying OverviewHubEnhanced.qml

Long way but succesfull in the end:
[image]

gurkc006, 2023-12-08:
Hi!
Just a comment from me: I used 0.7.5-beta for the last weeks without any problems on VenusOS 3.20_10. I use it to emulate one 3-phase PV Inverter from an 3-phase-AC-meter using a ESP32 and also for two temperature sensors. Worked very stable the last weeks.
Today I tried upgrading to VenusOS 3.20_33. After rerunning the install.sh script everything seems to run again. BUT: the despite the fact, that the different devices seem to be created correctly, there is no data available. this might be due to the fact that 3.20_33 switched now completley to FlashMQ. So the basic device creation seems to work, but there is no data available. I checked also with dbus-spy: only the name available, all other values are not visible.
[image]
[image]
Here are the topics for device creation in MQTT Explorer:
[image]
Here is a screenshot of the remote console:
[image]
Devices are visiblen, but empty...
So maybe I was a little bit naive thinking it might acutally work out of the box?! 
Can I provide further information? Or is there a plan to update the driver because the FlashMQ-driver will be the future for VenusOS and it would be great, if the dbus-mqtt-devices would evolve to a dbus-flashMQ-devices :-)
Thanks a lot for the great work! I really appreciate it!
Christian

freakent, 2023-12-09:
Thanks for providing the results of your research. Have you tried running the current release version (not the beta) on the latest beta Venus OS?

gurkc006, 2023-12-09:
Hi freakent,
yes I know, installing the latest beta of driver and OS was maybe a little be too much :-)
I tried installing 0.6.3 release, but had no luck with installing the YAML-dependency. I followed the steps from the installation instructions (ModuleNotFoundError: No module named 'dataclasses' -> opkg install python3-modules) but this doesn't seem to help. Always getting this dataclass error. Here you can find the console output AFTER installing the python3-modules and rerunning setup.sh: [console.txt](https://github.com/freakent/dbus-mqtt-devices/files/13624569/console.txt)
So with this verion I don't get the driver to run properly. I had this behaviour earlier also with older VenusOS versions (that's the reason I switched to 0.7.x at all). 
So for the moment I suppose switching back to VenusOS 3.20_10 and dbus-mqtt-devices 0.7.5_beta (that was working properly) and hoping, that the error will be fixed soon.
Thanks

freakent, 2023-12-10:
I would prefer to figure out why python3-modules is not installing for you. What is the output from opkg install python3-modules? What does the output from opkg list-installed tell you?

gurkc006, 2023-12-10:
Sorry for the delayed answer. Had some test this morning but had to get my system running again, as it's kind of productive... Here are some files from this evening (no solar at the moment, therefore I can test a little...)
I switched to VenusOS 3.20_33 and try using dbus-mqtt-devices 0.6.3 main branch from github. Symptoms are, that the driver seems to crash around every minute, probably when it receives new "device creation messages" via mqtt.
The logfile shows:
[vOS_3.20.33_dbus-mqtt-devices0.6.3main.txt](https://github.com/freakent/dbus-mqtt-devices/files/13629227/vOS_3.20.33_dbus-mqtt-devices0.6.3main.txt)
Here are the outputs from list-installed: 
[vOS_3.20.33_list-installed.txt](https://github.com/freakent/dbus-mqtt-devices/files/13629229/vOS_3.20.33_list-installed.txt)
and python -m pip list: 
[vOS_3.20.33_pip-list.txt](https://github.com/freakent/dbus-mqtt-devices/files/13629231/vOS_3.20.33_pip-list.txt)
Don't know, if that helpes? There seem to be a YAML parse error somewhere. 
I will switch back now to VenusOS 3.20_10 with running dbus-mqtt-devices 0.7.5_beta as this is running stable for me at the moment.
Good night