freakent/dbus-mqtt-devices #63: strings everywhere

foxos42, 2023-09-19:
Hi,
nice work! I use v0.6.2 inside a docker container.

Would be great if data inside the json (int and float) are not quoted with "" so the data type is correct and not string.
For tools like node-red it will then be possible to convert the json correctly into a javascript object. And then I no longer have to cast everything manually. ;)

freakent, 2023-09-19:
Please provide a code example. I don't know which part of the registration process this refers to. Remember it is Victron's code that you actually interact with when reading and writing values, this driver just handles the registration.

foxos42, 2023-09-19:
Arg.. damn sorry wasting your time. Just wrong repo. To much tabs. Issues are here ;) : https://github.com/wasilukm/hoymiles-mqtt
By the way ;) with you project I was able realize full phase balancing with my victron system without extra physical meters, so thanks for that!

closed 2023-09-19 by foxos42