freakent/dbus-mqtt-devices #64: null not "0"

foxos42, 2023-09-19:
Hi,
nice work! I use v0.6.2 inside a docker container.

If link status is down all data will be send with a "0". This is wrong, because it is not "0". In this case data should be set to "null" or should just not be send.

I push all the data right into a TS-database and "0" is a valid value so I see voltage drops to "0" just because the WR was not reachable. But this is apparently not the reality. I have a workaround in place but "0" is just not "null". ;)

freakent, 2023-09-19:
Please provide a code example. I don't know which part of the registration process this refers to. Remember it is Victron's code that you actually interact with when reading and writing values, this driver just handles the registration.

foxos42, 2023-09-19:
Arg.. damn sorry wasting your time. Just wrong repo. To much tabs. Issues are here ;) : https://github.com/wasilukm/hoymiles-mqtt
By the way ;) with you project I was able realize full phase balancing with my victron system without extra physical meters, so thanks for that!

closed 2023-09-19 by foxos42