freakent/dbus-mqtt-devices #65: Nodered Virtual sensor

Woodpeckercz, 2023-10-16:
Hi,
first of all great work.
I would like to take some of my existing mqtt sensors, connected to my mqtt broker and home assistant.
For example my hot water tank temperature sensor and add it to vrm. 
Your driver seems like ideal solution. 
How would I go about creating a flow in nodered to create a virtual sensor/bridge over to dbus?

freakent, 2023-10-16:
H, thanks for the feedback. A number of the folks who use this driver do so with NodeRed. I don't have a suitable NodeRed installation setup at them moment to provide an example. Hopefully someone else monitoring this project can provide an example.

Woodpeckercz, 2023-10-16:
I managed to register the NodeRed "virtual" device, and it shows up in the console but with blank temperature. 
I missunderstood how the registration process works.
Now I'm trying to figure out how to update the temperature.

Woodpeckercz, 2023-10-16:
getting bad string error on json payload  {"portalId": "<vrm portal id>", deviceInstance":"t1": 5, "t2":12}

freakent, 2023-10-16:
I highly recommend you use MQTT Explorer to experiment with registration procedure before moving what you have learned to NodeRed.

freakent, 2023-10-16:
Also take a look at the samples in the test-data folder. They are written in python but should be very easy to follow.

Woodpeckercz, 2023-10-17:
thanks, I'm using mqtt explorer.
I'm strugling with converting the mqtt input into JSON string that the system will accept and change the value,
Basicly,
I need to convert this:

{"battery":83,"humidity":60.41,"last_seen":"2023-10-16T23:26:46.650Z","linkquality":69,"power_outage_count":6,"pressure":1008.9,"temperature":19.02,"voltage":2975}

into this:

{"value":19.02}