freakent/dbus-mqtt-devices #66: Add AC Load energy meter

Woodpeckercz, 2023-10-18:
Hi,

Would it be possible to add a AC load energy meter to the services? I would like to add my immersion heater as separate load to the vrm because I have a solar diverter installed. When the battery is full, the excess solar goes to hot water, and I would like to see how much energy went in.

Woodpeckercz, 2023-10-18:
So I noticed there is a config for the PV inverter that allows it to be set to AC load but is comented out.
How would I go about enabling this?
```
# https://github.com/victronenergy/venus/wiki/dbus#pv-inverters
pvinverter:
  ProductId:
    default: 45069 # 0xB00D, ESS Demo mode uses 45058
  CustomName:
    default: "My PV Inverter"
    persist: true
  ErrorCode: # https://www.victronenergy.com/live/mppt-error-codes
    description: "0=No Error"
    default: 0 # not showing in in console???
  Position:
    description: "0=AC input 1; 1=AC output; 2=AC input 2"
    persist: true
    default: 0
    min: 0
    max: 2
  StatusCode: # not used by ESS demo mode pvinverter
    description: "0=Startup 0; 1=Startup 1; 2=Startup 2; 3=Startup 3; 4=Startup 4; 5=Startup 5; 6=Startup 6; 7=Running; 8=Standby; 9=Boot loading; 10=Error"
    default: 7
    min: 0
    max: 10
#  AllowedRoles:
#    default: '["grid", "pvinverter", "genset", "acload"]'
#  Role:
#    description: "grid, pvinverter, genset, acload"    
#    persist: true
#    default: "pvinverter"
```

freakent, 2023-10-19:
@stundenblume Sebastian, any advice you can give?

Woodpeckercz, 2023-10-25:
didn't get much chance to look into it further, I tried uncommenting the Allowed Roles but the setup in the console then just showed bunch of empty choices.