freakent/dbus-mqtt-devices #67: Different service weather forecast

DJZorro, 2023-10-20:
Dear Freakent
Thank you for your driver. It took me a while as newby to figure it out but in the end I got it working.
What I want is to display the weather on my RPi with Venus OS
I would like to publish the weater conditions derrived from Open weather based on the GPS coordinates on the Dbus
From there I would like to display it with the edited Guimods.
Question: how can I get a weather service. Now I only get a temperature /grid/tank etc.
Thks in advance

freakent, 2023-10-20:
Simple answer, you can't. You can only use the services that Victron Venus OS supports on the dbus. All the apis (MQTT, ModBus TCP) are just wrappers around the Venus OS dbus api. These dbus services are all described here. https://github.com/victronenergy/venus/wiki/dbus

DJZorro, 2023-10-20:
Dear Martin,
Thank you for your very fast and clear answer.
Bit disappointing as I thought it was a good idea and a nice project.
Have to give up this and can only hope the Victron will open its doors with
more services.
Thank you for your great driver, learned a lot from it, was hoping I could
publish any service on the dbus but that is obvious not the case.

Best regards from Holland, and I will be one of your future followers..

Rgds,

Paul

closed 2023-10-28 by freakent

DJZorro, 2023-10-30:
Dear Martin,
Just wanted to let you know I got it working with your tip about misusing a GPS and your driver.
Thanks again for the info