freakent/dbus-mqtt-devices #68: Battery support

odaigle, 2023-10-21:
Hi,
  Thanks for your great module. I'm using it to implement support of several sensors in my solar system and I am almost abusing of it.

  I've added battery support with the following addition in service.yml. I could not test everything, but Soc, Voltage, Current, temperature, max charge voltage, max charge and discharge current and power are correctly reported.

battery:
  ProductId:
    default: 65535
  CustomName:
    default: "My MQTT battery"
    persist: true
  ErrorCode: # https://www.victronenergy.com/live/mppt-error-codes
    description: "0=No Error"
    default: 0 # not showing in in console???
  DeviceType:
    description: ""
    persist: true
    default: 0
    min: 0
    max: 2
  Dc/0/Voltage:
    description: "V DC"
    format: "{} V"
  Dc/0/Current:
    description: "A DC"
    format: "{} A"
  Dc/0/Power:
    description: "W"
    format: "{} W"
  Dc/0/Temperature:
    description: "C"
  Dc/0/MidVoltage:
    description: "V DC"
    format: "{} V"
  Dc/0/MidVoltageDeviation:
    description: "V DC"
    format: "{} V"
  Dc/1/Voltage:
    description: "V DC"
    format: "{} V"
  ConsumedAmphours:
    description: "Ah"
    format: "{} Ah"
  Soc:
    description: "%"
  TimeToGo:
    description: "s"
  Info/MaxChargeCurrent:
    description: "A DC"
  Info/MaxDischargeCurrent:
    description: "A DC"
  Info/MaxChargeVoltage:
    description: "V DC"
  Info/BatteryLowVoltage:
    description: "V DC"
  Info/ChargeRequest:
    description: ""
    default: 0
    min: 0
    max: 1
  Alarms/Alarm:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowStarterVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighStarterVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowSoc:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighChargeCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighDischargeCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/CellImbalance:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/InternalFailure:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighChargeTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowChargeTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowCellVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/MidVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/Contactor:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/BMSCable:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighInternalTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/FuseBlown:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Settings/HasTemperature:
    description: ""
    default: 0
    min: 0
    max: 1
  Settings/HasStarterVoltage:
    description: ""
    default: 0
    min: 0
    max: 1
  Settings/HasMidVoltage:
    description: ""
    default: 0
    min: 0
    max: 1
  Relay/0/State:
    description: ""
    default: 0

freakent, 2023-10-21:
Good that you are finding the driver useful. Always good to hear feedback. As you raised an issue is there something you need help with?

odaigle, 2023-10-21:
Hi, thanks for your reply. Perhaps the only "issue" is for you to paste the addition to service.yml to add the support for batteries to the next release, if you consider it is worth. Here is the data with the correct indentation. 

For the context, in my case this allows me to integrate my Renogy 12V 400Ah battery with the Victron system. I am using the Renogy BT to MQTT scripts from https://github.com/cyrils/renogy-bt to send the data to my MQTT broker and Home Assistant, and then publish the battery information to dbus-mqtt-devices to get it to the Cerbo GX.

```
battery:
  ProductId:
    default: 65535
  CustomName:
    default: "My MQTT battery"
    persist: true
  ErrorCode: # https://www.victronenergy.com/live/mppt-error-codes
    description: "0=No Error"
    default: 0 # not showing in in console???
  DeviceType:
    description: ""
    persist: true
    default: 0
    min: 0
    max: 2
  Dc/0/Voltage:
    description: "V DC"
    format: "{} V"
  Dc/0/Current:
    description: "A DC"
    format: "{} A"
  Dc/0/Power:
    description: "W"
    format: "{} W"
  Dc/0/Temperature:
    description: "C"
  Dc/0/MidVoltage:
    description: "V DC"
    format: "{} V"
  Dc/0/MidVoltageDeviation:
    description: "V DC"
    format: "{} V"
  Dc/1/Voltage:
    description: "V DC"
    format: "{} V"
  ConsumedAmphours:
    description: "Ah"
    format: "{} Ah"
  Soc:
    description: "%"
  TimeToGo:
    description: "s"
  Info/MaxChargeCurrent:
    description: "A DC"
  Info/MaxDischargeCurrent:
    description: "A DC"
  Info/MaxChargeVoltage:
    description: "V DC"
  Info/BatteryLowVoltage:
    description: "V DC"
  Info/ChargeRequest:
    description: ""
    default: 0
    min: 0
    max: 1
  Alarms/Alarm:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowStarterVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighStarterVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowSoc:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighChargeCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighDischargeCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighCurrent:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/CellImbalance:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/InternalFailure:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighChargeTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowChargeTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowCellVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/LowTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/MidVoltage:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/Contactor:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/BMSCable:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/HighInternalTemperature:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Alarms/FuseBlown:
    description: "0=OK 0; 1=Warning 1; 2=Alarm 2"
    default: 0
    min: 0
    max: 2
  Settings/HasTemperature:
    description: ""
    default: 0
    min: 0
    max: 1
  Settings/HasStarterVoltage:
    description: ""
    default: 0
    min: 0
    max: 1
  Settings/HasMidVoltage:
    description: ""
    default: 0
    min: 0
    max: 1
  Relay/0/State:
    description: ""
    default: 0

```

freakent, 2023-10-22:
I very much appreciate and welcome contributions from the community, however there are number of reasons why I can not add this to the next release. Please let me explain, as I do not want to discourage contributions but I do have to manage contributions in a controlled way so as not to destabilise the driver for other users. 

1) I am working on a new release that replaces the single services.yml file with individual services files (see Beta). I do not want to add new services now and have to migrate them later. 

2) I will no longer accept contributions that do not come with out a test script. I must be able to regression test all contributions. 

3) with a test system I can not answer questions from other users on this service type. 

4) Issues are not the right place to submit code changes, pull requests are the correct way to merge changes to the code base in a controlled way. 

As I said, I absolutely want to encourage contributions from the community but I want to do it in a controlled way. 

I think the best way to share this with other users for now is to start using the project wiki to share WIP contributions and get more user feedback. I'll look into setting this up.

boumbah, 2023-11-04:
would you be able to elaborate how you got the battery to register as a device? 
I have been trying to do similar with my renogy battery and have got [renogy-bt](https://github.com/cyrils/renogy-bt) running locally on the pi4 running Venus OS, and copied your extra code into the .yaml

Modbus/TCP is now hitting me with:

> ERROR "Error processing function code 3, unit id 1, start address 1307, quantity 16, src 10.40.0.10: Error finding service with device type battery at device instance 1"

Thanks in advance.

odaigle, 2023-11-06:
Hi boumbah, registering the battery as a device is the same as registering another sensor such as a temperature sensor or a grid. Perhaps try registering a simple and officially supported sensor before attempting supporting a battery. Bear in mind that through dbus-mqtt-devices, it is through mqtt you are registering a device and pushing sensor updates, not through modbus.

Also, you have to restart dbus-mqtt-devices after having altered the services.yml file.