freakent/dbus-mqtt-devices #70: PVinverter is shown as "Fronius" in VRM

americanium, 2023-11-01:
Hello!

Not really a problem of this code I think, but when implementing my sungrow inverter as "pvinverter" it is shown as "Fronius" pvinverter.
I have another fronius in use, but i wan't to seperate them in two single inverters within VRM.

Thank you!
Alex

freakent, 2023-11-01:
Without any code examples or screenshots it's impossible to give you any advice.

americanium, 2023-11-02:
Hi!
Sorry for that.

This is my current dashboard
[image]

You see the "Fronius" which is my real fronius added by VRM automatically via (modbus?)

And here my current configuration in node red for testing adding another PV inverter.

[image]

The pvinverter is listed within vrm device overview with extra VRM instance, but the values are added to the already existing fronius inverter.

[image]

Do you need any specific part of code or screenshot ?

Thanks!

freakent, 2023-11-02:
The GX UI is very limited. I suspect it will not show multiple PVInverters. I have no experience with PVInverters but I do have 3 mppts in my installation, they are represented by one Mppt icon and the data aggregated. And they Icon is just a generic Mppt icon, not a picture of my actual MPPT models. 
[image]

>You see the "Fronius" which is my > real fronius added by VRM 
> automatically via (modbus?)
I'm not sure what you mean by this. Nothing is added by VRM, VRM is the portal in Victron's cloud.

americanium, 2023-11-02:
Sorry, I didn't mean VRM but the VenusUI.

I will cheeck victron forum for a solution, here you see pv inverters without "fronius" symbol:
https://community.victronenergy.com/questions/3848/how-to-add-pv-meter-on-ac-out1-multigrid-venus.html