freakent/dbus-mqtt-devices #74: New GUI-Beta requires Status DBus path for Temp sensors

gharshaw, 2023-11-19:
@freakent  
Hi Martin, While reviewing the new gui-beta in v3.20~26 at  http://ipaddress/gui-beta I discovered that gui-beta is using a  "Status" dbus-path for temperature sensors. even though the documentation, https://github.com/victronenergy/venus/wiki/dbus#temperatures,  says this path is for "wired inputs only.   Without the status path  the "Environment" screen is blank.

I added the Status path to the services.yml like so --
  Status:
    description: "0=Ok, 1=Disconnected, 2=Short circuited, 3=Reverse polarity, 4=Unknown"
    persist: true
    default: 0
    min: 0
    max: 4

[image]

Best Regards,
Gene

freakent, 2023-11-19:
Thanks for the info. Where in the UI you posted do you see the status?

gharshaw, 2023-11-19:
@freakent , I defaulted it to 0. 0=Ok
I agree with your thoughts on persist. I'll set it to false and see how the app behaves.

freakent, 2023-11-19:
If you change "persist", make sure you use a different device or service name in the registration so that it creates a new set of settings on the dbus.

johanvancalster, 2024-01-08:
I got the same issue with the new beta version. in the old gui and beta gui.
Nothing shows, exact samen code works perfect on release version of Venus OS.
MQQT explorer gives the result fine of level and remaining, but gui stays blank.
Anything else I can try.

freakent, 2024-01-08:
I'm sorry this issue has been open for such a long time, and was origionally related to the Status attribute. Can you summarise what the problem is your are experiencing currently? If it's not related to the status attribute then pleae open a new issue.

johanvancalster, 2024-01-08:
I'm testing out the beta release of Venus OS, to give the beta gui a try.
When using my working code to publish tank level, I get empty fields see below.
I added the status field like above, but still get empty values.

[image]
[image]