freakent/dbus-mqtt-devices #76: Inverter is not showing Values in VRM

KallerTobias, 2023-12-21:
Hello, I have installed ad set up according to your manual. 
I get the IDs and I can write the values to MQTT (showing up in MQTT Explorer" but VenusOS is not showing the values. 

I currently do not have the inverter speaking to MQTT, the hardware will arrive in a few days, but I have a virtual output I'm writing to MQTT.
I double checked all topics etc. I can't find the problem.

Thank you,

Tobias

[image]
[image]
[image]

stundenblume, 2023-12-21:
Hi Tobias,
maybe try less values for the beginning.
this is how my pvinverter looks like.
[image]
Sebastian

KallerTobias, 2023-12-21:
Hi Sebastian,

thank you for the reply. I now only transfered the Power for L3 and Total, still the same picture:
[image]
[image]

The Topic, starting with "W" is correct, right?

KallerTobias, 2023-12-21:
I will setup a clean system tomorrow, somethin is really strange there. I get the ID "2" for the PV Inverter, but when I analyse the N/.../pvinverter/ Topic I can see that the ID is 0. If I change the ID in my MQTT script, I can write at least the power, but no other Values. Also the MQTT server is breaking down constantly... I think I messed up something.

freakent, 2023-12-21:
Tobias, please share your code you are using to register and send data values.

KallerTobias, 2023-12-22:
HI, 
I send 
`{"clientId":"iv001","connected":1,"version":"v1.0","services":{"i1":"pvinverter"}}`
To the Topic: device/iv001/Status
I Recieved:
`{"portalId":"bxxxxxxxxxxf","deviceInstance":{"i1":"2"}}`

Data I transffered to Venus OS MQTT are e.g.:

Topic:
W/bxxxxxxxxxxf/pvinverter/2/Ac/Energy/Forward
`{"value":0.4}`
or
Topic:
W/bxxxxxxxxxxf/pvinverter/2/Ac/L3/Power
`{"value":460}`

Kind regards, 

Tobias

halve85, 2024-01-14:
which Venus Version you're using?
Release candidate? (had the same issue with that - it's caused due to "dbus MQTT" not supported in future anymore, there will be an alternative.)

freakent, 2024-01-14:
@halve, please do not leave comments like this. It is misleading to people new to Victron MQTT.  It is true that Victron have migrated off Mosquitto, but they have moved to FlashMQ and they functionality is mostly the same.

freakent, 2024-01-14:
@KallerTobias It looks like the driver is working for you and registering your device. 

Can I suggest for test purposes you try setting up a dummy temperature sensor? It's possibly the simplest and will check your set up.

disaster123, 2024-01-14:
> which Venus Version you're using? Release candidate? (had the same issue with that - it's caused due to "dbus MQTT" not supported in future anymore, there will be an alternative.)

This is just wrong. Victron supports mqtt like before. They‘re just using a different software / service.

KallerTobias, 2024-01-14:
> @KallerTobias It looks like the driver is working for you and registering your device. 
> 
> Can I suggest for test purposes you try setting up a dummy temperature sensor? It's possibly the simplest and will check your set up. 

Hi,
Thanks for the support but I switched to a Shelly as EM and the Integrations works much better than the all possible SolaX solutions.
4 years ago when I bought this inverter, I was not aware of the closed system. 
The shelly integration works perfekt.

closed 2024-01-14 by freakent