freakent/dbus-mqtt-devices #87: Investigate status dbus path for temperature

freakent, 2024-02-15:
As per discussion #37

freakent, 2024-02-15:
Temperature devices in Demo mode include this attribute so make sense to add it to the services config.  Same applies to Tanks. Doesn't appear to be affecting display in either of the new guis at this time.

/Status             0=Ok; 1=Disconnected; 2=Short circuited; 3=Reverse polarity; 4=Unknown

```
Status:
  description: "0=Ok, 1=Disconnected, 2=Short circuited, 3=Reverse polarity, 4=Unknown"
  persist: false
  default: 0
  min: 0
  max: 4
```

gharshaw, 2024-02-20:
@freakent, I just reinstalled a fresh Venus O/S 3.30~6  and dbus-mqtt-devices v0.6.4 on a Pi. I needed to go back and add back in the Temps and Tanks status attribute  to services.yml in order for the gui-beta to display properly .

gharshaw, 2024-02-20:
They did appear but my updates to  fluid type and custom name did not update and capacities were incorrect. I believe the devices were being seen as unknown.  I added the status and everything updated correctly.

closed 2024-02-26 by freakent

freakent, 2024-02-26:
I discovered that temeperature sensors were not showing up in VRM without the status attribute. Tanks were shoing up without the status attribute. I decided it safest to include status attribute in services.yml for both tanks and temperature. Hopefully it should also help witht the gui-beta too.