freakent/dbus-mqtt-devices #88: Still no support for FlashMQ?

dersch81, 2024-02-15:
VenusOS v3.20 is now released and after 4 month beta FlashMQ is the new broker. 

As i can see there is still no support for it by dbus-mqtt-devices?

freakent, 2024-02-15:
@dersch81 Please see the announcement I posted yesterday : https://github.com/freakent/dbus-mqtt-devices/discussions/85#discussion-6215729
A similar announecment was also posted on the Victron Modifications forum.

How are you coming to the conculsion the this driver still doesn't support FlashMQ?

dersch81, 2024-02-15:
Oh sorry, didn't notice that in the changelog of 0.6.4. My conclusion was because i have not my custom mqtt devices in place as before. I'm registering the devices with nodered like before but nothing happens in venus dbus.

I can see them here:

`tail -f /var/log/dbus-mqtt-devices/current
@4000000065ce113616496fcc INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce113616c8cd44 INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce117216f2c75c INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce117217447d54 INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce117217677e44 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce117217c9e994 INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce11ae221bfcac INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce11ae223be88c INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce11ae2267ced4 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce11ae229d1b5c INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce11ea149c2944 INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce11ea14d45814 INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce11ea15012ca4 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce11ea15356f8c INFO:device_manager:publish {'portalId': 'b827eb909369', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
`

I can see the W Topic
[image]

The devices are registered in dbus but no values are visible.

freakent, 2024-02-15:
So what version of the driver do you have installed? 

What do you see under the N/portalid>/temperature ?
[image]

dersch81, 2024-02-15:
i'm running 0.6.4 

nothing of my custom mqtt devices are at N. The keep alive is included in my node red flow. I'm running this since more than one year.

freakent, 2024-02-15:
Please can you reboot the GX device then upload the whole fresh log file. Don't cut and paste a few lines it's not enough to see what is happening.

dersch81, 2024-02-15:
```
@4000000065ce186619e5d83c *** starting dbus-mqtt-devices ***
@4000000065ce186d22d601c4 dbus_mqtt_devices v{} 0.6.4
@4000000065ce186d22d61934 INFO:logger:Loglevel set to INFO
@4000000065ce186d22f84734 INFO:dbus_mqtt_devices:-------- dbus_mqtt_devices, v0.6.4 is starting up --------
@4000000065ce186f316a332c INFO:device_manager:Using portalId xxx
@4000000065ce1872070bfe44 INFO:mqtt_gobject_bridge:[Init] Connecting to local broker
@4000000065ce18720f23f80c INFO:device_manager:[Connected] Result code 0
@4000000065ce187b1b0430ec INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce187b1b31345c INFO:device:**** Registering device: hygrometer0776, services: {'t1': 'temperature'} ****
@4000000065ce187b1b3f1edc INFO:device:Registering Service temperature for client hygrometer0776
@4000000065ce187b1b4e805c INFO:device_service_config:About to open config file
@4000000065ce187d09199cdc INFO:device_service:Registering service temperature for client hygrometer0776 at path com.victronenergy.temperature.mqtt_hygrometer0776_t1
@4000000065ce187d0d3bfa44 INFO:settingsdevice:Setting /Settings/Devices/mqtt_hygrometer0776_t1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000065ce187d1abe88e4 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.temperature.mqtt_hygrometer0776_t1
@4000000065ce187d1b3602d4 INFO:device_service:Registered Service com.victronenergy.temperature.mqtt_hygrometer0776_t1 under DeviceInstance 3
@4000000065ce187d1b6150c4 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce187d1da6242c INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce187d1dafc8ec INFO:device:**** Registering device: shellyht1E6D29, services: {'t1': 'temperature'} ****
@4000000065ce187d1dc92904 INFO:device:Registering Service temperature for client shellyht1E6D29
@4000000065ce187d1f410b94 INFO:device_service_config:About to open config file
@4000000065ce187f381ad8ac INFO:device_service:Registering service temperature for client shellyht1E6D29 at path com.victronenergy.temperature.mqtt_shellyht1E6D29_t1
@4000000065ce18800afd974c INFO:settingsdevice:Setting /Settings/Devices/mqtt_shellyht1E6D29_t1/ClassAndVrmInstance does not exist yet or must be adjusted
@4000000065ce188015e844cc INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.temperature.mqtt_shellyht1E6D29_t1
@4000000065ce188016b299ac INFO:device_service:Registered Service com.victronenergy.temperature.mqtt_shellyht1E6D29_t1 under DeviceInstance 1
@4000000065ce188016e98444 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce18b70761a024 INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce18b707ac8404 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce18b707dbaa54 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce18b7081056b4 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce18f30156e34c INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce18f3021e3a8c INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce18f302412fc4 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce18f3026c913c INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
@4000000065ce192f136683bc INFO:device_manager:Received device status message {'clientId': 'hygrometer0776', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '1.0'}
@4000000065ce192f13913184 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 3}} to device/hygrometer0776/DBus, status is 0
@4000000065ce192f13b451b4 INFO:device_manager:Received device status message {'clientId': 'shellyht1E6D29', 'connected': 1, 'services': {'t1': 'temperature'}, 'version': '0.14'}
@4000000065ce192f13f133a4 INFO:device_manager:publish {'portalId': 'xxx', 'deviceInstance': {'t1': 1}} to device/shellyht1E6D29/DBus, status is 0
```

freakent, 2024-02-15:
That looks like a stability issue with the dbus itself. The driver was unable to connect to the dbus.

In the first log file you seem to be re-registering the device multiple times, is there a reason for that?

dersch81, 2024-02-15:
Its pretty strange. Now everything is also at Topic N, all values are frequently written to W but not at N. 

In the first log i manually registered the items with the hope it will start to show values.

freakent, 2024-02-15:
Do you understand what the W, N and R topic branches are used for?
W = Write, N = Notification, R = read request
You should publish your values to W and the dbus will publish it's values to N. That is why you should monitor N.

dersch81, 2024-02-15:
Yes i know that

dersch81, 2024-02-15:
It is working again as before. Cause was my PoE HAT of the Pi and the additional connected USB powered touchscreen....

closed 2024-02-15 by freakent