freakent/dbus-mqtt-devices #96: Setup module dependecies on every reboot from /data/rc.local

freakent, 2024-02-21:
### Version

0.6.4 (current)

### What version of VenusOS are you running on your GX device

3.21 (currect release)

### Does this issue occur during installation or at runtime ?

Installation

### Is there a Discussion thread for this issue? Please paste a link.

https://github.com/freakent/dbus-mqtt-devices/discussions/95

### What is the issue you are actually experiencing?

After a venusOS upgrade the driver can fail with errors indicating missing pyyaml module. If the pyyaml module is missing then pip, python3-misc and pyyaml all need to be re-installed. This check can be automated at system boot up in /data/rc.local

gharshaw, 2024-02-21:
@freakent Are you aware of @kwindrem setuphelper? It's a package manager that reinstalls packages after Venus upgrades

freakent, 2024-02-21:
(moving this to discussions)
Yes I am aware. It's certainly on my radar but other priorities first.