freakent/dbus-mqtt-devices #110: Add support for EVCharger

freakent, 2024-03-06:
### Version

0.6.6 (current)

### What version of VenusOS are you running on your GX device

3.21 (current release)

### Does this issue occur during installation or at runtime ?

Other

### Is there a Discussion thread for this issue? Please paste a link.

https://github.com/freakent/dbus-mqtt-devices/issues/32#issuecomment-1979792013

### What is the issue you are actually experiencing?

Users are using the driver to add EV Chargers to their system using a customised service description. With each new release of the driver they have to manually add the service description to their services.yml file. 

The service description currently in use needs refining before inclusion in the released version. 

```
# https://github.com/victronenergy/venus/wiki/dbus#evcharger
evcharger:
  ProductId:
    default: 65535 # 0xFFFF
  CustomName:
    default: "MyCharger"
    persist: true
  Ac/Energy/Forward:
    description: "wh"
    format: "{:.2f}W"
  Ac/L1/Power:
    description: "W"
    format: "{} W"
  Ac/L2/Power:
    description: "w"
    format: "{} W"
  Ac/L3/Power:
    description: "w"
    format: "{} W"
  Ac/Power:
    description: "w"
    format: "{} W"
  AutoStart:
    description: "w"
    default: 0
#    persist: true
  ChargingTime:
    description: "in seconds"
    format: "{} s"
  Current:
    description: "A"
    default: 0
#    persist: true
  EnableDisplay:
    description: ""
  MaxCurrent:
    description: "A"
    default: 0
#    persist: true   #wird nicht gespeichert reset nach disconnect/reconnect
  Mode:
    description: ""
    default: 0
  Model:
    description: "w"
    default: "AC22NS"
  Position:
    persist: true
    default: 0
    min: 0
    max: 1
    description: "0=OUTPUT; 1=INPUT"
#  Connected:
#    description: "w"
  Role:
    description: "w"
  SetCurrent:
    description: "w"
#  StartStop:
#    description: "w"
#    persist: true
  Status:
    description: "w"
    default: 1
```

freakent, 2024-03-06:
Things to do
1) confirm the use of “persist: true” across all attributes
2) use the correct descriptions from the Victron documentation 
3) confirm why some fields are commented out entirely 
4) create a script to test/verify the configuration is working

freakent, 2024-03-07:
With correct descriptions:
```
# https://github.com/victronenergy/venus/wiki/dbus#evcharger
evcharger:
  ProductId:
    default: 65535 # 0xFFFF
  CustomName:
    default: "MyCharger"
    persist: true
  Ac/Energy/Forward:
    description: "wh"
    format: "{:.2f}W"
  Ac/L1/Power:
    description: "L1 Power used (W)"
    format: "{} W"
  Ac/L2/Power:
    description: "L3 Power used (W)"
    format: "{} W"
  Ac/L3/Power:
    description: "L3 Power used (W)"
    format: "{} W"
  Ac/Power:
    description: "AC Power (W)"
    format: "{} W"
  AutoStart:
    description: "w"
    default: 0
#    persist: true
  ChargingTime:
    description: "in seconds"
    format: "{} s"
  Current:
    description: "A"
    default: 0
#    persist: true
  EnableDisplay:
    description: ""
  MaxCurrent:
    description: "A"
    default: 0
#    persist: true   #wird nicht gespeichert reset nach disconnect/reconnect
  Mode:
    description: ""
    default: 0
  Model:
    description: "w"
    default: "AC22NS"
  Position:
    persist: true
    default: 0
    min: 0
    max: 1
    description: "0=OUTPUT; 1=INPUT"
#  Connected:
#    description: "w"
  Role:
    description: "w"
  SetCurrent:
    description: "w"
#  StartStop:
#    description: "w"
#    persist: true
  Status:
    description: "w"
    default: 1
```

freakent, 2024-03-07:
Settings - aka where to use persist=true

CustomName
AutoStart ??
MaxCurrent ??
Mode
Model
Position
Role

dersch81, 2024-03-07:
Sure, i have a quite big NodeRed Flow for this to read via MQTT the data from the Wallbox (i'm using two) in that way and registered two EVcharger. I can also control the charging mode via VRM, because i have mapped the mode ID's between VenusOS and OpenWB. But should work with any other "open" Wallbox, too.  

[image]

[EVCharger Flow 1.json](https://github.com/freakent/dbus-mqtt-devices/files/14524345/EVCharger.Flow.1.json)
[EVCharger Flow 2.json](https://github.com/freakent/dbus-mqtt-devices/files/14524346/EVCharger.Flow.2.json)

In VRM they appear like this:
[image]

Example of local Venus GUI:
[image]

MQTT Explorer:
[image]

[image]

dbus-spy:
[image]

[image]

freakent, 2024-03-07:
In your nodered flows, what does Gate do?

dersch81, 2024-03-07:
In that flow the gate is just preventing to flood the register with zero values. It opens only in case of >0. 

In general i'm using gates a lot to control main flows with subflows.

dersch81, 2024-03-07:
> In that flow the gate is just preventing to flood the register with zero values. It opens only in case of >0.
> 
> In general i'm using gates a lot to control main flows with subflows.

I'm sorry my information about the gate was wrong. It is controlling the counting of charging time. When the current is >0 then it opens the gate and the timestamp payload is able to flow. If current is =0 then the gate is closed and time counting stopps. The next process is starting from time 0 again.

freakent, 2024-03-08:
These are the attributes that need persist = true

Mode
SetCurrent
MaxCurrent
Position
AutoStart
EnableDisplay
CustomName 

[image]

dersch81, 2024-03-08:
persist = true just means to have the same state after restart, right?

freakent, 2024-03-09:
Yes, which also affects the device specific setup screen on the GX device (as per my screenshots above). If the correct fields are not set as persist = true the setup screen will not work correctly. It might have been better to call this parameter "setting" instead of "persist" to more accurately affect what it does.

freakent, 2024-04-26:
Fixed in v0.8.0.

closed 2024-04-26 by freakent