freakent/dbus-mqtt-devices #119: Temperature value as string

mhcore-m, 2024-05-13:
### Version

0.6.6 (current)

### What version of VenusOS are you running on your GX device

3.2

### Does this issue occur during installation or at runtime ?

Runtime

### What is the issue you are actually experiencing?

If you send a string temperature value to the W node, local display of the temperature works as expected, but the VRM data transfer dies and the Victron vrmlogger daemon causes high CPU load on Cerbo. I did not find any similar issue, but some that were relate to high CPU usage, which maybe have the same reason?

freakent, 2024-05-13:
Why are you sending a temperature value as a string? 

If the problem occurs when publishing to a "W/..." topic then that is Victron's flashmq not this driver. This driver only handles registration. 

P.s please upgrade to the latest version of the driver.

mhcore-m, 2024-05-13:
My node-red flow provided a string... ;) 
...and it seemed to work.  

Ok, but I got the point with the Victron flashmq, I will raise a issue there, then. Thanks.

freakent, 2024-05-13:
Eve if you are using nodered you should send the value as a number. If you are constructing the data as JSON, don't put quotes around the value. If you look in the latest v0.8.0 version of this driver there are example nodered flows in the samples folder.