freakent/dbus-mqtt-devices #120: Error while installing

devilmastah, 2024-05-18:
### Version

Other

### What version of VenusOS are you running on your GX device

Other

### Does this issue occur during installation or at runtime ?

Installation

### What is the issue you are actually experiencing?

When running the install script i run against following error, its the latest version as prescribed in the readme on this repo:

`ython3-pip.
dbus-mqtt-devices: Pip install module dependencies
Collecting PyYAML~=6.0
  Downloading PyYAML-6.0.1.tar.gz (125 kB)
     |████████████████████████████████| 125 kB 1.6 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", lin                                          e 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", lin                                          e 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line                                           177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line                                           333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line                                           282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", l                                          ine 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", l                                          ine 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py",                                           line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py",                                           line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 1                                          51, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 2                                          55, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last                                          ):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", lin                                          e 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-build-env-61vbp_88/overlay/lib/python3.8/site-pack                                          ages/setuptools/__init__.py", line 17, in <module>
    from .dist import Distribution
  File "/var/volatile/tmp/pip-build-env-61vbp_88/overlay/lib/python3.8/site-pack                                          ages/setuptools/dist.py", line 27, in <module>
    from .extern.packaging.markers import InvalidMarker, Marker
  File "/var/volatile/tmp/pip-build-env-61vbp_88/overlay/lib/python3.8/site-pack                                          ages/setuptools/_vendor/packaging/markers.py", line 11, in <module>
    from ._parser import (
  File "/var/volatile/tmp/pip-build-env-61vbp_88/overlay/lib/python3.8/site-pack                                          ages/setuptools/_vendor/packaging/_parser.py", line 10, in <module>
    from ._tokenizer import DEFAULT_RULES, Tokenizer
  File "/var/volatile/tmp/pip-build-env-61vbp_88/overlay/lib/python3.8/site-pack                                          ages/setuptools/_vendor/packaging/_tokenizer.py", line 3, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'

dbus-mqtt-devices: Set up Victron module libraries
dbus-mqtt-devices: Set up device service to autorun on restart
dbus-mqtt-devices: Adding device service to /data/rc.local
dbus-mqtt-devices: Setup complete
root@nanopi:/data/drivers# substitutions:
-sh: substitutions:: command not found
root@nanopi:/data/drivers#   device_name: "esp32_DSMR"
`

I am trying to run this on a brand new Multiplus II GX version. do not know if this has to do with it. 
When i do the last step catting rclocal this seems fine.

freakent, 2024-05-18:
I'm unsure why you are getting this error as DataClasses has been added as a dependency in requirements.txt. I would like to see the full log from the setup.sh. 

If you just want to get up and running quickly please check the troubleshooting section in the README. 

Troubleshooting

during installation

If you receive an error during setup that includes the lines

ModuleNotFoundError: No module named 'dataclasses'
Try running the following before running the setup.sh script again.

opkg install python3-modules

freakent, 2024-05-18:
I'm also unsure what this log entry refers to:

substitutions:
-sh: substitutions:: command not found
root@nanopi:/data/drivers# device_name: "esp32_DSMR"

Can you say what was happening here?

devilmastah, 2024-05-18:
Oh sorry, the last part should not have been included i accidently copy pasted some text in my putty window after the setup :) 
I will try the opkg command.

freakent, 2024-05-18:
One last thought from me is about using this on Multiplus II GX. I'm not sure how much RAM the venusos has available to it on the Multiplus GX. Keep an eye on system stability. If you get unexpected reboots you might not have enough RAM to support additional drivers etc. I had a similar issue with my original CCGX.

devilmastah, 2024-05-18:
If this happens i will post it here so you know :) Thought it was worth a try. 
dataclasses problem is solved, i completely missed it in the documentation

devilmastah, 2024-05-18:
@freakent just to let you know, its been running perfectly stable for the past 9 hours on a Multiplus-II 3000 GX. In combination with the esp project https://github.com/NuKeMtje/ESPHomeDSMRVictron this saved me 100 bucks on a energy meter + extra cabling effort :)

closed 2024-05-18 by freakent