freakent/dbus-mqtt-devices #122: installation fails

skrebber, 2024-05-29:
### Version

0.8.0 (current)

### VenusOS version

3.32

### Does this issue occur during installation or at runtime ?

Installation

### What is the issue you are actually experiencing?

The installation script fails. 

Really nothing more to say, the readme is really clear, what to do. And what has to be done is few, only download and unzip and run the script. Therefore I'm sorry, that I can't provide more details. 

### Relevant log output

```shell
root@einstein:/data/drivers# ./dbus-mqtt-devices-0.8.0/bin/setup.sh
dbus-mqtt-devices: Setup in /data/drivers/dbus-mqtt-devices-0.8.0 started
dbus-mqtt-devices: Checking to see if Python's Pip is installed
/usr/bin/python: No module named pip
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/einstein/Packages.gz.
Updated source 'einstein'.
Installing libjitterentropy2 (2.2.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libjitterentropy2_2.2.0-r0_cortexa7hf-neon-vfpv4.ipk.
Installing libsysfs2 (2.1.0) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/libsysfs2_2.1.0-r5_cortexa7hf-neon-vfpv4.ipk.
Installing python3-pydoc (3.8.18) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-pydoc_3.8.18-r0_cortexa7hf-neon-vfpv4.ipk.
Installing rng-tools (6.9) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/rng-tools_6.9-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-xmlrpc (3.8.18) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-xmlrpc_3.8.18-r0_cortexa7hf-neon-vfpv4.ipk.
Installing python3-pip (20.0.2) on root
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/python3-pip_20.0.2-r0_cortexa7hf-neon-vfpv4.ipk.
Configuring libjitterentropy2.
Configuring libsysfs2.
Configuring rng-tools.
 Adding system startup for /etc/init.d/rng-tools.
Starting random number generator daemon.
Configuring python3-pydoc.
Configuring python3-xmlrpc.
Configuring python3-pip.
dbus-mqtt-devices: Pip install module dependencies
Collecting PyYAML~=6.0
  Downloading PyYAML-6.0.1.tar.gz (125 kB)
     |████████████████████████████████| 125 kB 667 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-build-env-y03n4zqh/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 16, in <module>
    from .dist import Distribution
  File "/var/volatile/tmp/pip-build-env-y03n4zqh/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 27, in <module>
    from .extern.packaging.markers import InvalidMarker, Marker
  File "/var/volatile/tmp/pip-build-env-y03n4zqh/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/markers.py", line 11, in <module>
    from ._parser import (
  File "/var/volatile/tmp/pip-build-env-y03n4zqh/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_parser.py", line 10, in <module>
    from ._tokenizer import DEFAULT_RULES, Tokenizer
  File "/var/volatile/tmp/pip-build-env-y03n4zqh/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_tokenizer.py", line 3, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'

dbus-mqtt-devices: Set up Victron module libraries
dbus-mqtt-devices: Set up device service to autorun on restart
dbus-mqtt-devices: Adding device service to /data/rc.local
dbus-mqtt-devices: Setup complete
```

skrebber, 2024-05-29:
sry, rtfm.

closed 2024-05-29 by skrebber

freakent, 2024-05-29:
Thanks for reporting this. I thought I had fixed this problem once and for all with this release. Sounds like I need to look into why that library doesn't get automatically installed again.