freakent/dbus-mqtt-devices #126: Error during installation

schwalbe13, 2024-08-08:
### Version

0.8.0 (current)

### VenusOS version

v3.50~13 beta release

### Does this issue occur during installation or at runtime ?

Installation

### What is the issue you are actually experiencing?

After downloading and unzipping, the following mentioned to use installation command is not successful
`./dbus-mqtt-devices-0.8.0/bin/setup.sh`

### Relevant log output

```shell
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-install-ud95nhdj/PyYAML/packaging/_pyyaml_pep517.py", line 50, in <module>
    _bridge_build_meta()
  File "/var/volatile/tmp/pip-install-ud95nhdj/PyYAML/packaging/_pyyaml_pep517.py", line 8, in _bridge_build_meta
    from setuptools import build_meta
  File "/var/volatile/tmp/pip-build-env-hl4454n4/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 21, in <module>
    from .dist import Distribution
  File "/var/volatile/tmp/pip-build-env-hl4454n4/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 25, in <module>
    from packaging.markers import InvalidMarker, Marker
  File "/var/volatile/tmp/pip-build-env-hl4454n4/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/markers.py", line 13, in <module>
    from ._parser import MarkerAtom, MarkerList, Op, Value, Variable
  File "/var/volatile/tmp/pip-build-env-hl4454n4/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_parser.py", line 12, in <module>
    from ._tokenizer import DEFAULT_RULES, Tokenizer
  File "/var/volatile/tmp/pip-build-env-hl4454n4/overlay/lib/python3.8/site-packages/setuptools/_vendor/packaging/_tokenizer.py", line 5, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
```

freakent, 2024-08-08:
Please read the Trouble Shooting section of the ReadMe.

schwalbe13, 2024-08-13:
Thank you, it's working now by following the Troubleshooting section.

closed 2024-08-13 by schwalbe13