freakent/dbus-mqtt-devices #131: Minor issue: PV Inverter doesn't show power, nor phases

kds69, 2024-11-02:
### Version

0.8.0 (current)

### VenusOS version

3.50

### Does this issue occur during installation or at runtime ?

Runtime

### What is the issue you are actually experiencing?

Not sure this is a dbus-mqtt-service problem or a Venus OS problem.
Power per phases is not displayed on Venus monitor dashboard/overview. Instead it is shown as "undefined: 0 W"
Btw, I suspect this was already an issue since long, as since v2.8x already I never management to get power per phases displayed though it is well reported in other screens (see below screen print). But it doesn't seems to cause any harm anyway, so more of cosmetic issue on new GUI where it appears in an obvious manner now:

[image]

whereas:
[image]

...and thanks you so much for all this fantastic job btw!!

kds69, 2024-11-02:
what I wrote above is not fully correct: phases to show in the old gui, but on the alternative dashboard. see below:
[image]

freakent, 2024-11-02:
Hi, you'll have to help me understand the screen shots better. The PV Inverter config was community provided so I've never used it myself. 

Is the "undefined" valued an attribute value that you send via mqtt or is it a derived value? The reason I ask is that it could be the attribute is not properly defined or missing in the settings.yaml file.

freakent, 2024-11-02:
I don't have a value displayed under solar yield to know what should be displayed there. 

[image]

kds69, 2024-11-02:
Thanks for your quick reply!!
I think the issue is on Venus OS, not your driver, thus you can disregard and I'll check with the victron community rather.

I have a pv inverter connected to AC input of a Multiplus II. 
May I bet you have a mppt connected to your battery instead..(+a wind turbine or generator??)

kds69, 2024-11-03:
this will is fixed in Venus beta 3.51:
[https://community.victronenergy.com/t/venus-os-beta-series-announcement-and-changelog-v3-51-2/10041/2](url)

closed 2024-11-03 by kds69