freakent/dbus-mqtt-devices #132: if off-line the setup-dependencies.sh prevents rc.local to complete ansd create symlink to /service/dbus-mqtt-devices

kds69, 2024-11-03:
### Version

0.8.0 (current)

### VenusOS version

3.50

### Does this issue occur during installation or at runtime ?

Runtime

### What is the issue you are actually experiencing?

It seems drivers failed to work if GX is offline.
Most likely the symlink to latest drivers is not created (rc.local timeout?)
Tested by forcing in the command line the symlink creation after reboot:
`ln -s /data/drivers/dbus-mqtt-devices-0.8.0/bin/service /service/dbus-mqtt-devices`

Maybe 10 attempts every minute is too much? this is 1h... 
May I suggest to either reduce the number of attempts and time between attempts or add a "&" to let the symlink be created anyway. I believe this is harmless...

freakent, 2024-11-03:
The setup script does a lot more than just create the symlink. It also installs the dependencies from the internet. If you don't have an internet connection it won't install.  I remember thinking that retrying every 10 seconds was a bit brutal but it's such an edge case it's not worth over complicating it any more than that.

kds69, 2024-11-03:
sorry I didn't mentioned in the issue detail, only in the title: I am not speaking about the set-up script, speaking about the setup-dependencies.sh, which runs at every reboot/start-up.
+I figured out it is in reality a endless loop due to a missing "t" in "atempts"...:

`attempts=$((atempts + 1))`

to be replaced with

attempts=$((attempts + 1))

freakent, 2024-11-03:
Ah I see, that is a bug. Well spotted.

freakent, 2025-04-08:
Fixed in release[ 0.9.0-rc1](https://github.com/freakent/dbus-mqtt-devices/discussions/148).

closed 2025-04-08 by freakent